SpenceKonde/DxCore #123: prog.py verbose error

Di-Ny, 2021-06-29:
Hi, 
I have this issue when trying to upload via SerialUPDI method. It comes independent from any wiring issue.

Error message : 
    prog.py: error: unrecognized arguments: {program.verbose}

SpenceKonde, 2021-06-29:
Bizarre, whih version of the ide is that?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

Di-Ny, 2021-06-29:
I am using Visual Micro V2021.606.4 on VStudio Community 2019. I background Arduino 1.8.3. The full error message in the console :
Uploading 'Epheromon' to 'AVR DB-series (Optiboot)' using 'SerialUPDI with 4.7k resistor or diode (230400 baud)'
usage: prog.py [-h] [-a ACTION] [-b BAUDRATE] [--blocksize BLOCKSIZE]
               [-d DEVICE] [--fuses FUSES [FUSES ...]] [--fuses_print]
Error during upload using programmer
The uploader process failed
The uploader returned an error
               [-f FILENAME] [-t TOOL] [-u UART] [-s SERIALNUMBER] [-v]
prog.py: error: unrecognized arguments:   {program.verbose}

SpenceKonde, 2021-07-05:
Do you not get the same error with avrdude? o_o

It looks like it's not resolving program.verbose. which is supposed to be set by the IDE....

Di-Ny, 2021-07-05:
Ok I ran further tests: 
- it works correctly with the Arduino IDE : if this runs avrdude in background, then the problem is not with avrdude.
- and it also works correctly with my setup by changing the programmer to ATMEL-ICE.

The issue seems to be related to the SerialUPDI programmers on Visual Micro. 
But I wonder if there wouldn't be other compitibilty issues. I don't find the "Export binaries" button in Visual Micro (and I have it on the Arduino IDE), and there is no EESave option either.

SpenceKonde, 2021-07-06:
Export binaries is a feature of the IDE, so it's no surprise that that would be different (though it's a crying shame if they don't support that - especially with how I've pimped out that functionality with the full configuration specified in the filename, and memory map and assembly listing generation!)

But EESave should *really* be there if my other menus are! Are you sure you're nnot looking at an Optiboot definition on one of them? EESave is not present on Optiboot board definitions, because a chip erase never occurs when programming through the bootloader, so it's irrelevant.

Di-Ny, 2021-07-08:
My bad, of course it is only on no bootlaoder baords. So no issue with EESave.
Still the issue with SerialUPDI though.
By removing _{program.verbose}_ and _{bootloader.verbose}_ from _tools.pymcuprog.bootloader.pattern_ (l.193) in plateform.txt stored in "AppData\Local\arduino15\packages\DxCore\hardware\megaavr\1.3.6"  it works. 

BTW, it is the same issue MegaTinyCore, and also on an other computer with the same configuration. 
There is no issue with the avrdude programmers, but only with pymcuprog, where _{program.verbose}_ is not replaced by it's value in VisualStudio.

SpenceKonde, 2021-07-08:
Wow. Okay then. Well;. I guess I'll just do that. At present verbose uploads has no effect on SerialUPDI. The minimum output from SerialUPDI has about as much info as verbose output on avrdude does. and if the actual verbose modes are enabled, the output is *INSANELY* verbose and there are probably less than a dozen people who who would glean anything useful from  that wall of repetitive test/

SpenceKonde, 2021-07-08:
That said, I think this is a visualmicro bug too.

SpenceKonde, 2021-08-25:
This was fixed back in july (buit not in any release)/

closed 2021-08-25 by SpenceKonde