SpenceKonde/DxCore #124: Ability to link against archives

simengangstad, 2021-07-06:
Hello,

I'm in the situation where I want to link against an archive. What do you think of adding this to platform.txt?

`compiler.extra_archive_paths=`

 `recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" {compiler.extra_archive_paths} "-L{build.path}" -lm`

If I understand the project setup correctly, then one could just override these paths in platform.local.txt.

SpenceKonde, 2021-07-06:
I see no harm in it... Presumably you have a specific set of archives in mind.... Do you mind my asking what these are?

I'm no expert on the C details of the compilation process; I've only recently gotten to the point where I can start playing with things like thing, instead of just ganking configurations from others. As I understand would be precompiled files that, for whatever reason, can't be obtained as and compiled from source (like the Q-touch stuff?) or is this something different? We have long dreamed of getting the PTC working on megaTinyCore in particular (I'm sure MCUDude, who maintains cores for all the megaAVR parts that are plausible Arudino platforms, as well plus the ATtiny13 which I'm not interested in (I don't touch anything smaller than 2k flash with a ten foot pole - and only grudgingly support the 2k ones - it just involves so much optimization to the the Arduino bloat down to a reasonable size).  But the fact that it's all pre-compiled has stymied this so far; I know it's possible (the guy who did TinyCore did it for two of the - but I'm pretty sure he's got a grudge against me, and possible most of Arduino; some drama went down IIRC, so I think the chances that I'd be able to extract the secrets from him are close to 0.)

SpenceKonde, 2021-07-06:
If there is some pile of archive files that would unlock more of the true potential of these parts, maybe they should  be included in the core straight up? 

In any event, would it be useful to put this into megaTinyCore? I ask because there should be a release there later today, (god I hope so - I'm like 2 weeks late, and certain people are getting impatient).

simengangstad, 2021-07-06:
I'm linking against cryptoauthlib. I'm not too keen on forking it and porting it over to a arduino library format/including just the files, as it gets messy quickly and it will be more to maintain further on. I currently just build it with a custom CMakeLists outside the its repo with avr-gcc and link it with the setup I posted above with arduino-cli. 

That works when using arduino-cli directly and setting the flag using `build-property`. Though after looking at the platform.local.txt file requirements, does it seem like that file has to be placed along in the DxCore's platform.txt for this to work out of the box in the Arduino IDE (https://forum.arduino.cc/t/additional-compiler-options/631297/9). I might be missing something?

My goal is to build a library for supporting ATECC-devices on top of your core, so ideally the users of it would just need to place the library in their library folder and hopefully without copying this custom platform.local.txt to the core's directory. Feel free to await including this until further, as this might not be the optimal solution within a pure Arduino IDE environment.

SpenceKonde, 2021-07-08:
ATECC?  Reference about what you[re talking about?

What's most interesting to me is whether being able to use archive files would enable more widely applicable new features in Arduino cores - I checked and that is indeed how the core that got QTouch to work on some parts did it (though more stuff around it was needed to make it work and I never understood how he got there from here 

Also, you're that dude with Microchip who I was told would be contacting me? You should have been given my email (spencekonde@gmail.com - feel free to use that to contact me too.

simengangstad, 2021-07-08:
The secure element chip: https://www.microchip.com/wwwproducts/en/ATECC608B

Yeah, no doubt it could be useful. For almost everyone, making sure the platform.local.txt file is where it is supposed to be isn't a problem. We might just do it this way and include some small script to place it correctly for the library. 

Yeah, that's correct. I've sent you an email :)

SpenceKonde, 2021-07-17:
How long is the list of things that require an archive file to link against for support?
I am not firmly opposed to including a pile of archive files with the core - it is not the most graceful, and I'm sure it is stomach-turning to many opensource purists, but I care more about producing  something that *works* rather than proving my opensource ideology. If I'm legally in the clear to include it with the core, and we can't come up with a more graceful way to do it.... so be it. I already drag along a standalone copy of python a heavily modfiied version of pymcuprog and all of it's dependencies and my own bloody toolchain package versions. I wonder if it could even be handled by an extra, dummy "board package" that doesn't add any real boards, but installs the archive file(s) via tools-dependencies (I mean, that I know we can do) the question is whether we could get platform.txt to use it.

simengangstad, 2021-07-18:
I wouldn’t include that archive into the core, as it is not related to the core specifically. It will be just more to maintain and slow down the upload time for people building on the core. I think from our side we can just do the copying of the platform.local.txt in some setup script for the library we’re building. But having the option there at least people can decide if they want to link against something extra if they wish so.

SpenceKonde, 2021-07-28:
Aaah, I see.Qould it be  spend those checks as money to waste, they sazw a holy calling.

SpenceKonde, 2021-07-28:
Wouldn't the Not sure wjhere the post I deleted came from., it was words, but didn't make sense lexically. Copy paste? Autocorerect + zuzu's feet?

simengangstad, 2021-07-30:
Hahah, these computers right 😛 But anyway, what do you feel about including this (just the two lines in the first post)?

SpenceKonde, 2021-07-30:
Yeah, I can do that. 

.... I wonder if there's a way.... to at that point in the process, locate the sketch , or the libraries folder, or something like that - so instead of reaching into the void if it wasn't overwritten by the platform.txt.local......

SpenceKonde, 2021-07-30:
Actually - if it *can't* at that point locate the sketch folder itself, that tells you that that gets defined literally betweemn that and the next os command.

simengangstad, 2021-08-06:
The way I do it now is to append this to platform.local.txt:

```
compiler.c.extra_flags=-I{build.source.path}/lib/cryptoauth/cryptoauthlib/lib/ -I{build.source.path}/lib/cryptoauth/build/cryptoauthlib/lib
compiler.cpp.extra_flags=-I{build.source.path}/lib/cryptoauth/cryptoauthlib/lib/ -I{build.source.path}/lib/cryptoauth/build/cryptoauthlib/lib
combine.extra_paths={build.source.path}/lib/cryptoauth/build/cryptoauthlib/lib/libcryptoauth.a
```

The build.source.path points to the sketch's path, so it does know where things are. I have some time to look at this more thoroughly today, I'll let you know if I find something useful.

simengangstad, 2021-08-06:
So I found  a way to do this without messing with the platform.local.txt. Arduino has some functionality for it: https://arduino.github.io/arduino-cli/latest/library-specification/#precompiled-binaries when building a library targeting some core. The only change needed is to add some extra lines in the platform.txt. I've set up a PR with this. This makes this whole process far more streamlined for me, now I can just compile directly without messing with any of the files related to the core😄

SpenceKonde, 2021-08-06:
Awesome!! Would you mind sharing how to use it? Maybe a .md file in
megaavr/extras?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

simengangstad, 2021-08-09:
Added some documentation about it now, here: https://github.com/simengangstad/DxCore/blob/master/megaavr/extras/LinkingAgainstStaticLibrary.md

SpenceKonde, 2021-08-11:
Thanks, That content will likely be moved around and rephrased, since a key thing I hadn't realized was that only the library authors need concern themselves with it, among other things (let's not kid ourselves, nobody is going to turn their sketch into a library to save a few seconds of compile time if it costs them any flash - we all want to do the opposite! It's so libraries with closed source code supplied in the form of archives is usable.
Anyway, I added this to megaTinyCore.too, and pending one internally discovered issue investigation, that is ready to go for 2.4.0 releae, fucking finally. The release cycle that would not end.

closed 2021-08-11 by SpenceKonde

simengangstad, 2021-08-11:
Cool, I can make a small example for this. It looks like compiler.libraries.ldflags are missing in megaTinyCore by the way: https://github.com/SpenceKonde/megaTinyCore/blob/7af03650172901c6fec97750617faa7d396f4880/megaavr/platform.txt#L80