SpenceKonde/DxCore #126: AVR32DB48 not supported by pymcuprog

grandaspanna, 2021-07-20:
Only device I could get quickly :-)
Ouput from console:
```
Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port COM10 at 230400 baud.
Target: avr32db48
Set fuses: ['5:0b11001001', '7:0x00', '8:0x00']
Action: write
File: C:\Users\i007544\AppData\Local\Temp\arduino_build_793474/i2c_send4ints.ino.hex
Unable to setup stack for device avr32db48
Currently supported devices (in 'devices' folder):
atmega1608, atmega1609, atmega3208, atmega3209, atmega328p, atmega4808, atmega4809, atmega808, atmega809, atsamd21e18a, atsamd21g18a, attiny1604, attiny1606, attiny1607, attiny1614, attiny1616, attiny1617, attiny1624, attiny1626, attiny1627, attiny202, attiny204, attiny212, attiny214, attiny3216, attiny3217, attiny402, attiny404, attiny406, attiny412, attiny414, attiny416, attiny417, attiny804, attiny806, attiny807, attiny814, attiny816, attiny817, attiny84a, avr128da28, avr128da32, avr128da48, avr128da64, avr128db28, avr128db32, avr128db48, avr128db64, avr32da28, avr32da32, avr32da48, avr64da28, avr64da32, avr64da48, avr64da64, pic16f15244, pic16f15376, pic16f1768, pic16f1779, pic16f18446, pic16f18456, pic16lf18456, pic18f16q40, pic18f16q41, pic18f47k40, pic18f47k42, pic18f47q10, pic18f47q43, pic18f57q43, pic18f57q84, pic24fj128ga705, pic24fj64gu205, uc3a3256
Error: 
Error: 
```

SpenceKonde, 2021-07-20:
This will be fixed either by merging the latest microchip pymcuprog
definitions into the tool before the next release, or me just stuffing the
files we need in if we haven't gotten it sorted out in time for that
version.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

grandaspanna, 2021-07-21:
FWIW, and in case it helps anyone, I had a crack at it and seems to work.
[avr32db48.zip](https://github.com/SpenceKonde/DxCore/files/6853029/avr32db48.zip)

mraardvark, 2021-07-28:
It would probably be a good idea to log an issue on the pymcuprog github repo when device support is lagging
https://github.com/microchip-pic-avr-tools/pymcuprog/issues
(norwegian summer-holiday lag)

grandaspanna, 2021-07-28:
Thanks - makes sense. I'll head over there now
(Sydney lockdown boredom).

SpenceKonde, 2021-07-28:
The official version has support we just don't have that copied here yet.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

mraardvark, 2021-07-28:
Ah, ok.
(There is a lag between the pypi package and github repo which threw me off...)

SpenceKonde, 2021-07-28:
I thought the issue was that the official version supported it, it was just that the version in here didn't. 

Which would make sense since we're still using a very old codebase because our attempt to merge our changes with the github pymcuprog release took several agonizing hours.... it worked in the cases we tested... but everything else was totally hosed, and I needed to have something I could work with and release, and I continued using the version based on the old codebase until Quentin and I have time to do it correctly.

SpenceKonde, 2021-07-28:
I lacked both time and ability (and at the time, patience was also in extremely short supply as I recall) to reattempt that., but the code must upload,, so the codebase I'm using remains the old one, which has been so severely hacked up, mostly by me, that I don't call what the core uses pymcuprog anymore, I call it SerialUPDI, based on pymcuprog. (that's the only mode that I have any confidence would attempt to work, considering all the changeswe made and that we test only with the uart mode, since that's what we caare about.. So I decided to call it something even more generic. which highlights the Serial part, and doesn't mention anything else, because that other stuff is likely broken now :-)