SpenceKonde/DxCore #127: pinMode is broken

mikrocoder, 2021-08-04:
Hi,

your pinMode function with INPUT_PULLUP sets the wrong bits. It sets the bits in the OUT register. But it must set the bits in the corresponding PINnCTRL registers. Please test the pinMode method again completely.

```cpp
Serial2.swap(1);            // PF4 TXD2 / PF5 RXD2
Serial2.begin(250000);                
Serial2.println("\nµC Reset #### #### ####");
  
showPORTARegister(Serial2, 0);
  
PORTA.PIN3CTRL |= PORT_PULLUPEN_bm;
PORTA.PIN2CTRL |= PORT_PULLUPEN_bm;

showPORTARegister(Serial2, 0);

pinMode(PIN_WIRE_SDA, INPUT_PULLUP);
pinMode(PIN_WIRE_SCL, INPUT_PULLUP);

showPORTARegister(Serial2, 0);
```

Result:
```cpp
µC Reset #### #### ####
PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
IN:        0b0001'1100 0x0408
PINCONFIG: 0b0000'0000 0x040B
PIN2CTRL:  0b0000'0000 0x0412
PIN3CTRL:  0b0000'0000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
IN:        0b0001'1100 0x0408
PINCONFIG: 0b0000'0000 0x040B
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'1100 0x0404
IN:        0b0001'1100 0x0408
PINCONFIG: 0b0000'0000 0x040B
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413
```

SpenceKonde, 2021-08-05:
Your test case does not demonstrate a problem. If you were to run your test case without pre-setting those bits, you would see that they are indeed being set. 

```
Serial2.swap(1);            // PF4 TXD2 / PF5 RXD2
Serial2.begin(250000);                
Serial2.println("\nµC Reset #### #### ####");
  
showPORTARegister(Serial2);

pinMode(PIN_WIRE_SDA, INPUT_PULLUP);
pinMode(PIN_WIRE_SCL, INPUT_PULLUP);

showPORTARegister(Serial2);

```

pinMode sets both PULLUPEN bit in PORTx.PINnCTRL AND the bits in PORTx.OUT. This is correct and intended behavior in order to ensure backwards compatibility with code for classic AVRs which depends on pins set to input pullup and then flipped to output starting out HIGH. (such code does exist in the wild). 

In the case of the I2C, the Wire library, [as documented](https://github.com/SpenceKonde/DxCore/tree/master/megaavr/libraries/Wire) does NOT enable the pullups because the pullups are too weak to meet the I2C specificiations by a factor of around 3-5 (the pullup strength depends on operating voltage and process variation). Any system relying upon the internal pullups for I2C is wired incorrectly. While the internal pullups alone will function in some cases, it can lead to fiendishly difficult to debug issues - the sorts of issues which depend on things like the length of wires, number of connected devices, and potentially even really stupid stuff like whether the user is holding the circuit board in their hand vs letting it rest on the table. We provide Wire.usePullups() as an option which enables the pullups for users who understand this and wish to do so anyway to expedite prototyping; it is not the default option because then users could proceed without realizing that they are operating outside of the I2C specs and unprepared to recognize these problems if they occur. It is better for a configuration to fail consistently than to fail in such a perverse and unpredictable way.

Note that both Wire.begin() and Wire.usePullups() explicitly clear the relevant OUT registers in order to work around the silicon errata.

closed 2021-08-05 by SpenceKonde

mikrocoder, 2021-08-05:
Hello,

of course my code shows exactly the problem. Your pinMode(n, INPUT_PULLUP) sets the OUT register. You can see this quite clearly in my code shown with register output.

I don't understand what this should have to do with compatibility. Who configures the pullup in the old fashioned way, does it like this and this is simply wrong on the new µC. On the old controllers it didn't work differently anyway. But if you use the pinMode method correctly, you should not run into a trap, because you deliberately configure it wrong in the background. On the new controller you expect with pinMode(n, INPUT_PULLUP) that the pullup is set in the PINnCTRL register and does not take a fatal detour via OUT. At this point there is no compatibility thought. Because only therefore you must go with Wire a special way and correct that again. The question is why not immediately so and always exactly so. The I/O port can't be compared 1:1 anyway, because the new controllers are built differently.

And because of your compatibility. Arduino does this on the Nano Every also only with the PINnCTRL register and MCUdude likewise.

What I want to say is, your compatibility thought is wrong here. It does not need compatibility. I must say that unfortunately so clearly.

You must not be distracted by the Wire pins. This has only made the problem clear. This is how I first became aware of the error.
```cpp
// Arduino Nano Every (Arduino Framework & MCUdude Framework)
µC Reset #### #### ####
PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
PIN2CTRL:  0b0000'0000 0x0412
PIN3CTRL:  0b0000'0000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413
```
and

```cpp
// AVR128DB48 Nano Every (your Framework)
µC Reset #### #### ####
PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
PIN2CTRL:  0b0000'0000 0x0412
PIN3CTRL:  0b0000'0000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'1100 0x0404
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413
```
Testcode

```cpp
void setup (void)
{
  Serial.begin(250000);                
  Serial.println("\nµC Reset #### #### ####");
  showPORTARegister(Serial);
  
  PORTA.PIN3CTRL |= PORT_PULLUPEN_bm;
  PORTA.PIN2CTRL |= PORT_PULLUPEN_bm;

  showPORTARegister(Serial);
  PORTA.PIN3CTRL = 0;
  PORTA.PIN2CTRL = 0;
  
  pinMode(PIN_WIRE_SDA, INPUT_PULLUP);
  pinMode(PIN_WIRE_SCL, INPUT_PULLUP);

  showPORTARegister(Serial);
}
```

Translated with www.DeepL.com/Translator (free version)

SpenceKonde, 2021-08-05:
Tjhis is one of the big decisions that had to be be made - how colsely do we emulate the behavior of the classic AVR cores? 

Classic AVRx had 3 registers. DDRx, PORTx and PINx. The last one worked the same as PORTx.IN, except that writing to it toggles the value of PORTx , like writing to VPORTx.IN  does now (these registers were always in the low I/OP space allowing bit level access, except on the rare parts like the mega2560 that had too many I/O pins for that).
IF the pin's bit in DDRX was 1, it was an output, if it was 0 it was am input. of DDRx bit was 1. PORTx biot set oputput state: 1= high = 0 LOW. But if DDRx was 1 (inpui), while the bit in PORTx was 1, that;'s how you turned on the pul.up. This was the original sin of the Atmel designers. 

That mean s that on classic AVR 
pinMode(pin,INPUT);
digitalWrite(pin,HIGH); 
turned on the puillups. This is used extensivelty in existing code.  

Also that 
pinMode(pin,INPUT_PULLUP); 
pinMode(pIn,OUTPUT); 
resulted in the opin outputting a HIGH  in output mode

On a modern AVR, a naive implementations of pinMode and th. I decided that the most reasonable approach is that whenever the classic API is used, we should  emilate the weirdness where practical.

And when novel APIs are uesd we do not emulate the legacy ones (ex dgitalWritFast() called on input pins doesn't modify their pulliup setting.

mikrocoder, 2021-08-05:
Hello,

I don't understand your reasoning about some alleged compatibility constraint. Arduino.cc and MCUdude don't do that either. Only you make such nonsense.

If you still write such things, you can continue to do so, if you mistakenly think that it is correct on modern controllers.
`pinMode(pin,INPUT) and digitalWrite(pin,HIGH)` Only this was the code to switch on pullup on old controllers and `pinMode(n, INPUT_PULLUP)` does the same in the background. Yes.

But I want to teach you that exactly this is just wrong on new controllers. Who on new controllers can use
`pinMode(pin,INPUT) and digitalWrite(pin,HIGH)` thinks he switches on the pullup, I just feel sorry for him. Because the port pin behaves differently than old controllers.

The people who write `pinMode(n, INPUT_PULLUP)` just expect the pullup to be active and nothing more.
This is exactly what must happen on new controllers in the `PINnCTRL` register.

I can't see any constraint between old and new code. From this I cannot construct a problem which does not exist. There is a serious problem only if one holds on modern controllers wrongly like you on old braids. That would be a pity for all your work with your framework.

If you still don't want to understand that it is wrong. Then please explain to me what problem there should be if your `pinMode(n, INPUT_PULLUP)` only turns on the pullup in `PINnCTRL`? What specific problem would people have then? I don't see any problem. Besides, you are still turning on both. `OUT` and `PINnCTRL`. Which is now both together now really nonsense.

Translated with www.DeepL.com/Translator (free version)