SpenceKonde/DxCore #129: AVR-DU

MCUdude, 2021-08-24:
It doesn't seem like you're running out of work right away!

Introducing AVR-DU; with native USB!

https://ww1.microchip.com/downloads/en/DeviceDoc/AVR-DU-Product-Brief-DS40002328.pdf

SpenceKonde, 2021-08-24:
Did you manage to save it before it vanished? It's not there anymore. 

Thank fucking god we will finally have something better than micro! 

But I think it's going to be a long way off. We don't even have DD yet!

MCUdude, 2021-08-24:
The URL doesn't work for you?
Here's the PDF!

[AVR-DU-Product-Brief-DS40002328.pdf](https://github.com/SpenceKonde/DxCore/files/7041383/AVR-DU-Product-Brief-DS40002328.pdf)

SpenceKonde, 2021-08-24:
Nope, thanks! 

Interesting.... what's with their stepping back down a tier in flash and ram with newer parts? They did 128k for DA and DB, but now we're only seeing 64 for DD, DU, and EA....

SpenceKonde, 2021-08-25:
I'm getting a sinking feeling that the 14 and 20 pin AVR DD-series parts aren;'t going to be released in 64k versions :-(

closed 2021-08-25 by SpenceKonde