SpenceKonde/DxCore #130: Add missing clock overclocked speeds

SpenceKonde, 2021-08-25:
Speed options should be a superset of those available to megaTinyCore, so that means we need to pull in support for 25 and 30 back into the Dx-series, since somehow those weren't ever added as options (30 makes sense - it was added when it became clear that we couldn't turn 1-series parts up to 32 MHz and expect them to have a prayer of working without an external clock). I guess I didn't do 25 because it was so close to 24.

wa2mze, 2021-09-07:
While 25mhz isn't that different from 24mhz in performance, 25mhz crystals and oscillators are probably easier (and cheaper) to source!  I have a lifetime supply of 25mhz crystals in the junque box (if anyone needs one for testing I can mail you some).

SpenceKonde, 2021-10-01:
Easier to source from pulls, yes. not any cheaper new, though - both are common speeds, 25 probably a bit more common ("1/4th of 100 MHz"), but 24 being twice the base 12 MHz of USB 1 or half the 48 MHz base clock of USB 2.0 which is generally generated via pll from 12 or 24 - as well as being a multiple of halfway between two powers of 2, and thus a natural choice for parts that aren't rated for 32, but which need to be clocked higher than 16. 
Pricing is similar whether you get them in the West or from chinese knockoff sellers, too,

commit 837df06b (2021-10-04): Update platform and board defs

Fix #130., #148
closed 2021-10-04 by SpenceKonde