SpenceKonde/DxCore #134: Curiosity Nano programmer not working....

wa2mze, 2021-09-04:
I selected the Curiosity Nano as the programmer, I selected NO BOOTLOADER, and AVR128DA48 as the chip type.  Also 24mhz internal (no option listed for using the 32khz oscillator to calibrate the 24 mhz oscillator).   Selected TTYACM0 for serial port (which should NOT be used for programmer.  Selected for use with serial port (which isn't being used right now)   I created a simple led blink sketch:

void setup() {
  // put your setup code here, to run once:
  pinMode(20,OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(500);
  digitalWrite(20,1);
  delay(500);
  digitalWrite(20,0);
}

Sketch compiles OK.  Tried uploading and it fails.  See error messages (for AVRDUDE only) below.

Arduino: 1.8.13 (Linux), TD: 1.54, Board: "AVR DA-series (no bootloader), Disabled, AVR128DA48, 24 MHz internal, 1.9V, Disabled/Disabled, EEPROM retained, Hardware Reset (recommended), TCB2, 8ms"

Sketch uses 954 bytes (0%) of program storage space. Maximum is 131072 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 16380 bytes for local variables. Maximum is 16384 bytes.
/home/ken/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18/bin/avrdude -C/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.6/avrdude.conf -v -v -pavr128da48 -ccuriosity_updi -Pusb -Ufuse5:w:0b11001001:m -Ufuse7:w:0x00:m -Ufuse8:w:0x00:m -Uflash:w:/tmp/arduino_build_985540/AVR128DAnano.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.6/avrdude.conf"
         User configuration file is "/home/ken/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: jtag3_open_updi()
avrdude: usbdev_open(): Found nEDBG CMSIS-DAP, serno: MCHP3280031800001715
avrdude: usbdev_open(): WARNING: failed to set configuration 1: Device or resource busy
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
avrdude: usbdev_send(): wrote -5 out of 912 bytes, err = Input/output error
avrdude: jtag3_edbg_prepare(): failed to send command to serial port

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

SpenceKonde, 2021-09-06:
I don't have a linux system to test with but this behavior on linux, buyt all is well on windows.

wa2mze, 2021-09-06:
The curiosity Nano works fine under MPLAB under Linux, so there IS a Linux driver for the thing that works.  I did have a problem getting it recognized under MPLAB once that required resetting some of the project parameters though.  It's strange that Arduino is complaining about the port not being recognized, though the /dev/ACM0 does 'go away' while the programmer is being used (it's a USB device with multiple endpoints).  AVRdude DOES find the programmer endpoint, as reported in the error messages.

killerkalamari, 2021-09-11:
I needed to add my user to the dialout group on my Debian system for avrdude to be able access /dev/ttyUSB0 to program the AVR128DB28 using jtag2updi. I don't have a Curiosity Nano so this is a shot in the dark, but does `sudo adduser $USER dialout` and then rebooting (not just logging out), do the trick?

wa2mze, 2021-09-11:
I'm already in the dialout group, and a member of sudo (these are usually set by default in ubuntu and kubuntu but not Debian.)  So that's not the answer for me.  Also the Curiosity Nano does work under MPLAB native Linux.

pfoun, 2021-09-11:
I do not think just being a member of the dialout group will give you permission to use a serial usb device.  You need to create a rules file in the etc/udev/rules.d system folder.  The file can be named "98-anything.rules" and contain this line:  SUBSYSTEMS=="usb", ATTRS{idVendor}=="????", ATTRS{idProduct}=="????", MODE:="0666".  You will need elevated priviledges to do this.  This line containing the wild cards may be a breach of security, but it should match any usb device and has worked for me for /dev/ttyUSB0 as well as others.  I am not an expert in this area and direction from others is welcome.

killerkalamari, 2021-09-11:
Linux has a kernel module that can sniff USB traffic: https://www.kernel.org/doc/Documentation/usb/usbmon.txt
I tried it out on jtag2updi:

`sudo modprobe usbmon`

Then I took a look at `lsusb`:
`Bus 001 Device 010: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter`

To capture, I ran (`1u` comes from lsusb `Bus 001`...  just add a `u` to the bus): 

`sudo cat /sys/kernel/debug/usb/usbmon/1u >bus1.txt`

In my case, there were multiple devices on the bus, cluttering the output. I filtered the output to just the device I cared about (`010` comes from lsusb `Device 010`): 

`grep ":010:" bus1.txt >avrdude.txt`

Perhaps comparing captured USB outputs for MPLAB and avrdude will reveal something?

SpenceKonde, 2021-09-12:
I thought dialout normally gave you access to serial USB devices. But the programmer on the curiosity nanos is not just a serial port (the serial port isn't the thing that is used to program it; it's a composite device that shows up as several devices)
You can get the USB PID and VIDs for them from various programmers from avrdude,conf. Of course, we don't even know if it's a permissions issue or what.

leo60228, 2021-09-28:
pymcuprog works, but unlike the serial programmer it verifies the device ID, so the MCU always being `avr128da28`/`avr128db28` doesn't work.

leo60228, 2021-10-07:
This seems to be specific to mEDBG, or possibly the Dx series. I can program an Xplained Nano ATtiny416 with avrdude on the same system perfectly fine.

SpenceKonde, 2022-01-19:
Since this is a bug in avrdude, fixing it is out of my hands. If this problem still occurs with latest avrdude, raise the issue with the avrdude team

closed 2022-01-19 by SpenceKonde