SpenceKonde/DxCore #138: PIN_WIRE1_SDA undeclared for 28pin-standard

killerkalamari, 2021-09-15:
Compilation error:
```
In file included from [...]/cores/dxcore/Arduino.h:354:0,
                 from [...]/cores/dxcore/wiring_private.h:31,
                 from wiring_digital.c:26:
[...]/cores/dxcore/pinswap.h:74:29: error: 'PIN_WIRE1_SDA' undeclared here (not in a function); did you mean 'PIN_WIRE_SDA'?
 static const uint8_t SDA1 = PIN_WIRE1_SDA;
                             ^~~~~~~~~~~~~
                             PIN_WIRE_SDA
[...]/cores/dxcore/pinswap.h:75:29: error: 'PIN_WIRE1_SCL' undeclared here (not in a function); did you mean 'PIN_WIRE_SCL'?
 static const uint8_t SCL1 = PIN_WIRE1_SCL;
                             ^~~~~~~~~~~~~
                             PIN_WIRE_SCL
```

For a 32-pin part these are defined as:
```
// TWI 1
#define PIN_WIRE1_SDA           PIN_PF2
#define PIN_WIRE1_SCL           PIN_PF3
```

However, for a 28-pin part:
`// TWI 1 not available - no supported pins present!`

SpenceKonde, 2021-09-16:
Oh goddamnit. I accidentally committed pinswap before it was done.Yeah all pin swapping should be expected to be broken :-(

SpenceKonde, 2021-09-16:
Actually maybe it i considerably more complex than that, needs to be dealt with as part of that whole mess though, (which is mostly not yet in github)

SpenceKonde, 2021-10-04:
This and other issues should be fixed with the new variant files checked in over the weekend

closed 2021-10-04 by SpenceKonde