SpenceKonde/DxCore #146: First try compiling Blink with DxCore 1.3.6

mrWheel, 2021-09-30:
@SpenceKonde 

Due to the Wire / I2C problem I try to make the switch to an AVR128DB28.

First try:
```
In file included from /Users/me/Documents/ArduinoProjects/MySketches/Blink/Blink.ino:31:0:
/Users/me/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.6/libraries/DxCore/src/DxCore.h: In function 'bool setTCA0MuxByPort(uint8_t)':
/Users/me/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.6/libraries/DxCore/src/DxCore.h:88:24: error: 'digitalPortToPinZero' was not declared in this scope
     uint8_t base_pin = digitalPortToPinZero(port);
                        ^~~~~~~~~~~~~~~~~~~~
/Users/me/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.6/libraries/DxCore/src/DxCore.h:88:24: note: suggested alternative: 'digitalPinToTimer'
     uint8_t base_pin = digitalPortToPinZero(port);
                        ^~~~~~~~~~~~~~~~~~~~
                        digitalPinToTimer
/Users/me/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.6/libraries/DxCore/src/DxCore.h:90:60: error: expected ')' before ';' token
     for (byte i = base_pin; i < (min(max_pin, base_pin + 6); i++)) {
                                                            ^
/Users/me/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.6/libraries/DxCore/src/DxCore.h:90:66: error: expected primary-expression before ')' token
     for (byte i = base_pin; i < (min(max_pin, base_pin + 6); i++)) {
                                                                  ^
Using library DxCore at version 1.3.1 in folder: /Users/me/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.6/libraries/DxCore 
exit status 1
Error compiling for board AVR DB-series (no bootloader).
```
Is this expected behaviour?

mrWheel, 2021-09-30:
It must be me! Suddenly it compiles and PA7 blinks....

SpenceKonde, 2021-09-30:
DxCore.h is the problem....

SpenceKonde, 2021-09-30:
I named that macro something different in every damned file

mrWheel, 2021-09-30:
> I named that macro something different in every damned file

Ha ha!

But the funny thing is: I have not included `DxCore.h` (or is it includes by the core somehow?)

mrWheel, 2021-09-30:
> Weird.... I cannot explain why it would have been compiling that library

Well, you know me by now!

commit d363bcb2 (2021-10-01): Fix #146
closed 2021-10-01 by SpenceKonde