SpenceKonde/DxCore #150: How to configure USART for RS485 communication

mrWheel, 2021-10-02:
In the documentation I see this text

> On all supported devices, where the appropriate pins are present, they can be pin-swapped - each PORT except PORTD (AVR DD parts don't get one one PORT) gets a USART, which defaults to pins 0 and 1 for RX, TX (2 and 3 for XCK and **XDIR** - though these are not supported through the Serial class)

XDIR is used for RS485 communication. Is there a way to configure an USART for RS485 anyway?

SpenceKonde, 2021-10-03:
Yeah, you can enable RS485 and loopback mode, they show a diagram of that in the datasheet. . Sounds like you need to set USART_LBME and RS485 bits in CTRLA, and if you need open drain mode for that, ODME (make sure the pin is an input if ODME is set per errata). Note that unlike the tiny3216, they only have one RS485 option described, where it drives XDIR when it's transmitting. XDIR may need to be set output. If it's doing the opposite of what the transceiver expects, set the pin inverted. (the tinyAVR parts described a second mode, where that was an input, and it would... not output serial unless it was in the right state or something, and the data would just sit there, or something. I suspect it was flaky. Something like that has a bunch of corner cases that could go wrong, and is conceptually kind of tricky... It's not described in the 2-series tinyAVR datasheet either.

But making XDIR drive the transceiver is a standard thing to want to do. 

Serial doesn't put a wrapper around any of that, but I don't think it would be hard to make work, I'd set up with Serial.begin()  then USARTn.CTRLB &= 0x3F; to disable, USARTn.CTRLA |= 0x09; (set LBME for single wire and RS485 bits), then set USARTn.CTRLB |= 0xC0; to reenable, or 0xC8 with ODME too (for ODME you also need to set the pin to be an input, otherwise it's not open drain at all) then see what it's doing and if I might need to invert XDIR output or take other corrective actions.. You might or might not need to set XDIR as an output and you might not need to disable and reenable it, just change them,- I haven['t played with that part of the product. There's a good app note on the one-wire/open drain mode. 
You are expected to receive every character you send, and it provides a way of making sure that something wasn't also trying send something at the same time in a way that ended with both messages getting stomped on. 

There's a plan to add the one wire modes and xdir modes as part of a larger project to make some changes to serial- mostly invisible to the user; it's mostly about being able to support large numbers of portmux options for the DD). There's just no way that the current pin swapping method is extensible to the DD series with like, what 5 mux options on USART0? Currently the constructor takes at least 5 arguments per mux option; it's already onerous with just default and alt1.

SpenceKonde, 2021-10-08:
Was there any issue here or can this be closed?