SpenceKonde/DxCore #159: [question] What is the best way to install the master-branch?

mrWheel, 2021-10-18:
What is the best way to install the master-branch?

Can it be done with the board-manager?

At the moment I just manually copy changed files in the ../megaavr/ map of the installed latest-release but it’s a lot of work with the risk of missing a file.

SpenceKonde, 2021-10-18:
I only ever have two types of Arduino installations:

Portable installation, which I install board manager packages on

And main install which I never install AVR board packages on. 

On the portable installs, I do not use the github branch. 

On the main install, I sync the repo to (documents)/Arduino/hardware and can easily update it. 

The problem is that since nothing is getting installed through board manager, the toolchain doesn't get updated though, so when I have to do a toolchain update (1.3.5 was the last one that got such an update, Azduino4b, 1.3.7 and bugfix releases won't, but 1.4.0 will (well some version between the availability of the first DD-silicon (DD64) and smaller sizes) - when that happens you have to take the new toolchain package from the json; for latest version:
```json
        {
          "name": "avr-gcc",
          "version": "7.3.0-atmel3.6.1-azduino4b",
          "systems": [
            {
              "size": "41351111",
              "checksum": "SHA-256:a0b8a6327a4ce0b2aee3e559a71ea0b61ccecc48e358ec5fac50ba10710981a0",
              "host": "i686-mingw32",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4b-i686-w64-mingw32.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4b-i686-w64-mingw32.tar.bz2"
            },
            {
              "size": "37878659",
              "checksum": "SHA-256:23becaaf746e8cf3b23ebcd261bf52d5eebcc094c3500fe8d533f0911b870c60",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-aarch64-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "34220996",
              "checksum": "SHA-256:dacef5c90b38e6728e2cd161a54133b008352ff6f6513e67ab71f3117331c758",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-arm-linux-gnueabihf.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-arm-linux-gnueabihf.tar.bz2"
            },
            {
              "size": "36887243",
              "checksum": "SHA-256:a3f4833034b5526a7375b20fed7031e5fd0ff6956703e7b4bcf48e914fb94d6c",
              "host": "i686-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-i686-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "37530243",
              "checksum": "SHA-256:61c9ad88f9e75664fc6b96350dd4b3f95946e1c6eeca2158504aecc515f475a3",
              "host": "x86_64-apple-darwin14",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-apple-darwin14.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-apple-darwin14.tar.bz2"
            },
            {
              "size": "37412841",
              "checksum": "SHA-256:60d879e7052db01383cd89c54e46fa20c81ca180e20a8c0a681ffecec20c24d8",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-pc-linux-gnu.tar.bz2",
              "url": "https://github.com/SpenceKonde/DxCore/releases/download/1.3.4/avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-pc-linux-gnu.tar.bz2"
            }
```
extract it, and then copy it over the (non-board-manager Arduiino install directory/hardware/tools/avr, being sure to get it the right number of folders down, replacing several thousand files, 

Note that you can't do this if it's installed in program files on windows, windows interferes in wacky bizarro ways) - I always use the Arduino zip package,  I normally have 4-10 versions of Arduino 
```
Directory of C:\

07/18/2020  08:19 AM    <DIR>          Anaconda3
12/22/2020  12:32 PM    <DIR>          arduino-1.0.6
11/22/2020  01:41 PM    <DIR>          arduino-1.0.6-5.4.0compiler
11/22/2020  01:41 PM    <DIR>          arduino-1.0.6-7.3.0compiler
03/02/2021  09:02 PM    <DIR>          arduino-1.6.9-portable
07/14/2020  08:09 PM    <DIR>          arduino-1.8.13
07/15/2020  01:16 AM    <DIR>          arduino-1.8.13-portable
07/28/2020  02:17 PM    <DIR>          arduino-1.8.13-portable-skbk
03/07/2021  07:21 PM    <DIR>          arduino-1.8.13-portableclean
06/09/2021  02:45 PM    <DIR>          arduino-1.8.15
10/13/2021  01:38 AM    <DIR>          Data
08/03/2020  06:38 AM    <DIR>          EAGLE-7.7.0
10/17/2021  05:11 AM    <DIR>          Intel
09/20/2020  01:13 PM    <DIR>          PADSViewer Projects
10/13/2021  05:07 AM    <DIR>          Program Files
09/10/2021  11:12 PM    <DIR>          Program Files (x86)
05/08/2021  05:07 AM    <DIR>          SiliconLabs
07/14/2020  12:15 AM    <DIR>          Users
08/30/2021  07:13 PM    <DIR>          Windows
08/18/2020  01:43 PM    <DIR>          xampp
```
and worst case if I trash the main installation somehow, there is no state saved in there, it'[s all saved in the Documents/arduino (or the portable locations, orthe board manager install location) so I can just delete it andextract fresh copy of the IDE.

SpenceKonde, 2021-11-03:
I do all my development on Windows and have never had anything but woe and misery trying to run anything with a GUI (like the arduino IDE) on Linux and don't get along well with apple products in general (I had an awful time trying to do "export compiled binary" on a borrowed mac on an older version, I forget what the complication was, but I've never been able to do basic stuff on mac - at least on linux you've always got the command line and it's generally easy to find what the incantation you need to type should be, approximately).

The master branch needs to be cloned into (sketchbook)/hardware, such that you have (sketchbook)/hardware/DxCore and so on.  That location does not exist until you create it and cores located there will show (in sketchbook) after the core name (now - at one point the behavior was horrifyingly bad). 
I don't think it will use the toolchain version from the board manager installation; here;'s the original sin of the third party hardware package format:
The dependency list is in the json file. There is nothing like that associated with a manually installed hardware package. 
In some versions of the IDE, it would use the *most recent* installed version of the toolchain (including from board manager). I don't know what the current behavior is, but I know that at one point that's what it did, and the behavior wound up being bad enough that I changed my protocol to "never make a board manager install on a non-portable installation, only on portable installations". 

Hence why my main development instance (hence the default arduino15 location) never has any board packages installed. 

From the json file, you can search for the most recent version of the core that was released. It will list a version of avr-gcc. Search the json file for that version string, and you'll find an entry with a file for each OS, and a URL. That can be downloaded and extracted. It must then be copied over (arduino install location)/hardware/tools/avr - which will replace thousands of files - but it is very easy to get the wrong number of levels of depth, especially because some tools extract evertyrhing into a directory others don't and copying a toolchain next to the existing one but the wrong number of directories deep is hopeless to entangle because there are like 3 layers of directories all with the same names. 
I've trashed arduino insrtall like that that a few times. And I have made the analogous error almost every time I try to make a new toolchain package. 

And no, my toolchain package creation is NOT automated. There is an automated way to do it, but it's part of a script that also expects to be able to compile the compiler executables for the target platform, but I don;t know jack shit about how to set up that kind of build environment. I run it on a VM in the cloud on linux (a big fast instance that only stays up for the half hour or so it takes to build the toolchain) getting me a copy for x64 linux. IIRC the linker doesn't work. In any event, I don't use the compiler executables from that. I copy it to a local linux VM, along with the most recent arduino official toolchain packages for the 5 linux-like platforms (file permissions are lost if you extract and recompress on windows) and then have a text file full of commands and regex replacement strings so that (I think now it works) I copy a blob of commands into sublime text, and run a regex on it to fix version numbers and stuff, then copy paste to terminal. repeating in like a dozen steps with difrferent blocks of commands and regexes. And the final result is a set of toolchain packages, and I have a block of commands I can paste in that generates the sizes and checksums and file names, and a regex that will wrap them in brackets and all. It's a fucking shitshow, but I don't know how to do it the way you're supposed to, and this allows me to release cores that support new parts and keep doing to interesting things I know how to do (developing for AVR) instead of the tedious and completely uinteresting things I don't have a clue how to do (crosscompiling for systems with a thousand times the memory at which it becomes unpleasant to program them.). 

What that last paragraph should basically tell you is "I am completely unqualified a to make a bash script for anything" and that I'm probably less competent than you think. My competence is very narrow, because I cannot control what I focus on and get good at. I am good at making cores, low level programming, and designing hardware of certain types, but in other areas I am quite incompetent, sometimes spectacularly so. I barely manage to eat food and sleep on a vaguely human schedule. My life is falling apart at the moment, and all my plans are upended but I am unable to apply myself to fixing those things. But I can sure make an AVR do cool shit!

SpenceKonde, 2021-11-05:
I recognize that this method of installation is not pleasant. I do not have a solution to how to fix it which I know how to implement.

closed 2021-11-05 by SpenceKonde