SpenceKonde/DxCore #166: Serial Port may have wrong baud rate, ver 1.3.8

microPaul, 2021-11-01:
So I've been working on a project using the AVR128DB28 for about two months.  Everything was going great, and most of it still is going great.  But today I loaded in the most recent version of the DX core 1.3.8, and now I find that the serial port UART baud rate generator seems to be running fast.  I discovered this when both Serial Monitor and RealTerm print my power-on message as gibberish (it's been printing correctly on both for months).

Here's some test code to show the problem:

`
 void setup() {
  Serial.begin(1200, SERIAL_7N1); 
  for (int i = 0; i < 1000; i++) {
    // send a bunch of 0x7e ('~')to UART using 7N1, output should be:
    // start bit low, B0 low, B1-B6 high, stop bit high.  9 bits total.
    // at 1200 bps the interval from leading edge of one
    // start bit to the leading edge of the next start bit should
    // be 7.5 ms.  but when I check with a scope I see 1.356 ms
    // between leading edges of two start bits.
    Serial.printf("~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~");
  }
    
}

void loop() {
}`

So what I'd expect to see at the UART TX output when the UART is set to 1200 bps with a 7N1 format, is that the time between the leading edges of two start bits would be 7.5ms ( (1/1200) * 9).  But what I see with a scope is that the time between two start bits is 1.356 ms, which would mean that the baud rate was 6637 bps.  I tried going back to an older version of the DX core, to see if that would solve the problem, but it doesn't.  So, in addition to some problematic change to the baud rate generator, I wonder if 1.3.8 changed some fuse in the MCU that doesn't get corrected when I run with older versions of the Core.

I'm running with the internal 24 MHz oscillator.

SpenceKonde, 2021-11-02:
It is impossible to generate 1200 baud output from a 24 MHz system clock. There are not enough bits in the USARTn.BAUD register to hold the value that would be needed to slow it down by that much. The best it can do is off by 22%.  https://docs.google.com/spreadsheets/d/1uzU_HqWEpK-wQUo4Q7FBZGtHOo_eY4P_BAzmVbKCj6Y/edit?usp=sharing

microPaul, 2021-11-02:
Ugh.  Thank you.  I knew it had to be something stupid.

SpenceKonde, 2021-11-03:
I've clarified this in the documentation as well now.

closed 2021-11-03 by SpenceKonde