SpenceKonde/DxCore #167: AVR-DA: Unrecognized combination of flash size and chip type

aikopras, 2021-11-02:
I've just updated to V1.3.8 but run into the compile time error message "Unrecognized combination of flash size and chip type". I'm using a AVR128DA48.

To better understand the problem, I tried the blink example and experimented with different chip settings. Whatever combination of DA cores I try  (AVR128..., AVR32..., ...DA48, ...DA32), I consistently get the same compile error. However, if I compile using a AVRDB chip, I don't have any error.

I tried using the Arduino IDE 1.8.16 as well as 2.0.0-beta.7. Same problem.
I also tried a fresh install of DxCore on another Mac. Same problem.
I assume it is a small problem, but a bit irritating ;-)

I'll attach the error message I received
[Error-Message.txt](https://github.com/SpenceKonde/DxCore/files/7460350/Error-Message.txt)
.

SpenceKonde, 2021-11-02:
Yrd I'm aware of this and will be releasing a critical bugfix for it today. This is why we desperately need CI like we have for megaTiinyCore

closed 2021-11-02 by SpenceKonde

aikopras, 2021-11-06:
Took me a while to realise that the tar file via the Arduino board manager still downloaded an old version, but a "direct download" from github solved the problem. I'm also happy to see a rewritten (and faster) attachInterrupt(). I'll play with that next. And the different gcc options decreased code size with roughly 10%. All very nice!
Thanks for your great work!!

hoka83, 2024-04-24:
> I've just updated to V1.3.8 but run into the compile time error message "Unrecognized combination of flash size and chip type". I'm using a AVR128DA48.
>
> To better understand the problem, I tried the [...]

Hello can you please , clarify how did you fix this error , i have the same error as well