SpenceKonde/DxCore #170: Does variables keep there value after sleep [Question]

mrWheel, 2021-11-04:
It seems my AVR128DB48 looses the values stored in variables.

Is that "normal behaviour"?

Is there an easy way to let them keep there value (other than storing in EEPROM).

SpenceKonde, 2021-11-04:
Nope  not normal. I reckon your board isn;t sleeping, and is instead
resetting. Do you have the WDT enabled by any chance or something? Or
brownout detection tripping? Or voltage falling below 1.6v while it's
sleeping?

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

mrWheel, 2021-11-04:
Your right.
It is even worse..
I have a controller board sleeping and when it wakes-up it powers the communication board! That’s where I lose the values but that’s of-course expected as it doesn’t sleep but is completely switched of!

so stupid

thanks for your reply

closed 2021-11-04 by mrWheel