SpenceKonde/DxCore #171: [Question] 24Mhz Crystal not working, at least I couldn't make it happen

aliihsanhavan, 2021-11-04:
I am working on building my own hardware and use external 24Mhz Crystal. I couldn't make it work properly. 24Mhz Internal works well, no issue there. If I change it to 24 Mhz Crystal, code doesn't run. By the way code is just a simple blink sketch to check. Nothing complicated. 

Strange behaivor that if I change setting to 28 Mhz (Overclocked), then blink sketch seems to work fine. Here is my design an parts used.

Can you please help me identify what might be the issue? Has anyone faced this kind of an issue before? 

PS: I am pretty newbie to electronics, maybe it is a very very silly one. :)

[image]
[image]

Crystal: https://www.ozdisan.com/pasif-komponentler/kristaller-osilatorler-ve-rezanatorler/kristaller/X1E0000210065
Capacitors: https://www.ozdisan.com/pasif-komponentler/kondansatorler/smt-smd-ve-mlcc-kondansatorler/CL21C150JBANNNC

SpenceKonde, 2021-11-04:
You're using a 7pf crystal with 15 pf caps? That would be your problem. You
want loading caps to be equal to crystal caps, less 4 pf to account for pin
capacitance and 3 pf approx for stray capacitance, give or take. Try
removing the loading caps entirely, (I have found that this has fixed
problems of that sort me where I dont have full specs for xtal) and get a
higher capacitance crystal next time. Also use a ground pour, both sides of
the board.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every pf microcontroller
Contact: ***@***.***

aliihsanhavan, 2021-11-04:
Wow! That's the quickest reply I have ever had in my entire work life :) Thanks.

I tried to replicate the same desing in Curiosity Nano Board, but definetly there is a calculation mistake on my end. 

[image]

closed 2021-11-04 by aliihsanhavan

SpenceKonde, 2021-11-04:
Just luck on the fast response, and a question that I could quickly answer.
I've never found success the spec'ed crystal capacitance < loading cap
value, and the formula that I found suggested that the capacitors should be
more like Ccrystal - Cpin - Cstray, or 4.5- 7pf below the crystal
capacitance. I don't know where those example calculations came from; they
don't match my understanding of how those calculations should be done, and
the values from the datasheet don't match the datasheet (the AVR128DB
datasheet doesn't spec max load capacitance). That said though, I can't say
I've ever really felt like I understand the crystal formulas, frankly. I
always end up using rules of thumb and empirical testing to verify it, and
as long as the two caps have similar values, I haven't seen
noticeable inaccuracies in frequency due to the wrong values of caps, just
a greater sensitivity to insults like touching the crystal. What I have
always observed is that it either works and is as accurate as I can
measure, or it doesn't work at all.

Are you saying that you had success with 28 MHz crystal option, but not the
24 MHz crystal option? If so, I think I know what's going on there - going
over 24 MHz will switch to the maximum crystal drive strength option; I
have also observed a similar phenomenon on one of my boards, which used a
40 MHz crystal (It's for a lighting controller, and, WS2812B update speed
taking a fixed time per pixel, I wanted to be able to make the most of the
limited time that it's not outputting that signal for calculating the next
frame - since I didn't need super-accurate timing in general, I could just
use internal at 24 or 32 MHz if I found I didn't need that speed. As it
happens It's so far irrelevant because I don't seem to be able to wire up
power supply boards without burning shit out lately, so the fairy light
splitters that I counted on to achieve those optimistic numbers of LEDs are
0 for 3 right now - 1 failed to program, 1 generated 0v output voltage, and
the third applied 20V to the tiny412 I'm using as a splitter, making a loud
pop and thoroughly failing - it all worked fine in the prototype!), and it
wouldn't oscillate at 24 MHz crystal option (a mistaken selection) but I
then found that it would in fact work fine at 40 when a higher speed was
selected. I'm starting to think I should change the tests from F_CPU <= (8,
16, or 24 MHz) to F_CPU < (8, 16, or 24 MHz).

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

aliihsanhavan, 2021-11-05:
Removing caps work fine as you suggested. Thanks a lot for your advice.

Regarding the second paragraph, It was exactly like you described. It wasn't working with 24Mhz while caps are connected but when I select 28Mhz Overclocked, everything worked fine. I am sure that there is definitely a technical answer for this (as you explained). :)

Thanks again for your help and additional thanks for this great Core.