SpenceKonde/DxCore #173: EXTERNAL_NUM_INTERRUPTS in spi.h

sshwarts, 2021-11-05:
Spence:
This is a weird one. With 1.3.8, the compiler complains about EXTERNAL_NUM_INTERRUPTS not being defined in spi.h. (line 293) In 1.3.6 this is not reported as an issue.

Bear in mind I'm using Visual Studio Code.

SpenceKonde, 2021-11-05:
Oh not wierd at all, that would be breakage from attachInterrupt changes.

SpenceKonde, 2021-11-05:
But I have no fucking clue what the hell it is messing with the pin interrupts in SPI.h for! I never have understood that part of the code or what it hoped to achieve, but now it needs to be confronted because it is incompatible with the new attachInterrupt implementation. Like really, what the bloody hell business does SPI.h have messing with pin interrupts?!

SpenceKonde, 2021-11-05:
Anyone know why SPI.h has to go turning off interrupts at all??? This seems like it just makes things messier and worse....

sshwarts, 2021-11-05:
Two reasons (maybe). if you didn't have hardware SPI but also if the interrupt handler tries to do an SPI operation.

SpenceKonde, 2021-11-05:
Ohhhhh if both the interrupt handler and interrupt tried to use SPI so the
ISR was trying to begin a transaction when one was already begun.

@MX682X - wasn't there an analogous situation in TWI? Do we handle it
there? It seems like the way those beginTransaction functions in both are
written they'd have the same problem, but it only showed signs of ever
being handled in SPI case

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

MX682X, 2021-11-06:
Sorry, I can't follow what you asked exactly about the interrupts. 
First of all, TWI is Half-Duplex, it can only either transmit or receive, and this is done with a while loop and interrupts respectively. 
SPI, on the other hand, is Full-Duplex and can transmit and receive bytes simultaniously, this is done with a write, wait for complete and read method. Since the SPI library does not support Slave mode, nor Buffered read/writes (would be fun to do one though, with an extern SPI_Transfer_Finished callback function) it will never need interrupts. All transfers are initiated from user code and all transfers are blocking. (If I didn't miss anything)

closed 2021-11-06 by SpenceKonde