SpenceKonde/DxCore #174: Bootloader/No-Bootloader Fuse issues?

microPaul, 2021-11-06:
I will freely admit that this may be an operator error, but I'm working on a AVR128DB64 board of my own design.  I had a blink sketch working which I programmed via UPDI using a serial adapter and a 4.7K/Schottky diode.  That all worked fine.   Although I'm no longer a Bootloader user I thought I should test out a bootloader, since this is a new board, and to ensure that the DTR reset circuit was working.  I thought I had followed the procedure to burn in the boot loader, saw diagnostic messages related to setting fuses, etc.  But I couldn't get serial download via bootloader to work, I kept getting an error message about an array problem (I know, I should have saved that message, but I didn't, and now my problem seems to have worsened and I can't get back there).  So I thought, Ok, no problem, I'll go back to UPDI download without bootloader.  But now I can't get the blink sketch to program via UPDI directly (without bootloader).  

Here's the error I get when I try to compile and download a blink sketch using UPDI, no bootloader:

`C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\cores\dxcore\UART0.cpp:58:56: error: expected primary-expression before ',' token
   UartClass Serial0(HWSERIAL0, HWSERIAL0_DRE_VECTOR_NUM, PIN_HWSERIAL0_RX, PIN_HWSERIAL0_TX, HWSERIAL0_MUX, PIN_HWSERIAL0_RX_PINSWAP_1, PIN_HWSERIAL0_TX_PINSWAP_1, HWSERIAL0_MUX_PINSWAP_1);
                                                        ^
exit status 1
Error compiling for board AVR DB-series (no bootloader).
`
I don't have code like that in my blink sketch so I'm guess that that code is something from the DX core, but I don't know for sure.

Any advice greatly appreciated.

microPaul, 2021-11-06:
I tried burning the bootloader again, and here's the error message I get with that, in case that information is helpful:

`Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port COM8 at 57600 baud.
Target: avr128db64
Set fuses: ['0:0x00', '1:0x00', '5:0b11001000', '6:0b00001100', '7:0x00', '8:0x01']
Action: write
File: C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8/bootloaders/hex/optiboot_dx128_ser0_alt.hex
Pinging device...
Ping response: 1E970B
Setting fuse 0x0=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x1=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x5=0xc8
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x6=0xc
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x7=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x8=0x1
Writing literal values...
Verifying literal values...
Action took 0.16s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.06s
Writing from hex file...
Writing flash...
Writing flash...
Traceback (most recent call last):
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8/tools/prog.py", line 266, in <module>
    main()
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8/tools/prog.py", line 121, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8/tools/prog.py", line 240, in pymcuprog_basic
    blocksize=args.blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8/tools/prog.py", line 131, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\pymcuprog_main.py", line 341, in _action_write
    _write_memory_segments(backend, result, args.verify, blocksize=args.blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\pymcuprog_main.py", line 392, in _write_memory_segments
    backend.write_memory(segment.data, memory_name, segment.offset, blocksize=blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\backend.py", line 526, in write_memory
    self.programmer.write_memory(data=data, memory_name=memory_name, offset=offset_byte, blocksize=blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\programmer.py", line 190, in write_memory
    self.device_model.write(memory, offset, data, blocksize=blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\nvmserialupdi.py", line 175, in write
    self.avr.nvm.write_flash(offset_aligned, chunk, blocksize=-1, bulkwrite = bulk)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\serialupdi\nvm.py", line 248, in write_flash
    return self.write_nvm(address, data, use_word_access=True, blocksize=blocksize, bulkwrite=bulkwrite)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\serialupdi\nvm.py", line 313, in write_nvm
    if not self.wait_flash_ready():
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\serialupdi\nvm.py", line 58, in wait_flash_ready
    status = self.readwrite.read_byte(self.device.nvmctrl_address + constants.UPDI_NVMCTRL_STATUS)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\serialupdi\readwrite.py", line 56, in read_byte
    return self.datalink.ld(address)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\tools\libs\pymcuprog\serialupdi\link.py", line 349, in ld
    return self.updi_phy.receive(1)[0]
IndexError: bytearray index out of range
Error while burning bootloader.
`

SpenceKonde, 2021-11-06:
I cannot reproduce the bootloader write issue, that error just means it got no reply from the board. What kind of adapter is it?

microPaul, 2021-11-06:
For UPDI programming, In my original tests, I was using an FTDI (likely a knockoff) with 4.7K resistor - Schottky diode.   But after you asked about the adapter I also just now tried a SNAP, and I get the same results:

`C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\cores\dxcore\UART0.cpp:58:56: error: expected primary-expression before ',' token
   UartClass Serial0(HWSERIAL0, HWSERIAL0_DRE_VECTOR_NUM, PIN_HWSERIAL0_RX, PIN_HWSERIAL0_TX, HWSERIAL0_MUX, PIN_HWSERIAL0_RX_PINSWAP_1, PIN_HWSERIAL0_TX_PINSWAP_1, HWSERIAL0_MUX_PINSWAP_1);
                                                        ^
exit status 1
Error compiling for board AVR DB-series (no bootloader).
`

I do have a question about going back and forth between a bootloader and plain old UPDI programming, because maybe I'm not doing the correct procedure.  If my device (AVR128DB64) at one time had a bootloader (or a botched bootloader based on error message I get when trying to burn the bootloader) and I now want to go back to UPDI programming, to get rid of the bootloader is the correct procedure to select BOARD>DX-CORE>AVR DB Series (No Bootloader) and then simply do an UPLOAD operation (the right pointing button located below EDIT on the Arduino IDE window that starts the compile process followed by the program process)?  Is that the process I should use to blow away the bootloader and just use UPDI?  Or is the process to remove the bootloader to select BOARD>DX-CORE>AVR DB Series (No Bootloader) and then click Burn Bootloader to remove the bootloader (since I selected BOARD>DX-CORE>AVR DB Series (No Bootloader)?

microPaul, 2021-11-06:
Here's something interesting, that may be helpful.

This is a new simplified blink sketch I'm using.

`
void setup(void) {
  pinMode(PIN_PA7, OUTPUT);  // init PA7 at output for LED
}

void loop(void) {
  while (1) {
    digitalWrite(PIN_PA7, LOW); 
    delay(200); 
    digitalWrite(PIN_PA7, HIGH);
    delay(1000);
  }  
}
`

If I set up the Arduino IDE for BOARD>DX-CORE>AVR DB Series (No Bootloader) and the device for AVR128DB28 (the 28 pin part) and VERITY (check mark icon with no upload) with no programmer connected (and no target connected), the result I get is:

`Sketch uses 936 bytes (0%) of program storage space. Maximum is 131072 bytes.
Global variables use 4 bytes (0%) of dynamic memory, leaving 16380 bytes for local variables. Maximum is 16384 bytes.`

which is what I would expect.

But if I then change the target device to be AVR128DB64, again with no programmer and no target attached, but still using that same simple sketch, and click VERITY (check mark icon) I get:

`
C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.8\cores\dxcore\UART0.cpp:58:56: error: expected primary-expression before ',' token
   UartClass Serial0(HWSERIAL0, HWSERIAL0_DRE_VECTOR_NUM, PIN_HWSERIAL0_RX, PIN_HWSERIAL0_TX, HWSERIAL0_MUX, PIN_HWSERIAL0_RX_PINSWAP_1, PIN_HWSERIAL0_TX_PINSWAP_1, HWSERIAL0_MUX_PINSWAP_1);
                                                        ^
exit status 1
Error compiling for board AVR DB-series (no bootloader).
`

Is it possible that my Arduino Installation or DX Core installation is corrupt?  Maybe I should remove both and reinstall?

SpenceKonde, 2021-11-06:
Yeah that issue is fixed and will be in 1.3.9, which I'm to fix variant files because they aren't passing linter.

microPaul, 2021-11-06:
ok, thanks.  when 1.3.9 is out I will retest the bootloader thing on my DB64.

SpenceKonde, 2021-11-08:
Can you confirm whether this is still misbehaving?

microPaul, 2021-11-08:
Thanks to your new DX core Ver 1.3.9, everything is peachy from my perspective.  

On my DB64 board my console serial port is UART4-primary.  via UPDI and a home brew USB-serial 4.7K/Schottky interface I was able to burn a bootloader with UART4-primary as the download port, and then confirmed that I was able to serial download a blink sketch over that serial port.  I then selected the (no bootloader) option and burned a modified blink sketch (different LED timing) over UPDI and confirmed that that sketch worked as expected as well.  So, from my limited testing, the bootload/download problems that I was having with 1.3.8 on a DB64  appear to be resolved.  Thank you for fixing this so quickly!

closed 2021-11-08 by microPaul

SpenceKonde, 2021-11-08:
Wonderful! 

Thanks for confirming.