SpenceKonde/DxCore #178: Compiler warning flags

sshwarts, 2021-11-07:
I like compiler warnings. I noticed though that the compiler warning flags are now set to:
compiler.warning_flags=-Wall
compiler.warning_flags.none=-Wall
compiler.warning_flags.default=-Wall
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

Just a question, is the norm to have them set to:
compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

So that the Arduino IDE users can easily turn on warning as they see fit? As you have in your AtTiny platform?
Just asking...

SpenceKonde, 2021-11-07:
I intentionally force warnings on. They almost always indicate a problem in the program and the default settings of the IDE lead the majority of Arduino users to run with warnings disabled. And of course this is embedded so we can't spare the overhead for exceptions. So relatively inexperienced programmers are trying to debug problems in the absence of ANY feedback on what went wrong. 

For fucks sake, avr-gcc calls it a warning when you misspell the name of an ISR, which will result in the device crashing as soon as it would execute that! And 90% of Arduino users don't even realize that they can or should enable warnings, so the argument that users could enable or disable it is specious - the users who benefit most from warnings are least likely to know to enable them. If this was an opt-out type of option, I wouldn't feel a need to change this (not that there's ever a correct time to disable warnings), since only reasonably competent users would know how to disable them. But the default is "none", so the inexperienced users were all running with warnings turned off. This isso widespread that I've run into *libraries* that people were recommending... which throw a bunch of warnings because of sloppy coding practices without the author even knowing. Because they run with warnings disabled because that's the bloody default! 
This is part of the misguided ideology behind many aspects of the Arduino IDE and core libraries, where error conditions are hidden from the user so that their code will compile, even it's bloody obvious at compile time that it's not going to work. But don't tell them or show them where the suspicious parts of their code are or anything. I don't understand it. How does that help anyone?!  Wherever we can determine that something the user has done is wrong, we should let them know, not hide it and let them wander onwards until the consequences of that error finally result in undesired behavior.

In megaTinyCore and DxCore, in every place that I can catch a compile-time-known function argument that we know won't work for some API function with no return value through which an error can be given (which is basically all of them, because Arduino doesn't seem to believe in that), I've been adding checks for this which are optimized away for valid inputs and compile errors for compile-time known invalid inputs. Examples of this are things like digitalWrite(NUM_TOTAL_PINS _(or other thing that isn't a pin number - NUM_TOTAL_PINS is the lowest positive number that isn't a pin number)_ ,HIGH (unfortunately not the NOT_A_PIN value - there's existing code that relies upon passing NOT_A_PIN to the digital I/O functions silently doing nothing) or analogReference(1234) and so on. 

This change was made in megaTinyCore and will be made in the next release of ATTinyCore. This is forced on me by the perverse defaults of the IDE. I'm not going to contribute to people we all see floundering on forums with problems which warnings would flag, and complaining about issues that are bugs in their code - like, what am I doing putting in specific tests to catch common errors if people aren't even running with warnings? But they don't turn off warnings to make life harder for themselves, they do it _simply because it's the default_

closed 2021-11-07 by SpenceKonde

sshwarts, 2021-11-08:
I agree completely with you.  Warnings are your friend and it's especially annoying though when you get them from Arduino supplied libraries!

What I was wondering was for those who use the Arduino IDE (I don't), if they can change the warning levels if in platforms it's all turned on. I presume they can...