SpenceKonde/DxCore #180: VREF_REFSEL_VDD_gc undefined in VS Code

dpetican, 2021-11-08:
It still compiles properly though.  I think this affects all the reference macros.  I get a similar problem with things like `ADC0.SAMPCTRL = 128 `for instance

[image]

SpenceKonde, 2021-11-08:
At the preprocessor level it is an undefined constant because preprocessor can't see names from enums. But it diesn't need to get resolved at that time soit;'sfine.  

When it comes time to actually compile them, it can see the enums so everything works, that name is defined in the microchip-supplied io header files.

And I have no idea what you are talking about with "a similar problem with" what you claim, - I don[t understand how a similar problem could possibly come about with ADC0.SAMPCTRL = 128. 

Are you compiling this on some bizzaro non-standard tool that doesn't have access to the IO headers?

SpenceKonde, 2021-11-08:
(and yes, i know that many of those show as azduino4a - only windows version of 4a was bad.

dpetican, 2021-11-08:
The problem is as you said that the io.h or some file is not being read when VS Code is linting (correct term?) but at compile time everything resolves okay.  Without being able to resolve the define, autocompletion does not work and it appears there are errors where none actually exists.  To be fair I've had similar issues with standard cores like the Mega2560 so its likely a problem in VS Code's Intellisense feature or the Arduino extension.

Including the file` avr/ioavr128db28.h` makes the errors go away in the editor but the compile fails with this error:

```
c:\users\admin\appdata\local\arduino15\packages\dxcore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr\ioavr128db28.h:38:4: error: #error "Attempt to include more than one <avr/ioXXX.h> file."
 #  error "Attempt to include more than one <avr/ioXXX.h> file."
```
So that got me thinking about VS Code's Arduino extension.  When you initialize a project it creates a file called c_cpp_properties.json  This is the relevant section:

```
            "name": "Arduino",
            "compilerPath": "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\bin\\avr-g++",
            "compilerArgs": [
                "-w",
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\cores\\dxcore\\api\\deprecated",
                "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\cores\\dxcore",
                "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\variants\\28pin-standard",
                "C:\\Users\\admin\\Documents\\Arduino\\libraries\\BlockNot\\src",
                "C:\\Users\\admin\\Documents\\Arduino\\libraries\\ArduinoJson\\src",
                "C:\\Users\\admin\\Documents\\Arduino\\libraries\\Adafruit_MCP23017_Arduino_Library",
                "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\libraries\\Wire\\src",
                "c:\\users\\admin\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\lib\\gcc\\avr\\7.3.0\\include",
                "c:\\users\\admin\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\lib\\gcc\\avr\\7.3.0\\include-fixed",
                "c:\\users\\admin\\appdata\\local\\arduino15\\packages\\dxcore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b\\avr\\include"
            ],
            "forcedInclude": [
                "C:\\Users\\admin\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\cores\\dxcore\\Arduino.h"
            ],

```

But I can't force include `ioavr128db28.h` because the `c_cpp_properties.json` file gets rebuilt every time I compile.  I used to be able to hand edit this file and have the changes stick in earlier versions, but alas that appears to not be the case anymore.  Please leave this issue open.  When I have time I will raise this with the VS Code Arduino extension team.  Before I can be more helpful with this I need to learn how VS Code knows what information to put in the json file when initializing a project.

SpenceKonde, 2021-11-09:
Thething is that the Arduino.h file itself should be core_devices.h which should be pulling in avr/io.h, which in turn should be o pulling in the right io header  and it sounds luike it does for compiling, but isnt fully resolving them for linting >.< this seemsto be a VS code issue that I can't be particularly helpful The IO headers were meant to be pulled in by including io.h and then from there it sorts out what file to include.\

closed 2021-11-09 by SpenceKonde