SpenceKonde/DxCore #181: Compile error in Wire.h

dpetican, 2021-11-08:
Error message:

```
C:\Users\admin\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.9\libraries\Wire\src\Wire.cpp:258:9: error: 'else' without a previous 'if'
       } else
```

Unfortunately I have to downgrade to 1.6 for now to continue working.

SpenceKonde, 2021-11-08:
Ugh. That section of code hasn't been touched since prior to release of 1.3.7. WTF part are you using that triggers it

SpenceKonde, 2021-11-08:
OH! This is a bug that cannot manifest for any supported parts,. You are cvompilinh in soime bizzare environment that doen't have access to io.h somehow.

SpenceKonde, 2021-11-08:
Will be fixed in next release, though there are not any current or released products which will cause that conditionally compiled branch of code to be used.

closed 2021-11-08 by SpenceKonde

dpetican, 2021-11-08:
I am using the AVR128DB28 part.  Interesting that you mention io.h again  Argh this stuff is hard.  I appreciate the work you are doing on the core.  My project wouldn't be possible without it.  I'm hoping once I get a little further along I will have some extra bandwidth to actively help out with testing.

SpenceKonde, 2021-11-11:
My previous statement was inaccurate, it could have reproduced on existing parts. And my "fix" broke swap badly. 1.3.10 is coming out tonight to address these critical bugs before we switch to the new version of Wire