SpenceKonde/DxCore #182: HardwareSerial library support

nabelekt, 2021-11-08:
I am trying to compile with the RadioHead library (the Adafruit version) for a DA MCU. Compilation fails with:
```
"C:\\Users\\nabel\\OneDrive\\Documents\\ArduinoData\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=8000000L -DARDUINO=10816 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 "-IC:\\Users\\nabel\\OneDrive\\Documents\\ArduinoData\\packages\\DxCore\\hardware\\megaavr\\1.3.9\\cores\\dxcore/api/deprecated" "-IC:\\Users\\nabel\\OneDrive\\Documents\\ArduinoData\\packages\\DxCore\\hardware\\megaavr\\1.3.9\\cores\\dxcore" "-IC:\\Users\\nabel\\OneDrive\\Documents\\ArduinoData\\packages\\DxCore\\hardware\\megaavr\\1.3.9\\variants\\28pin-standard" "-IC:\\Users\\nabel\\OneDrive\\Documents\\ArduinoData\\packages\\DxCore\\hardware\\megaavr\\1.3.9\\libraries\\SPI\\src" "-IC:\\Users\\nabel\\OneDrive\\Documents\\Arduino\\libraries\\RadioHead" "C:\\Users\\nabel\\OneDrive\\Documents\\Arduino\\libraries\\RadioHead\\RH_Serial.cpp" -o nul
Alternatives for HardwareSerial.h: []
ResolveLibrary(HardwareSerial.h)s\Arduino\libraries\RadioHead\RH_Serial.cpp:11:11: fatal error: HardwareSerial.h: No such file or directory

  -> candidates: []eSerial.h>

           ^~~~~~~~~~~~~~~~~~
compilation terminated.
```
It looks like HardwareSerial hasn't been implemented for DxCore? Any particular reason (other than time/bandwidth/etc.) for the lack of support?

Thanks!

SpenceKonde, 2021-11-08:
Huh? Hardware serial is in UART.h and I didn't think radiohead has ever
made any effort to support modern avr devices- which are all like this in
the naming of the files

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

nabelekt, 2021-11-08:
Hmm, gotcha. Yeah, looks like the fix isn't as simple as including UART.h in place of HardwareSerial.h. Guess I need to look for a different library. Any suggestions for a library supporting LoRa radios and friendly to Dx MCUs?

closed 2021-11-08 by nabelekt

nabelekt, 2021-11-08:
It looks like examples in the [library from Seeed Studio](https://github.com/Seeed-Studio/Grove_LoRa_433MHz_and_915MHz_RF) at least compile with DxCore boards selected.