SpenceKonde/DxCore #184: OneWire (not to be confused with 1Wire)

sshwarts, 2021-11-10:
This is a report void of details, and I apologize but I just wanted to document it. I tried to get the common OneWire lib (Jim Studt) working with the AVRDB32 to read from a DS18B20 temp sensor and could not see any data being exchanged on the logic analyzer. I saw the reset but that's it. I ended up rolling my own with the help of my partner Kevin Darrah, and it works. 

I can't tell you where in OneWire it's failing bit if I had to guess, it's most likely the DIRECT_WRITE_LOW, DIRECT_READ etc macros. Might have to just fork it for a DxCore version or ask them to add definitions for the platform?

SpenceKonde, 2021-11-10:
Did you see https://github.com/SpenceKonde/megaTinyCore/blob/master/megaavr/extras/LibraryCompatibility.md  (it should be linked from somewhere in the documentation, I forget where that link was) - the tinyAVR and Dx-series are close enough that a single list is sufficient and the overhead of maintaining multiple lists was too much. I forked it, added support, and then created a PR that hasn't gotten a response in over a year.

 Jim Studt? I thought it was Paul Stoffregen who did OneWire.h? See link from above linked page.

sshwarts, 2021-11-10:
I did not see! I had searched for it, I appreciate it. Jim Studt was the original author. Paul Stoffregen has been the maintainer for quite some time.

closed 2021-11-10 by sshwarts

SpenceKonde, 2021-11-10:
Well he ain't exactly doing much maintenance.....