SpenceKonde/DxCore #186: Unable to program AVR128DB64 with DxCore 1.3.9 and SerialUPDI programmer

dbuchwald, 2021-11-12:
I was able to compile/upload simple Blink sketch to my DIP AVR128DB28 and it works fine. However, once I moved to AVR128DB64, I'm not able to upload the sketch correctly. Bootloader seems to have been written OK, but when trying to upload the sketch I get:

```
Arduino: 1.8.13 (Mac OS X), Board: "AVR DB-series (no bootloader), AVR128DB64, 24 MHz internal, 1.9V, Disabled/Disabled, EEPROM retained, Hardware Reset (recommended), TCB2 (recommended), Default, 1.5k flash, no floats, On all pins, like usual, Disabled, 8ms, Disabled"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dawid/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dawid/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dawid/Documents/Arduino/libraries -fqbn=DxCore:megaavr:avrdb:chip=128DB64,clock=24internal,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,printf=default,attach=allenabled,mvio=disabled,startuptime=8,appspm=no -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563 -warnings=none -build-cache /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_cache_72544 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.python3.path=/Users/dawid/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/dawid/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -verbose /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_modified_sketch_514845/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/dawid/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/dawid/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/dawid/Documents/Arduino/libraries -fqbn=DxCore:megaavr:avrdb:chip=128DB64,clock=24internal,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,printf=default,attach=allenabled,mvio=disabled,startuptime=8,appspm=no -vid-pid=0000_0000 -ide-version=10813 -build-path /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563 -warnings=none -build-cache /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_cache_72544 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.python3.path=/Users/dawid/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/dawid/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Users/dawid/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -verbose /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_modified_sketch_514845/Blink.ino
Using board 'avrdb' from platform in folder: /Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9
Using core 'dxcore' from platform in folder: /Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9
Detecting libraries used...
/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr128db64 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore/api/deprecated -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/variants/64pin-standard /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563/sketch/Blink.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr128db64 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore/api/deprecated -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/variants/64pin-standard /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563/sketch/Blink.ino.cpp -o /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=avr128db64 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DMILLIS_USE_TIMERB2 -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 -DCORE_ATTACH_ALL -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore/api/deprecated -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore -I/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/variants/64pin-standard /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563/sketch/Blink.ino.cpp -o /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_build_479563/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /var/folders/ng/m4ppdfjj1wqcrqgcbnn5j2_w0000gn/T/arduino_cache_72544/core/core_788c587183c50d575b5d2dd4ed1768f1.a
Linking everything together...
/Users/dawid/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-gcc -Wall -Os -g -flto -fuse-linker-plugin -mrelax -Wl,--gc-sections,--section-start=.text=0x0,--section-start=.FLMAP_SECTION1=0x8000,--section-start=.FLMAP_SECTION2=0x1 [... cut ...]
    return self.readwrite.read_data_words(address, words)
  File "/Users/dawid/Library/Arduino15/packages/DxCore/hardware/megaavr/1.3.9/tools/libs/pymcuprog/serialupdi/readwrite.py", line 102, in read_data_words
    return self.datalink.ld16(self,address)
TypeError: ld16() takes 2 positional arguments but 3 were given
TypeError: ld16() takes 2 positional arguments but 3 were given

[=================================                 ] 2/3
```
As it turns out, it probably requires fix in `readwrite.py`, will submit PR soon.

SpenceKonde, 2021-11-12:
Huh! I'd been trying to figure out that one for a while. 

Pity you didn;'t get that to me like... 2 hours earlier so it could have been in 1.3.10. 

Worth noting that this error only impacts verify - if you power-cycled or reset-button-reset the part, it would work without it.

See, I don't actually *know* python, I see code and fiddle with it to make it do my bidding, but I don't really understand why some things have a self argument, and others don't or anything like that.

SpenceKonde, 2021-11-12:
Thanks, will be in next release

closed 2021-11-12 by SpenceKonde