SpenceKonde/DxCore #188: Wire does not compile for DX core (

wa2mze, 2021-11-14:
I tried building the i2c_scanner example using the DX core.
Got the following error:

_**/tmp/arduino_build_154902/libraries/Wire/Wire.cpp.o
/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src/Wire.cpp: In member function 'bool TwoWire::swap(uint8_t)':
/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src/Wire.cpp:258:9: error: 'else' without a previous 'if'
       } else
         ^~~~
Using library Wire at version 1.1.2 in folder: /home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire 
exit status 1
Error compiling for board AVR DA-series (no bootloader).
Arduino: 1.8.13 (Linux), TD: 1.54, Board: "AVR DA-series (no bootloader), AVR128DA28, 24 MHz internal, 1.9V, Disabled/Disabled, EEPROM retained, Hardware Reset (recommended), TCB2 (recommended), Default, 1.5k flash, no floats, On all pins, like usual, 8ms, Disabled"**_

I examined the code, error is in this section of wire.cpp.  Works for atmega4809, same error for avr128da series.  I may try doing a diff between the two libraries (DXcore and megacoreX).

    #if (defined(PIN_WIRE_SDA) && defined(PIN_WIRE_SCL))
      if (state == 1) {
        // Use pin swap
        PORTMUX.TWIROUTEA = (PORTMUX.TWIROUTEA & 0xFC) | 0x01;
        return true;
      } else {
        // Use default configuration
        PORTMUX.TWIROUTEA = (PORTMUX.TWIROUTEA & 0xFC);
        // return false if we did that because the state they asked for didn't exist
        return  (state == 0);
      } else
    #else
      {
        // Assume default configuration
        PORTMUX.TWIROUTEA = (PORTMUX.TWIROUTEA & 0xFC | 0x02);
        return false;
      }
    #endif

Full compiler output below:
/home/ken/Arduino/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/ken/Arduino/arduino-1.8.13/hardware -hardware /home/ken/.arduino15/packages -hardware /home/ken/Arduino/Sketches/hardware -tools /home/ken/Arduino/arduino-1.8.13/tools-builder -tools /home/ken/Arduino/arduino-1.8.13/hardware/tools/avr -tools /home/ken/.arduino15/packages -built-in-libraries /home/ken/Arduino/arduino-1.8.13/libraries -libraries /home/ken/Arduino/Sketches/libraries -fqbn=DxCore:megaavr:avrda:chip=128DA28,clock=24internal,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,printf=default,attach=allenabled,startuptime=8,appspm=no -vid-pid=0000_0000 -ide-version=10813 -build-path /tmp/arduino_build_154902 -warnings=none -build-cache /tmp/arduino_cache_769563 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=/home/ken/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/ken/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=/home/ken/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/ken/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avrdude.path=/home/ken/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=/home/ken/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -verbose /home/ken/Arduino/Sketches/I2C_Scanner/I2C_Scanner.ino
/home/ken/Arduino/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/ken/Arduino/arduino-1.8.13/hardware -hardware /home/ken/.arduino15/packages -hardware /home/ken/Arduino/Sketches/hardware -tools /home/ken/Arduino/arduino-1.8.13/tools-builder -tools /home/ken/Arduino/arduino-1.8.13/hardware/tools/avr -tools /home/ken/.arduino15/packages -built-in-libraries /home/ken/Arduino/arduino-1.8.13/libraries -libraries /home/ken/Arduino/Sketches/libraries -fqbn=DxCore:megaavr:avrda:chip=128DA28,clock=24internal,bodvoltage=1v9,bodmode=disabled,eesave=enable,resetpin=reset,millis=tcb2,printf=default,attach=allenabled,startuptime=8,appspm=no -vid-pid=0000_0000 -ide-version=10813 -build-path /tmp/arduino_build_154902 -warnings=none -build-cache /tmp/arduino_cache_769563 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=/home/ken/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/home/ken/.arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1 -prefs=runtime.tools.arduinoOTA.path=/home/ken/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/ken/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avr-gcc.path=/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avrdude.path=/home/ken/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=/home/ken/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino18 -verbose /home/ken/Arduino/Sketches/I2C_Scanner/I2C_Scanner.ino
Using board 'avrda' from platform in folder: /home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9
Using core 'dxcore' from platform in folder: /home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9
Detecting libraries used...
/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore/api/deprecated -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/variants/28pin-standard /tmp/arduino_build_154902/sketch/I2C_Scanner.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Wire.h: [Wire@1.1.2]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.1.2]
/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore/api/deprecated -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/variants/28pin-standard -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src /tmp/arduino_build_154902/sketch/I2C_Scanner.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr128da28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.9\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=9UL -DDXCORE_RELEASED=1 -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore/api/deprecated -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/cores/dxcore -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/variants/28pin-standard -I/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src /home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src/Wire.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/home/ken/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/avr-g++ -c -g -Os -Wall -std=c++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narr [... cut ...]
/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src/Wire.cpp: In member function 'bool TwoWire::swap(uint8_t)':
/home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire/src/Wire.cpp:258:9: error: 'else' without a previous 'if'
       } else
         ^~~~
Using library Wire at version 1.1.2 in folder: /home/ken/.arduino15/packages/DxCore/hardware/megaavr/1.3.9/libraries/Wire 
exit status 1
Error compiling for board AVR DA-series (no bootloader).
[I2C_Scanner.zip](https://github.com/SpenceKonde/DxCore/files/7534103/I2C_Scanner.zip)

SpenceKonde, 2021-11-14:
This is fixed in the 1.3.10 release - in fact it's one of the two bugs that prompted it.

closed 2021-11-14 by SpenceKonde

wa2mze, 2021-11-14:
Modified line 258 in swap() in wire.cpp  as:
      }
//      else  //remove this and look elsewhere if problems with other parts in this core

Looks like an extra 'else' was cut and pasted here....

wa2mze, 2021-11-14:
Opps we seem to have crossed posting here....
Is 1.3.10 available via board manager now?

SpenceKonde, 2021-11-15:
Sure should be, its listed there http://drazzy.com/package_drazzy.com_index.json