SpenceKonde/DxCore #189: typeof() not defined 1.3.10

Di-Ny, 2021-11-15:
I upgraded from 1.3.6 to 1.3.10 via the board manager. Since then, the compiler returns errors regarding Arduino functions. Is there something changed with the toolchain ? I am not sure if it is a bug or I am missing something.

It is tested under Windows, Arduino IDE or VisualStudio, the same, AVR128DB48. The min function in Common.h does not compile due to the typeof() function that is not found.
Test code:
`void setup() {
	min(12,14);
}`
`void loop() { }`

Compiler error:
 `void setup()':
C:\Users\username\AppData\Local\arduino15\packages\DxCore\hardware\megaavr\1.3.10\cores\dxcore/api/Common.h:34:5: error: 'typeof' was not declared in this scope
   ({typeof (a) _a = (a); \
     ^
C:\Users\ndauy\source\repos\Test\Test.ino:13:2: note: in expansion of macro 'min'
  min(12,14);
`

It compiles fine under version 1.3.6, or when I choose an ATMega328. Any idea where does it come from ?

SpenceKonde, 2021-11-15:
Oh for fucks sake what the hell happened? 

Its is reproducing here. HOW CAN A BUILTIN KEYWORD NOT BE DEFINED?!!!?!

SpenceKonde, 2021-11-15:
Fucking hell, it's the C++17 stuff I think
It looks like we're in hair-across-ass mode where it doesn't avr-gcc standards without double-underlines on both sides of them
That is *not* intended behavior, and I think it only impacts cpp files not c files, or I would have stumbled over it

SpenceKonde, 2021-11-15:
Unfotunately doing an emergency release to fix it, as of like 5-6 hours ago is impractical.

SpenceKonde, 2021-11-15:
Fixed in Repo via c434ad9

SpenceKonde, 2021-11-16:
Nope, this is still busted - it just breaks differently now...

SpenceKonde, 2021-11-16:
Dude, it doesn't work on 1.3.6 either!

SpenceKonde, 2021-11-16:
```
void loop();
void setup() {
  float arg1 = 1.1;
  float arg2 = 1.6;
  uint8_t test min(arg1,arg2);
};
```
```

C:\arduino-1.8.13-portable\arduino-builder -dump-prefs -logger=machine -hardware C:\arduino-1.8.13-portable\hardware -hardware C:\arduino-1.8.13-portable\portable\packages -hardware C:\arduino-1.8.13-portable\portable\sketchbook\hardware -tools C:\arduino-1.8.13-portable\tools-builder -tools C:\arduino-1.8.13-portable\hardware\tools\avr -tools C:\arduino-1.8.13-portable\portable\packages -built-in-libraries C:\arduino-1.8.13-portable\libraries -libraries C:\arduino-1.8.13-portable\portable\sketchbook\libraries -fqbn=DxCore:megaavr:avrdbopti:chip=128DB28,clock=24internal,bodvoltage=1v9,bodmode=disabled,resetpin=reset,millis=tcb2,startuptime=8,timeout=1sec,bootloaderusart=ser0_alt -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_336445 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.python3.path=C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.avr-gcc.path=C:\arduino-1.8.13-portable\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=C:\arduino-1.8.13-portable\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avrdude.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\avrdude\6.3.0-arduino18 -verbose C:\arduino-1.8.13-portable\portable\sketchbook\sketch_nov16a\sketch_nov16a.ino
C:\arduino-1.8.13-portable\arduino-builder -compile -logger=machine -hardware C:\arduino-1.8.13-portable\hardware -hardware C:\arduino-1.8.13-portable\portable\packages -hardware C:\arduino-1.8.13-portable\portable\sketchbook\hardware -tools C:\arduino-1.8.13-portable\tools-builder -tools C:\arduino-1.8.13-portable\hardware\tools\avr -tools C:\arduino-1.8.13-portable\portable\packages -built-in-libraries C:\arduino-1.8.13-portable\libraries -libraries C:\arduino-1.8.13-portable\portable\sketchbook\libraries -fqbn=DxCore:megaavr:avrdbopti:chip=128DB28,clock=24internal,bodvoltage=1v9,bodmode=disabled,resetpin=reset,millis=tcb2,startuptime=8,timeout=1sec,bootloaderusart=ser0_alt -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_336445 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.python3.path=C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\tools\python3\3.7.2-post1 -prefs=runtime.tools.avr-gcc.path=C:\arduino-1.8.13-portable\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-azduino4b.path=C:\arduino-1.8.13-portable\portable\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b -prefs=runtime.tools.avrdude.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\avrdude\6.3.0-arduino18 -prefs=runtime.tools.avrdude-6.3.0-arduino18.path=C:\arduino-1.8.13-portable\portable\packages\arduino\tools\avrdude\6.3.0-arduino18 -verbose C:\arduino-1.8.13-portable\portable\sketchbook\sketch_nov16a\sketch_nov16a.ino
Using board 'avrdbopti' from platform in folder: C:\arduino-1.8.13-portable\portable\packages\DxCore\hardware\megaavr\1.3.6
Using core 'dxcore' from platform in folder: C:\arduino-1.8.13-portable\portable\packages\DxCore\hardware\megaavr\1.3.6
Detecting libraries used...
"C:\\arduino-1.8.13-portable\\portable\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=avr128db28 -DF_CPU=24000000L -DARDUINO=10813 -DARDUINO_avrdb -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.3.6\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=3UL -DDXCORE_PATCH=6UL -DDXCORE_RELEASED=1 -DMILLIS_USE_TIMERB2 -DCLOCK_SOURCE=0 -DSERIAL_PORT_BOOT=Serial -DSERIAL_PORT_BOOT_PORTMUX=1 -DUSING_OPTIBOOT "-IC:\\arduino-1.8.13-portable\\portable\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\cores\\dxcore/api/deprecated" "-IC:\\arduino-1.8.13-portable\\portable\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\cores\\dxcore" "-IC:\\arduino-1.8.13-portable\\portable\\packages\\DxCore\\hardware\\megaavr\\1.3.6\\variants\\28pin-standard" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_336445\\sketch\\sketch_nov16a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\arduino-1.8.13-portable\\portable\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sect [... cut ...]
C:\arduino-1.8.13-portable\portable\packages\DxCore\hardware\megaavr\1.3.6\cores\dxcore/api/Common.h:36:25: error: expected unqualified-id before ')' token
     _a < _b ? _a : _b; })
                         ^
C:\arduino-1.8.13-portable\portable\sketchbook\sketch_nov16a\sketch_nov16a.ino:5:16: note: in expansion of macro 'min'
   uint8_t test min(arg1,arg2);
                ^~~
sketch_nov16a:6:1: error: expected declaration before '}' token
 };
 ^
exit status 1
expected declaration before '}' token
```

closed 2021-11-16 by SpenceKonde