SpenceKonde/DxCore #190: assembly listing generation has stopped demangling names of core provided functions or including source

SpenceKonde, 2021-11-16:
Since 1.3.7, this seems to have been lost.... Unfortunately that was a huge bunch of changes. But I also know I was doing stuff late in the development of 1.3.7 that I couldn't have done without being able to generate usable assembly listings. And **it works on megatinycore** which is using nearly identical settings. This can't have been broken for long

C names are not being demangled, 
No source code is being included. 

It's as if it doesn't know about the core files, This always "just worked" before, and i have no idea how to go about solving it - but I do know that I won;'t be able to do the things I have planned for the core without the ability to get a useful assembly listing.

Marked as critical because it compromises my ability to maintain the core.

Di-Ny, 2021-11-16:
It won't solve the issue but maybe it will help locating the it. When I peek the definition of "RTC_PERIOD_CYC32768_gc", which is not known, it sends me to the file "ioavr128dd32.h" (I am using the avr128db48). So I included the absolute path for right file in the .ino

`#include "C:\Users\ndauy\AppData\Local\arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b\avr\include\avr\ioavr128db48.h"`

And I comment thoses lines in the file (l.35-39) : 
`//#ifndef _AVR_IOXXX_H_`
`//#  define _AVR_IOXXX_H_ "ioavr128db48.h"`
`//#else`
`//#  error "Attempt to include more than one <avr/ioXXX.h> file."`
`//#endif`

Now "RTC_PERIOD_CYC32768_gc" is known and autocompletion is back.

SpenceKonde, 2021-11-16:
I'm not sure how you did that, or what gives you those definitions. Do you see the same thing as I do? 

Arduini export compiled binary generates an assembly listing with mangled names for functions like analogWrite() and no comments for those functions but comment for other stuff? (and also that analogWrite is horrifyingly large? That was when I realized that it was broken - I was alarmed by how large it was - I knew it was somewhat bloated but did;t realize it had gotten that bad,.

SpenceKonde, 2021-11-20:
Only happens if the board is optiboot, and only effects the core itself. Libraries come through ,even some core files seem to, but not all....

See attached, sketch was this:
```
void setup() {
  // put your setup code here, to run once
}

void loop() {
  // put your main code here, to run repeatedly:
  analogWrite(GPIOR0,GPIOR1);
}
```
the double GPIORn is just a quick way to force the compiler to include the full function rather than optimizing for known pin. 

[sketch_nov20a.ino.avr128db28opti.24cs0.mB2.txt](https://github.com/SpenceKonde/DxCore/files/7575127/sketch_nov20a.ino.avr128db28opti.24cs0.mB2.txt)
[sketch_nov20a.ino.avr128da28.24cs0.mB2.txt](https://github.com/SpenceKonde/DxCore/files/7575128/sketch_nov20a.ino.avr128da28.24cs0.mB2.txt)

Removing critical since this is not quite as pressing anymore

I would still like to fix this

MX682X, 2021-11-21:
Interesting, the optiboot version has no demangling between the vectors and main. It can't be dependent on the files aswell, because the relevant files are being demangled below.
Also, I noticed that the Optiboot version is not using the TCB2 OVF interrupt vector 30. For whatever reason.

Arduino also generates .lst files. Can you upload them too? They usually have some demangling too.

SpenceKonde, 2021-11-24:
I don't know what you mean by that? Those two files attached are the .lst files - github doesn't let you attach .lst files to an issue, hence renaming them to .txt 
And vector differences (31 vs 30) are because one looks like it was a DB export and the other a DA (which doesn't make a difference, I had a pile if them accumulated from trying to understand this.

SpenceKonde, 2021-12-05:
more data - I've seen small changed impact what gets demangled in the listing :-(

SpenceKonde, 2021-12-26:
Okay, I'm left to conclude that this is a compiler (well objdump demangling) issue when the code isn't starting at 0x0000.

closed 2021-12-26 by SpenceKonde