SpenceKonde/DxCore #191: 1.3.10 Bootloader Issues

microPaul, 2021-11-16:
I'm using my same home made AVR128DB64 board with a likely knockoff FTDI USB<->Serial with 4.7K/Schotty on the UPDI.  When I use DXcore 1.3.10 to burn a bootloader with UART4-primary as the console (I accepted the defaults on all other options) I get the following error messages

`Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port COM8 at 345600 baud.
Target: avr128db64
Set fuses: ['0:0x00', '1:0x00', '5:0b11001000', '6:0b00001100', '7:0x00', '8:0x01']
Action: write
File: C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/bootloaders/hex/optiboot_dx128_ser4.hex
Pinging device...
Ping response: 1E970B
Setting fuse 0x0=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x1=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x5=0xc8
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x6=0xc
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x7=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x8=0x1
Writing literal values...
Verifying literal values...
Action took 0.16s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.06s
Writing from hex file...
Writing flash...
Writing flash...
Traceback (most recent call last):
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 266, in <module>
    main()
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 121, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 240, in pymcuprog_basic
    blocksize=args.blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 131, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\pymcuprog_main.py", line 341, in _action_write
    _write_memory_segments(backend, result, args.verify, blocksize=args.blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\pymcuprog_main.py", line 392, in _write_memory_segments
    backend.write_memory(segment.data, memory_name, segment.offset, blocksize=blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\backend.py", line 526, in write_memory
    self.programmer.write_memory(data=data, memory_name=memory_name, offset=offset_byte, blocksize=blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\programmer.py", line 190, in write_memory
    self.device_model.write(memory, offset, data, blocksize=blocksize)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\nvmserialupdi.py", line 175, in write
    self.avr.nvm.write_flash(offset_aligned, chunk, blocksize=-1, bulkwrite = bulk)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\nvm.py", line 248, in write_flash
    return self.write_nvm(address, data, use_word_access=True, blocksize=blocksize, bulkwrite=bulkwrite)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\nvm.py", line 313, in write_nvm
    if not self.wait_flash_ready():
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\nvm.py", line 58, in wait_flash_ready
    status = self.readwrite.read_byte(self.device.nvmctrl_address + constants.UPDI_NVMCTRL_STATUS)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\readwrite.py", line 56, in read_byte
    return self.datalink.ld(address)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\link.py", line 349, in ld
    return self.updi_phy.receive(1)[0]
IndexError: bytearray index out of range
Error while burning bootloader.
`

I have no idea what these error messages mean, but I hope they are helpful.

SpenceKonde, 2021-11-16:
It errored out because the programming attempt failed. 

My guess is that it's because you're running it in Turbo (345600 baud) instead of the normal speed. Does it work if you back off to 230400? 

I have had problems with turbo with some FT232's and adapters - the 3.3v logic levels don't always recover to a logic high from the perspective of the AVR before it drives it low again, and if you miss a whole byte within page write, it's out of sync at the end of the write when the command to reenable the response signature comes through, so it's got the response signature disabled so when it tries to do the next chunk, it gets no response, and no response results in a traceback error instead of the correct errors because pyserial returns something that gets an index out of range error when you ask for index zero, and I don't know python thus don't know how I would go about handling the error more gracefully

microPaul, 2021-11-16:
Ok, thanks for this info.  I'll try it at slower speeds later this afternoon or evening and report back what I encounter.  I'll also try a different USB-Serial-4.7K-Schottky programmer.   Thanks for the quick response.

SpenceKonde, 2021-11-16:
You say 4.7k/schottky - what does that mean? Where is the resistor here?

microPaul, 2021-11-16:
The diagram below shows how I've configured the 4.7K resistor and the Schottky diode (1N5819).  The LEDs have been removed so there's no loading current due to LEDs.  Also R3 in the diagram is shorted.  The diagram shows a CP2102 but for the module I was using earlier today the USB bridge chip was a FTDI (very likely a knockoff FTDI).  Both the FTDI module and the target are 5 volt systems.   Later today I'll try again using various speeds with a CH340C as the bridge chip.

I've checked the voltage levels with a scope in the past for this resistor-schottky configuration and they look well within limits, which doesn't mean that something couldn't be wrong with my current setup.  I'll check voltage levels again when I rerun these tests later today.

[image]

SpenceKonde, 2021-11-17:
5V FTDI? Really? I've always seen those with 3.3v logic levels, - power at 5v - but the logic is only 3v3...I checked and the ones I use look to have had 10k pullups to5v  from Rx/UPDI,, while the CH340;'s that I added a resistor to  have the 10k resistor to the Tx pin. 

Where did you hear that the TTL logic levels were used for UPDI? I didn't think they were... At least, I never saw anything written to indicate that it was any different from another I/O pin for default configuration...

microPaul, 2021-11-17:
Thanks for the reminder re FTDI and 3 vs 5 volts.  I'll have to look at that again.  I'm sidetracked at the moment trying to resolve a problem with my application code (unrelated to bootloader issues).  I will come back to the bootloader issue and let you know what I find, but it will be a day or so.  sorry for the delay.

microPaul, 2021-11-18:
Yeah, it just won't burn the bootloader at any speed using the FTDI knockoff, maybe because the target is 5V and the FTDI is 3V, although the signals on the UPDI pin look good, clean transitions between 5V and 0V, no weird voltage offsets on the UPDI pin.  The odd thing is that the FTDI knock off will download sketches over UPDI to the target, but won't program a bootloader over UPDI.

When I use my homemade UPDI programmer based on the CH340C it will burn the bootloader at top speed (345600), no problems, no glitches.  I didn't test at slower speeds.  The CH340C is configured for 5V and the target is 5V.   And that seems to work perfectly.

Ugh.  I need to stay away from those FTDI thingys when the target is 5V (and probably should avoid the FTDI knockoffs all together).  The CH340C seems very reliable.  I just wish that the windows drivers for it would respect XON/XOFF.

closed 2021-11-18 by microPaul

SpenceKonde, 2021-11-18:
Huh - that's really weird that it doesn't work for bootloader but does work for uploads, I wonder why that works, but the bootloader doesn't. Do you get the same failure mode if you tryto upload a super-tiny sketch with it? Try something like Bare Minimum with millis disabled; it needs to compile to no more than 512b to take the same codepath.

microPaul, 2021-11-18:
So the FTDI knockoff with a 5V target does do a direct UPDI sketch download even for small sketches.  Here's my small sketch:

`
void setup() {
  PORTA.DIRSET = PIN7_bm;  // LED DDR set for output
  PORTA.OUTSET = PIN7_bm;  // LED on  
}

void loop() {
}`

I did the direct download of this sketch, with mills disabled, as you recommended.  Here's the response I get from that process:

`Sketch uses 482 bytes (0%) of program storage space. Maximum is 131072 bytes.
Global variables use 0 bytes (0%) of dynamic memory, leaving 16384 bytes for local variables. Maximum is 16384 bytes.
Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port COM8 at 345600 baud.
Target: avr128db64
Set fuses: ['5:0b11001001', '7:0x00', '8:0x00']
Action: write
File: C:\Users\p\AppData\Local\Temp\arduino_build_298135/small-sketch-211118-01.ino.hex
Pinging device...
Ping response: 1E970B
Setting fuse 0x5=0xc9
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x7=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x8=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.06s
Writing from hex file...
Writing flash...
Action took 0.14s
Verifying...
Verify successful. Data in flash matches data in specified hex-file
Action took 0.06s
`

I got no error message and the LED on my target is active.  I can then change the sketch to make LED output port to LOW, recompile and download and the LED goes off.  So that would confirm that the sketch is functioning correctly.

I'd be happy to "dump the flash to a hex file" after a failed boot loader and post it here, but I don't know how to find the hex file.  I've done some searching but I'm not seeing any recently generated hex files in the places that I've thought to look.  Can you give me some pointers as to where I should find that hex file?

thanks,