SpenceKonde/DxCore #199: printf implementation is buggy??

mrWheel, 2021-11-26:
Hi,

Given the next function:
```
//=============================================================================
void printElapsedTime(const char *message, uint32_t startMilliSecs)
{
  uint32_t eSecs, milliSecs;
  uint32_t elapsTime = millis() -startMilliSecs;

  Monitor.printf("\r\n[1]elapsTime[%d], ", elapsTime);
  eSecs     = (int)(elapsTime / 1000);
  Monitor.printf("eSecs[%d], ", eSecs);
  milliSecs = (int)(elapsTime - (eSecs*1000));
  Monitor.printf("milliSecs[%d]\r\n", milliSecs);

  Monitor.printf("\r\n[2]elapsTime[%d], eSecs[%d], milliSecs[%d]\r\n", elapsTime, eSecs, milliSecs);
  
  Monitor.printf("\r\n[3]Time Elapsed to %s took %d.%d seconds\r\n\n", message, eSecs, milliSecs);
  Monitor.flush();
  
} //  printElapsedTime()
```

The **expected** output for the call:
```
uint32_t startTime = millis();
   .
   .
   .

printElapsedTime("Testing", startTime);
```
.. would be:
```
[1]elapsTime[9602], eSecs[9], milliSecs[602]

[2]elapsTime[9602], eSecs[9], milliSecs[602]

[3]Time Elapsed to Testing took 9.602 seconds
```
But the output **is**:
```
[1]elapsTime[9602], eSecs[9], milliSecs[602]

[2]elapsTime[9602], eSecs[0], milliSecs[9]

[3]Time Elapsed to Testing took 9.0 seconds
```
As you can see the lines [2] and [3] are messed up!

Tested with DxCore 1.3.8 and 1.3.10 and compiler option `printf(): "Full, 2.6k, prints floats"`

SpenceKonde, 2021-11-26:
Does it reproduce with the default printf configuration, (not using floats of course)?

mrWheel, 2021-11-26:
> Does it reproduce with the default printf configuration, (not using floats of course)?

Yes, it does :-(

> It looks like it manifested when printing mutltiple variables on onw line.

My observation also. I changed the code to:
```
  Serial.printf("\r\nTime Elapsed to %s took ", message);
  Serial.printf("%d.", eSecs);
  Serial.printf("%d seconds\r\n\n", milliSecs);
```
and that works just fine..

SpenceKonde, 2021-11-26:
Huh.  Well, well, I stole it from MCUDude. so I opened an issue over there.

MCUdude, 2021-11-26:
I could actually reproduce this issue of MiniCore as well.

It's a simple fix though; use the correct formatter for the datatype:

| Formatter | Datatype       |                              |
|-----------|--------------|------------------|
| `%d`        | signed 16-bit | `int` or `int16_t` |
| `%u`        | unsigned 16-bit | `unsigned int` or `uint16_t` |
| `%ld`        | signed 32-bit | `long` or `int32_t` |
| `%lu`        | unsigned 32-bit | `unsigned long` or `uint32_t` |

so in @mrWheel's case, the correct printf statement should look like this:

``` c++
void printElapsedTime(const char *message, uint32_t startMilliSecs)
{
  uint32_t eSecs, milliSecs;
  uint32_t elapsTime = millis() -startMilliSecs;

  Monitor.printf("\r\n[1]elapsTime[%lu], ", elapsTime);
  eSecs     = (int)(elapsTime / 1000);
  Monitor.printf("eSecs[%lu], ", eSecs);
  milliSecs = (int)(elapsTime - (eSecs*1000));
  Monitor.printf("milliSecs[%lu]\r\n", milliSecs);

  Monitor.printf("\r\n[2]elapsTime[%lu], eSecs[%lu], milliSecs[%lu]\r\n", elapsTime, eSecs, milliSecs);
  
  Monitor.printf("\r\n[3]Time Elapsed to %s took %lu.%lu seconds\r\n\n", message, eSecs, milliSecs);
  Monitor.flush();
  
} //  printElapsedTime()
```

mrWheel, 2021-11-26:
@MCUdude,
Hm.. ok! Thanks for elaborating!

Strange that `printf()` on other systems do not have that limmitation.
But I’m glad it is solved.

SpenceKonde, 2021-11-29:
Likely other systems have more room for guard rails around printf and/or you just don't see it because ints are 32 bits on most platforms, so you have to get rather large values to manifest this, or things explicitly declared as smaller datatypes

Documentation was amended to mention this, though.

closed 2021-11-29 by SpenceKonde