SpenceKonde/DxCore #200: SerialUPDI programming bug: ld16() takes 2 positional arguments but 3 were given

nabelekt, 2021-11-26:
Trying to program with "SerialUPDI with 4.7k... (230400 baud)", I'm getting:
```
...
Erased.
Action took 0.01s
Writing from hex file...
Writing flash...
[==================================================] 90/90
Action took 2.73s
Verifying...
[================================================  ] 87/90Traceback (most recent call last):
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 266, in <module>
    main()
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 121, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 246, in pymcuprog_basic
    filename=args.filename)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10/tools/prog.py", line 131, in run_pymcu_action
    status = func(backend, *args, args_pymcu)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\pymcuprog_main.py", line 267, in _action_verify
    verify_status = verify_flash_from_hex(args.filename, backend)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\hexfileutils.py", line 120, in verify_flash_from_hex
    verify_status = backend.verify_memory(segment_data, 'flash', segments[i][0])
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\backend.py", line 545, in verify_memory
    return self.programmer.verify_memory(data=data, memory_name=memory_name, offset=offset_byte)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\programmer.py", line 208, in verify_memory
    data_verify = self.read_memory(memory_name, offset, len(data))[0].data
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\programmer.py", line 290, in read_memory
    data = self.device_model.read(meminfo, offset_adjusted, numbytes_adjusted)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\nvmserialupdi.py", line 212, in read
    data += self.avr.read_data_words(offset, read_chunk_size>> 1)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\application.py", line 147, in read_data_words
    return self.readwrite.read_data_words(address, words)
  File "C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.10\tools\libs\pymcuprog\serialupdi\readwrite.py", line 102, in read_data_words
    return self.datalink.ld16(self,address)
TypeError: ld16() takes 2 positional arguments but 3 were given
[================================================= ] 89/90 does not exist or your board is not connected
```
I believe line 102 of readwrite.py should be changed from 
`            return self.datalink.ld16(self,address)`
to
`            return self.datalink.ld16(address)`

SpenceKonde, 2021-11-27:
Check source in github I think that was fixed, I got a pr to fix an issue
that looks like that like. 2 hours after 1.3 10 went out

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2021-11-27:
Yeah, this is the one that got fixed just before 1.3.10 - it happened when the final page had only a single word; it's special-cased since it means that no REP command is sent. 

Python is weird. Says "takes 2 positional arguments", I see two arguments, it sees a third, but when one of them is removed, it works. I don't know what self is and when to use it and when not to.

closed 2021-11-27 by SpenceKonde

nabelekt, 2021-11-27:
Indeed, what's on master looks correct. Thanks!