SpenceKonde/DxCore #203: printf doesn't error on compilation when missing an argument

nabelekt, 2021-12-01:
Note: I am still on 1.3.10 (not sure where to clone the repo so that I can off of master).

I have the following code:
```
#include <SD.h>

void setup()
{
    Serial.begin(19200);

    SD.begin(7);
}

void loop()
{
    delay(1000);

    unsigned long ta, tb, t0, t1, t2, t3, t4 = {0};

    ta = millis();
    t0 = millis();
    t1 = millis();
    t2 = millis();
    t3 = millis();
    t4 = millis();
    tb = millis();
    
    Serial.printf("\n\n00: %ld\n01: %ld\n02: %ld\n03: %ld\n04: %ld", t1, t2, t3, t4);
    Serial.printf("\n00: %ld", t0);
    Serial.printf("\n a: %ld\n b: %ld", ta, tb);
}

}
```
running on an AVR128DA28.

Note that the first call to printf references five variables, but only four are provided. I don't get any warnings or errors on compilation. I am pretty sure that `printf()` from `stdio.h` would provide a warning or error.
This is what I see in the Serial Monitor:
```

00: 1015
01: 1015
02: 1015
03: 1015
04: 268435455
00: 1015
 a: 1015
 b: 1015

00: 2021
01: 2021
02: 2021
03: 2021
04: 268435455
00: 2021
 a: 2021
 b: 2021

00: 3028
01: 3028
02: 3028
03: 3028
04: 268435455
00: 3028
 a: 3028
 b: 3028
...
```
What is a bit disconcerting to me is if I remove the `SD.begin(7)` line, I get:
```

00: 1000
01: 1000
02: 1000
03: 1000
04: 1000
00: 1000
 a: 1000
 b: 1000

00: 2003
01: 2003
02: 2003
03: 2003
04: 2003
00: 2003
 a: 2003
 b: 2003

00: 3007
01: 3007
02: 3007
03: 3007
04: 3007
00: 3007
 a: 3007
 b: 3007
```
which dangerously looks correct and could be very misleading.

SpenceKonde, 2021-12-01:
Um, okay, I dont' know what to do about this,.,. I'm busy writing assembly
implementations of serial so they'll be efficient.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

SpenceKonde, 2021-12-01:
Like, this is avrlibc code doing the prinf-ing, and I think that might be a bug with their implementation.  @MCUdude 

Does it give warnings with other kinds of printf-ing?

softhack007, 2021-12-01:
Just one comment on the expected unix-like printf() behaviour:
``    Serial.printf("\n\n00: %ld\n01: %ld\n02: %ld\n03: %ld\n04: %ld", t1, t2, t3, t4);`` - as you say, the format string requires 5 parameters, but you only provided 4.

In principle, the compiler (gcc) cannot help you in this case, because it does not know what the format string means. Actually your scenario it is a typical cause of software crash. Maybe some clever libc implementations would recognize this, because stdarg.h allows to scan through all the arguments provided so the runtime might warn you, or not.

But still, it is considered as "undefined behaviour" on linux or unix, so you could be lucky and libc might catch this, or you might be not so lucky and the missing parameter is filled in with random crap from the CPU stack. Or with some luck, the software segfaults and you get a core dump, to inspect in the debugger ;-)

TL;DR: at least on unix/linux, the user has to ensure that the number of arguments passed to printf() is consistent with the format string.

nabelekt, 2021-12-01:
Thanks for your comment @softhack007. Are you saying that it's not gcc that gives me this warning:
```
thomas@ThinkPad-3:~$ cat printf_test.c
#include <stdio.h>

int main()
{
        printf("%d %d %d", 1, 2);
        return 0;
}
thomas@ThinkPad-3:~$ gcc printf_test.c -o printf_test.exe
printf_test.c: In function ‘main’:
printf_test.c:5:17: warning: format ‘%d’ expects a matching ‘int’ argument [-Wformat=]
    5 |  printf("%d %d %d", 1, 2);
      |                ~^
      |                 |
      |                 int
thomas@ThinkPad-3:~$ gcc --version
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
?

nabelekt, 2021-12-01:
> Does it give warnings with other kinds of printf-ing?

I don't think I've seen any warnings at all when using `printf`.

softhack007, 2021-12-01:
Hi @nabelekt, 
For sure what you showed from your pad comes from gcc. A very recent version actually. What I'm saying is that your example is undefined behaviour, so the compiler or libc might be nice and warn you, or not. The C language itself does not forbid it.

Newer gcc versions include analyzer functionality that was previously only available in separate tools like "lint" or "purify". No idea when this check was introduced into gcc, however older versions are not doing this kind of analysis. So in case that you did not get a warning using DxCore, there is probably not much that can be done about it.

nabelekt, 2021-12-01:
Got it, make sense. I'll go ahead and close this issue then.

softhack007, 2021-12-01:
PS just took a small dive into compiler docs at gcc.gnu.org. The warning that @nabelekt would like to see seems to be enabled when passing "-Wformat" on the command line. In case that "our" gcc (used by DxCore) supports this flag, it could be added in the compilation process. 

@SpenceKonde maybe you can quickly check if -Wformat is supported?

From https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html

-Wformat
-Wformat=n

Check calls to printf and scanf, etc., to make sure that the arguments supplied have types appropriate to the format string specified, and that the conversions specified in the format string make sense. This includes standard functions, and others specified by format attributes (see Function Attributes), in the printf, scanf, strftime and strfmon (an X/Open extension, not in the C standard) families (or other target-specific families). Which functions are checked without format attributes having been specified depends on the standard version selected, and such checks of functions without the attribute specified are disabled by -ffreestanding or -fno-builtin.

The formats are checked against the format features supported by GNU libc version 2.2. These include all ISO C90 and C99 features, as well as features from the Single Unix Specification and some BSD and GNU extensions. Other library implementations may not support all these features; GCC does not support warning about features that go beyond a particular library’s limitations. However, if -Wpedantic is used with -Wformat, warnings are given about format features not in the selected standard version (but not for strfmon formats, since those are not in any version of the C standard). See Options Controlling C Dialect. 

[....]

nabelekt, 2021-12-02:
Reopened so `-Wformat` can be investigated.
Just opened #204 (semi-related).

nabelekt, 2021-12-02:
Actually, I think the `-Wformat` flag is already in use:
```
main\main.ino:239:81: warning: format '%lld' expects argument of type 'long long int', but argument 3 has type 'uint32_t {aka long unsigned int}' [-Wformat=]
   sprintf(&output_str_buffer[strlen(output_str_buffer)], ", %3lld", lifetime_sec);
```
(Trying to use the correct format causes issues discussed in #204.)

softhack007, 2021-12-02:
Then maybe the header that defines Serial.printf just needs  "__attribute__((format(printf(" to tell the compiler to check the function arguments.

MCUdude, 2021-12-02:
> Then maybe the header that defines Serial.printf just needs "attribute((format(printf(" to tell the compiler to check the function arguments.

Good catch! Replacing the print definition in Print.h with the code below enables warnings:

```c++
int16_t printf(const char *format, ...) __attribute__ ((format (printf, 2, 3)));
```

However, I'm not able to get it to work with the F() macro one:

```
int16_t printf(const __FlashStringHelper *format, ...) __attribute__ ((format (printf, 2, 3)));
// format argument does not have a string type

softhack007, 2021-12-02:
Probably it should be "__attribute__ ((format (printf, **1, 2**)));" - format string is first arg, variable args start with second arg.

SpenceKonde, 2021-12-02:
wouldn't surprise me at all if that warning functionality didn't support F() and could not be made to....

MCUdude, 2021-12-02:
> Probably it should be __attribute__ ((format (printf, 1, 2))); - format string is first arg, variable args start with second arg.

I can't get it to compile with `__attribute__ ((format (printf, 1, 2)));`, but `__attribute__ ((format (printf, 2, 3)));` compiles and works just fine...

> wouldn't surprise me at all if that warning functionality didn't support F() and could not be made to...

You're probably right. I don't see how I could trick the compiler into accepting `const __FlashStringHelper*` instead of `const char*`, which is a bummer.

softhack007, 2021-12-03:
> can't get it to compile with __attribute__ ((format (printf, 1, 2)));, but __attribute__ ((format (printf, 2, 3))); compiles and works just fine...

After a second loo into the gcc docs, I think your solution is correct. "Since non-static C++ methods have an implicit this argument, the arguments of such methods should be counted from two, not one".

If the same does not work for `__FlashStringHelper*` , maybe `__attribute__ ((format_arg (2)))` can be used, which is somehow a "weaker" check.

SpenceKonde, 2021-12-03:
Yeah - considering the roundabout way that flashStringHelper works I;'d be surprised if it was anything else.

closed 2021-12-03 by SpenceKonde

nabelekt, 2021-12-04:
@SpenceKonde, your commit at https://github.com/SpenceKonde/DxCore/commit/2788d05612e4f0f5d00b36480249c38130c080df says that 1.4.0 will have warnings enabled, but only updates the documentation?

SpenceKonde, 2021-12-04:
https://github.com/SpenceKonde/DxCore/commit/61d7c06595d497e602a965ec5997d7d004cf41d0

SpenceKonde, 2021-12-04:
The readme update was spread out over several commits, while the print.h one was earlier.