SpenceKonde/DxCore #204: printf buggy with uint32_t, uint64_t, %s

nabelekt, 2021-12-02:
Is there somewhere else that I should post printf/sprintf issues?

Here's my code:
```
void setup()
{
    Serial.begin(19200);
    Serial.printf("STARTING\n");

    uint64_t a = 1;
    uint32_t b = 25;

    char output_str_buffer[256] = {"\0"};
    sprintf(output_str_buffer, "%s", "THIS IS MY TEST STRING\n");

    Serial.printf("%s", output_str_buffer);
    Serial.printf("\n%d", b);
    Serial.printf("\n%5d %3d", a, b);
    Serial.printf("\n1: %5d %3d: %s",     a, b, output_str_buffer);
    Serial.printf("\n2: %5d %3ld: %s",    a, b, output_str_buffer);
    Serial.printf("\n3: %5ld %3d: %s",    a, b, output_str_buffer);
    Serial.printf("\n4: %5ld %3ld: %s",   a, b, output_str_buffer);
    Serial.printf("\n5: %5ld %3lld: %s",  a, b, output_str_buffer);
    Serial.printf("\n6: %5lld %3ld: %s",  a, b, output_str_buffer);
    Serial.printf("\n7: %5lld %3lld: %s", a, b, output_str_buffer);
    Serial.printf("\n8: %5d %3lld: %s",   a, b, output_str_buffer);
    Serial.printf("\n9: %5lld %3d: %s",   a, b, output_str_buffer);
    Serial.printf("\n10: END");
}

void loop()
{
}
```

Here's my serial console output:
```
⸮STARTING
THIS IS MY TEST STRING

25
    1   0
1:     1   0: ⸮
2:     1   0: ⸮
3:     1   0: ⸮
4:     1   0: 
5:     1 
6: 
7: 
8:     1 
9: 
10: END
```
What I'm trying to get:
```
    1  25: THIS IS MY TEST STRING
```

Any ideas?

Thanks!

nabelekt, 2021-12-02:
I think there are a couple different issues here. I think one of them is that trying to print a `uint64_t` with `%lld` is not supported.

SpenceKonde, 2021-12-02:
Did you try all the versions of printf? There are three options in the tools menu, that give different flash-to-feature tradeoffs. 

My impression was that 64-bt types were generally really janky on the AVRs, and it wouldn't surprise me if none of the versions supported it. I'm just exposing the library that's included with the compiler package. Writing and maintaining that is not my department.

nabelekt, 2021-12-02:
Yes, same issues with all three options.

Is their a repo that you can point me to where it'd be more appropriate to fille an issue?

Thanks!

SpenceKonde, 2021-12-02:
It's part of avr-libc. I'm not sure where or if there's a place to file bugs for that. 

I think it's "maintained" by Microchip now

closed 2021-12-02 by SpenceKonde

nabelekt, 2021-12-03:
Hmm. Seems like avr-libc isn't being maintained?
https://www.nongnu.org/avr-libc/bugs.html has a last updated date in 2010.
The last release at http://download.savannah.gnu.org/releases/avr-libc/ was in 2016.

SpenceKonde, 2021-12-03:
It's being maintained by Microchip, for certain very permissive definitions of "maintain". Though, for the most part, it's done. The bits and pieces that have needed fixes have all been in headers, stuff like eeprom.h (not to be confused with EEPROM.h) which was wrong for the Dx for a while, not the core libc files that get precompiled to .a or .o files. That stuff gets fixed in the packs, not the actual avr-libc proper.

The latest version is 2.0.0 "with modifications". I think it's considered to be "done" by Microchip (which isn't a huge open source booster, either. Just try to find the download for the actual source code of the latest toolchain version (3.6.2, not 3.6.1) - the definitions of open source they subscribe to are also a bit of a stretch)... They more often issue fixes via the packs, because the stuff that's in flux is all in the headers, not the stuff that gets precompiled to .a/.o files.

 After the better part of a decade of Atmel letting the architecture languish, with exorbitant pricing and stale features as it became increasingly obvious that xmega wasn't gaining the traction they expected (maybe because it's a byzantine mess of complexity most of which doesn't add value, in contrast to the relative simplicity of AVRs traditionally? Or because it doesn;'t have 5v tolerant pins? Or that they lacked the computational power of a 32-bit ARM?). it's hardly surprising that the people who volunteered time for the toolchain