SpenceKonde/DxCore #209: Add 27mhz external option for uSDX?

wa2mze, 2021-12-06:
The uSDX ham transceiver group is looking at porting the code currently running on the ATmega328 to either the atmega4096 or the AVR128DB-32 (or 48).  It's a case of feature creep vs available flash, and the DXcore devices do have A/D converters with 2 extra bits (better S/N ratio) plus a D/A converter (the rig is currently using PWM to generate audio from the DSP code).  The rig uses an Si5351a clocked at either 25 or 27 mhz, with 27mhz being the preferred frequency to support various digital modes.  SO, it would be nice to have the 27mhz clock rate available in DXcore, then a single clock could drive BOTH the PLL/DDS and the processor.
PLEASE, if not too difficult to add.  TNX

SpenceKonde, 2021-12-07:
Good god, don't use the 4809. The 4809 as Microchip's stopgap so they weren't trying to compete for 5 years with just the classic AVRs, which Atmel had allowed to languish. The DB-series beats the stuffing out of them.

Adding 27 MHz clock isn't hard, I think all the math is done automatically except for one part

Micros timekeeping, which sounds kind of important for your application, requires dividing a number such that 1/2 of the clock speed in kHz comes out to comes out to be 1000. That is, it must be divided by 13.5. The problem is that you can't use division; if you did, micros would take too long to return. 

So we use a hand-determined combination of right-shifts and addition/subtraction. 

In the particularly important cases of the troublesome tens 5/10/20/40 and terrible twelves, 12/24/48 (also gets 6 and 3 covered), I wasn't satisfied with the job the compiler did  and wound up implementing it in assembly, cutting the execution time of micros down by around 20%, such that it runs in 5ish us at 24 MHz, and 6us at 20..... Which probably wasn't worth it, but anyway. 

Oh, and you'd probably want delayMicroseconds() to work with non-constant values for the delay....but IIRC, the 24MHz calculation does 3 passesthrough an 8 clock loop per microsecond delay, I can just copy the 8-cycle loop and make it a 9 cycle one. and burn a few extra clocks as we enter...

SpenceKonde, 2021-12-07:
GODDAAAAMNNNNNN 27 MHz comes out clean.... 

>>4 -> intermediate. (13.5/16ths of target) 
+ intermediate >> 2  (16.875/16ths of target)
- (intermediate >> 2) >> 2  (16.03125/16ths -  insanely close - 2 parts per thousand off); 5 more rightshifts of that value and subtracting it again results in essentially perfect micros calculation. Who the hell would have expected that from 27?

SpenceKonde, 2021-12-07:
That was considerably less painful than I was expecting..

SpenceKonde, 2021-12-07:
It's now available in github, and will be in the (coming very soon) 1.4.0 release (megatinycore 2.5.0 is coming out at same time - both are huuuuuge.  Serial and Wire both got major improvements, especially Wire.....

are you aware that you can use a crystal on the DB, and have it output the clock on PA7 and use that for the rest of the system?  (not sure how accurate it has to be though)

BTW, Speaking of megaTinyCore; if half a percent is good enough, you could tune a 2-series tinyAVR's internal oscillator to run at 27 too.....

closed 2021-12-07 by SpenceKonde

wa2mze, 2021-12-07:
Thanks Spence!  The accuracy of the 27mhz crystal isn't super critical as we can compensate in the software calculations for the Si5351 settings.  However, there shouldn't be any reason why one (or both) of the 22pf (typical) loading caps couldn't be shunted with a trimmer capacitor so the oscillator could be netted to frequency.  Then again, if we wanted that much accuracy it would make more sense to use an external oscillator circuit with the crystal in an oven and just use the external oscillator input on the 'DB.  (Is there an option for that vs using the crystal?)

SpenceKonde, 2021-12-07:
Good lord, trimmer caps? How the hell accurate are you trying for (also, 22pf sounds high. Most modern crystals are more 18pf at most, and then the crystal pins on the chip cover like 4pf of that.... particularly with these newer parts, they seem to be more able to deal with low load capacitance than high load capacitance.
Yeah, all the modern AVRs (and every classic AVR I can think of) sucpports an external *clock* - the DB is just the first one that had support for a crystal (the DD series and future EA-series will too; the EA being notable for it's having a mindblowing ADC, like the one in the 2-series tinies (they use tinyAVR as a trial run for new technologies - a bunch of the stuff that showed up on 1-serties tinies was refined and featured on the DA/DB-series. 

Sounds like the tuning the internal on a tinyAVR isn;t going to do the if you're considering that sort of thing heh.... . It's only accurate down to 0.25-0.5%, but figured it was worth mentioning, in case that sort of margin was okay (that's a tinyAVR 2-series, not 0/1 - they have twice the granularity of oscillator adjustment that the 0/1's do, even less voltage and temperature dependance, though all the modern AVRs; internal oscillator is pretty amazing. Totally different animal than the classic AVR oscillators. (right now the 27 MHz clock support is only in DxCore, though it will at some point get ported to megaTinyCore, but I don't think it's a particul;ar priority.

wa2mze, 2021-12-07:
Yes, 18pf sounds better, I don't remember which value I have more of in the junque box (how about 15pf caps?).  I strongly suspect that using a 'standard' crystal with the usually recommended load caps will be quite close enough, most Si5351 circuits don't have trimmers or even external load capacitors and just use the internal caps on the IC (IIRC there are two or three options selectable by a register).  If the oscillator isn't spot on frequency, then one adjusts the assumed value in software to compensate.  But some anal types would use trimmers (I say why add hardware to fix something you can fix in software?).