SpenceKonde/DxCore #211: Bug in pin pullup configuration

mikrocoder, 2021-12-15:
Hello,

I have to come back to your wrong pullup configuration. This does not work correctly at all. Depending on the pin at your whim. I still don't know why you are deliberately misconfiguring this and leading every user absolutely astray.
The pin must be set to input, DIR and OUT 0 and only the pullup bit must be set.

```cpp
#include <utilities.h>

void setup (void)
{
  Serial1.begin(250000);                
  Serial1.println("\nµC Reset #### #### ####");
  showPORTARegister(Serial1, 0);

  pinMode(7, INPUT_PULLUP);
  pinMode(6, INPUT_PULLUP);
  pinMode(3, INPUT_PULLUP);
  pinMode(2, INPUT_PULLUP);
  showPORTARegister(Serial1, 0);
}

void loop (void)
{
  
}

void showPORTARegister (Stream &out, const unsigned long interval)
{
  static unsigned long lastMillis {0};

  if (millis() - lastMillis >= interval)
  {
    lastMillis += interval;
    out.println("PORT.A:  ");
    out.print("DIR:       ");  formatBIN(out, PORTA.DIR, true);        out.print(" "); formatHEXln(out, (uint16_t)&PORTA.DIR, true);
    out.print("OUT:       ");  formatBIN(out, PORTA.OUT, true);        out.print(" "); formatHEXln(out, (uint16_t)&PORTA.OUT, true);
    out.print("PIN2CTRL:  ");  formatBIN(out, PORTA.PIN2CTRL, true);   out.print(" "); formatHEXln(out, (uint16_t)&PORTA.PIN2CTRL, true);
    out.print("PIN3CTRL:  ");  formatBIN(out, PORTA.PIN3CTRL, true);   out.print(" "); formatHEXln(out, (uint16_t)&PORTA.PIN3CTRL, true);
    out.flush();
    out.println();
  }
}
```

result is

```cpp
µC Reset #### #### ####
PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b0000'0000 0x0404
PIN2CTRL:  0b0000'0000 0x0412
PIN3CTRL:  0b0000'0000 0x0413

PORT.A:  
DIR:       0b0000'0000 0x0400
OUT:       0b1100'1100 0x0404
PIN2CTRL:  0b0000'1000 0x0412
PIN3CTRL:  0b0000'1000 0x0413
```

Please just make it right without hidden wrong bit settings.

SpenceKonde, 2021-12-16:
Can you describe a situation where this produces wrong behavior that can be detected by any means other than reading the registers? 

The reason I did this change, which I was not happy to make, was that there is apparently classic AVR code in the wild that relies on 
pinMode(pin,INPUT_PULLUP);
pinMode(pin, OUTPUT);

resulting in the pin ending up as an OUTPUT set HIGH. From their perspective, it is a bug because "your pinMode doesn't work like the real one". 

I eventually relented and implemented this. 

And digitalWrite() works the same stupid way, turning on the pullups when a pin gets written high while input... BECAUSE THIS IS ALSO USED IN THE GODDAMNED WILD

I hate this behavior too, but users demand it or they complain to me and then shittalk the core on forums "Oh don't bother with that core it doesn't even implement pinMode right!" that kind of crap. 

 pinMode and digitalWrite in general are garbage functions, even on classic AVRs, but implementing them on a part that isn't a classic AVR is pounding a square peg (made out of garbage) into a round hole. 

I can't fucking win can I?

SpenceKonde, 2021-12-16:
Of course I could add a tools submenu to select this, except that the number1 complaint about all of my cores is that there are too many tools submenus!

mikrocoder, 2021-12-17:
Hello,

so if you switch on the pullup with the old controllers like this
```cpp
pinMode(pin,INPUT);
digitalWrite(pin, HIGH);
```
There was a problem when programming an I2C bus scanner. It doesn't work if OUT and PULLUP are set at the same time. I know there is a function called Wire.usePullups(). But this is used as a workaround and is not really necessary. The problem is that OUT and Pullup are set at the same time. This is wrong anyway no matter how you look at it. usePullups clears the OUT bit which is correct, so you might as well do it right and please don't use OUT wrong.

With the old controllers it is different. With them you had to set the pin to input and the PORT register to '1'. This activates the pullup. Only the bit in the PORT register must be set. There is no other way. This would mean with the new controllers that you only have to set the bit in the OUT register. But this is nonsense, because the new controllers have a pullup bit which can be set. You don't have to carry an earlier method with you if there are better possibilities. Furthermore it is not guaranteed that the pin structure really works like before. The controllers are not 1:1 comparable. With the not working I2C scanner the I2C unit could not pull down the level if OUT and Pullup are active at the same time. So it could not detect that there is no answer at the requested address if no one is answering.

Furthermore I would not be influenced by people who have no idea. No matter what they think or want. The controller is not for beginners anyway. By the way MCUdude does it right and sets only the pullup bit. Since you both know each other well, I am surprised that you stick to the OUT register.

I have no problems with the menu. There is a lot to set so you have to set a lot. That is completely normal.

Translated with www.DeepL.com/Translator (free version)

SpenceKonde, 2021-12-18:
Oh I know that the PORTx.OUT register has no effect when the PORT.DIR is not set. 

All the cores special-case "digitalWrite(pin,HIGH)" to turn on the pullups (it breaks tons of code if it doesn't do that - this is not optional. code that relies on this behavior is very common in the wild)

My cores (but not other cores) also do it in the other direction, making pinMode(INPUT_PULLUP) set the PORTx.OUT register HIGH. 

This wouldn't be a problem if it weren't for the fucking stupid silicon bug that Microchip has left unfixed for like 5 years on all modern AVRs which prevents I2C from functioning correctly if the pins aren't set LOW, because the damned fools put the override on DIR but not OUT and never bothered to fix it..... 

Let me be clear: I HATE THIS FEATURE TOO! But other people hated that switching an input with pullups on into an output wouldn't result in an OUTPUT HIGH..... 

Altering this behavior constitutes a huge change that will impact hundreds of users. and more discussion than one issue on one of the two impacted cores is required.

With my "planned" move in high gear now I am not going to put a change like this into the 1.4.0 release going out tonight. Major DxCore and megaTinyCore issues should not expect to see work other than panic bugfixes until ATTinyCore 2.0.0 (it's their turn, seriously look at how many releases we've gotten since I last touched it - and there are far more users on attinycore). And that in turn won't see much work until I've finished the move to my new apartment

mikrocoder, 2021-12-18:
Hello,

I can understand your thoughts only partially. But I don't understand how you can transfer the old way of programming to new controllers. If someone doesn't know a completely new controller and programs blindly, he will end up in a dead end at some point. I have reproduced "your" problem on a Nano Every with standard Arduino IDE and MCUdude package. They are 99,9% compatible. In both cases the OUT register is not set. From this I assume that there is no compatibility problem. I don't know where this should come from. Otherwise Arduino.cc would have changed it or built in the old programming method.

This setting of the OUT register is also not the expectation if you explicitly use pinMode INPUT_PULLUP. If anyone wants to toggle a pin, then according to normal logic he must turn off the pullup, toggle to output and set to high. This would be understandable behavior at any time.

This logic alone is wrong.
_"But other people hated that switching an input with pullups on into an output wouldn't result in an OUTPUT HIGH....."_
I cannot comprehend this statement. These people should please speak up. Because people who have no idea about hardware can't get the idea that beside the pullup the push-pull output stage is active. So they can't assume that they can switch quickly. 

If someone wants to use a magic trick to save 2 clocks or 2 lines of code, he can do it for himself. No problem. But please not hidden and please not general. The function must do what the name says.

Because of the I2C Override Bug Errata 2.11.1. This is a bug, but it doesn't matter. Why should you set the OUT register before if you use these pins for I2C? Makes no sense.

Don't understand me wrong. Your DxCore and all the work is totally good and thanks a lot. It's just the pin configuration that I disagree with.

I wish a problem-free apartment move.

Translated with www.DeepL.com/Translator (free version)

SpenceKonde, 2021-12-18:
You have more faith in the userbase than I do, I think....

You see things in *LIBRARIES* that have non-trivial userbases that are
enough to make one physically ill. Hell, you see them do horrible things in
the official core. And don't even get me started about what I found in
ATTinyCore last time I was refactoring it.

The change in the way pullups are enabled caused a *lot* of pain for
people, it would seem - the old ideoms on classic AVRs were *really*
widespread - often inside libraries that were used by people who didn't
know how the library they used works (that's the usual reason that things
cause compatibility dustups)

On the other hand, nobody seemed to even notice when I made digitalRead()
not turn off PWM (turning off PWM is a really brutal hit, and it's much
worse on modern AVRs. (I've got a plan for that issue, and some related
ones, but I still on a conceptual level think that digitalRead() should not
change the state of a pin; luckily nobody has argued with me over it :-P) .

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

microPaul, 2021-12-20:
> Of course I could add a tools submenu to select this, except that the number1 complaint about all of my cores is that there are too many tools submenus!

Not that my opinion matters much, but I for one appreciate that you offer all the tools submenu choices.  Makes the system very flexible and easy for me to get exactly what I want it to be.

pfoun, 2021-12-20:
> Not that my opinion matters much, but I for one appreciate that you offer all the tools submenu choices. Makes the system very flexible and easy for me to get exactly what I want it to be.

I agree.  I also appreciate the flexibility the tools submenus provide.  I wonder if complaints of too many tools submenus are followed by complaints of parts not being configured properly for specific needs?

mikrocoder, 2022-01-23:
The flexibility of the submenus is good. But this one with the pullups is not one of them. My opinion is that it is a basic decision to program cleanly without surprises.

SpenceKonde, 2022-01-24:
40 days, and nobody has spoken in support of this behavior, it will be removed from the next release.

XFer012, 2022-02-05:
My 2 cents:

`pinMode(pin,INPUT_PULLUP);
pinMode(pin, OUTPUT);`

this is insane.
If people use this, be it in tutorial or in libraries, it's their fault.
They should update their code to a better standard; one cannot cripple the behaviour of new MCU families because of existing ugly & wrong code.

mikrocoder, 2022-02-05:
What are you talking about?
Who are you talking to?
What is wrong code?

SpenceKonde, 2022-02-05:
code that relies on pinMode(pin,INPUT_PULLUP) setting the port register to 1

mikrocoder, 2022-02-05:
I don't understand what XFer012 means. What's wrong with that? What is wrong with pinMode? XFer012 jumped in here in the middle and I'm just missing his point of reference.

SpenceKonde, 2022-02-06:
You were the one who originally complained about this..  He is arguing the same thing you are, that setting pins input_pullup should not change port register, and code that relies on 
```
pinMode(pin,INPUT_PULLUP);
pinMode(pin,OUTPUT); 
```
setting the pin high, which is normal behavior on classic AVRs and which is what I was emulating with the behavior that you complained about in the original post

XFer012, 2022-02-06:
Exactly that. @SpenceKonde is very nice of you trying not to break ugly existing code, but if this somehow impairs the code for these new MCU families, I'd say let's do things "the right way" and let ugly existing code on the shoulders of people who wrote it. :)

mikrocoder, 2022-02-06:
Now I understand what XFer012 means. That was not recognizable for me. His text was too short without further explanation. Then I can only say - correct.

SpenceKonde, 2022-03-11:
This functionality has been removed in accordance with user consensus., and was a while ago without having been properlly noted at the time.

closed 2022-03-11 by SpenceKonde