SpenceKonde/DxCore #214: Improvements to pinConfigure

MCUdude, 2021-12-20:
Due to limited time the last six months (I have two small kids now, life is busy...) I haven't been able to catch up to all the work you have been putting into DxCore and megaTinyCore. 

Anyways, I decided to implement pinConfigure to MegaCoreX, because I see the benefits of having such functionality. It works well, but is so compact and optimized it's very difficult to keep track of what's going on without reading in the datasheets and the io.h file at the same time.

The first improvement would be to convert this into a template function. I've done it, and it works great. Instead of having to OR the pin settings together, you can just add them as separate parameters.

```c
pinConfigure(PIN_PA0, PIN_DIR_OUT);
pinConfigure(PIN_PA1, PIN_DIR_OUT, PIN_OUT_HIGH, PIN_INVERT_SET);
pinConfigure(PIN_PA1, PIN_DIR_IN, PIN_PULLUP_SET);
```

The second improvement is to use a typedef enum instead of defined constants that provides no guard rails. By using enums, the user is only allowed to pass the PIN constants, unless not casted. We done need any ::'s The will still appear and work with

What do you think? Here's what all this looks like (Because this is for MegaCoreX, I've removed the MVIO part):

```c++
typedef enum
{
 // OUTPUT
  PIN_DIR_SET        = 0x0001,
  PIN_DIRSET         = 0x0001,
  PIN_DIR_OUTPUT     = 0x0001,
  PIN_DIR_OUT        = 0x0001,
 // INPUT
  PIN_DIR_CLR        = 0x0002,
  PIN_DIRCLR         = 0x0002,
  PIN_DIR_INPUT      = 0x0002,
  PIN_DIR_IN         = 0x0002,
 // TOGGLE INPUT/OUTPUT
  PIN_DIR_TGL        = 0x0003,
  PIN_DIRTGL         = 0x0003,
  PIN_DIR_TOGGLE     = 0x0003,
 // HIGH
  PIN_OUT_SET        = 0x0004,
  PIN_OUTSET         = 0x0004,
  PIN_OUT_HIGH       = 0x0004,
 // LOW
  PIN_OUT_CLR        = 0x0008,
  PIN_OUTCLR         = 0x0008,
  PIN_OUT_LOW        = 0x0008,
// CHANGE/TOGGLE
  PIN_OUT_TGL        = 0x000C,
  PIN_OUTTGL         = 0x000C,
  PIN_OUT_TOGGLE     = 0x000C,
//Interrupt disabled but input buffer enabled
  PIN_ISC_ENABLE     = 0x0080,
  PIN_INPUT_ENABLE   = 0x0080,
 // Interrupt on change
  PIN_ISC_CHANGE     = 0x0090,
  PIN_INT_CHANGE     = 0x0090,
// Interrupt on rising edge
  PIN_ISC_RISE       = 0x00A0,
  PIN_INT_RISE       = 0x00A0,
// Interrupt on falling edge
  PIN_ISC_FALL       = 0x00B0,
  PIN_INT_FALL       = 0x00B0,
// Interrupt and input buffer disabled
  PIN_ISC_DISABLE    = 0x00C0,
  PIN_INPUT_DISABLE  = 0x00C0,
// Interrupt enabled with sense on low level
  PIN_ISC_LEVEL      = 0x00D0,
  PIN_INT_LEVEL      = 0x00D0,
// PULLUP ON
  PIN_PULLUP_ON      = 0x0100,
  PIN_PULLUP         = 0x0100,
  PIN_PULLUP_SET     = 0x0100,
// PULLUP OFF
  PIN_PULLUP_OFF     = 0x0200,
  PIN_PULLUP_CLR     = 0x0200,
  PIN_NOPULLUP       = 0x0200,
// PULLUP TOGGLE
  PIN_PULLUP_TGL     = 0x0300,
  PIN_PULLUP_TOGGLE  = 0x0300,
// PIN INVERT ON
  PIN_INVERT_ON      = 0x4000,
  PIN_INVERT_SET     = 0x4000,
// PIN INVERT OFF
  PIN_INVERT_OFF     = 0x8000,
  PIN_INVERT_CLR     = 0x8000,
// PIN_INVERT_TOGGLE
  PIN_INVERT_TGL     = 0xC000,
  PIN_INVERT_TOGGLE  = 0xC000,
} pin_configure_t;

// Helper function to catch the last argument
pin_configure_t pinConfigure(const uint8_t digital_pin, const pin_configure_t mode)
{
  (void)digital_pin;
  return mode;
}

// PinConfigure variadic template function
template<typename MODE, typename... MODES>
void pinConfigure(const uint8_t digital_pin, const MODE& mode, const MODES&... modes)
{
  // Start by or-ing together the arguments
  uint8_t pin_config = pinConfigure(digital_pin, mode) | pinConfigure(digital_pin, modes...);
 
  uint8_t bit_mask = digitalPinToBitMask(digital_pin);
  if(bit_mask == NOT_A_PIN || !pin_config) // Return if digital pin is invalid or the other parameters or out to zero
    return;

  uint8_t bit_pos  = digitalPinToBitPosition(digital_pin);
  volatile uint8_t *portbase = (volatile uint8_t*) digitalPinToPortStruct(digital_pin);

  // Write to selected pin direction register
  uint8_t setting = pin_config & 0x03; // Mask out direction bits (DIR, DIRSET, DIRCLR, DIRTGL)
  if(setting) 
    *(portbase + setting) = bit_mask;

  // Write to selected output register
  pin_config >>= 2;
  setting = pin_config & 0x03;
  if(setting)
    *(portbase + 4 + setting) = bit_mask;

  // Return if there is nothing more to configure
  if(!(pin_config & 0x3FFC))
    return;

  uint8_t oldSREG = SREG; // Store SREG
  cli(); // Disable interrupts

  // PINnCTRL register
  pin_config >>= 2;
  uint8_t pinncfg = *(portbase + 0x10 + bit_pos);
  // Input sense configuration (ISC)
  if(pin_config & 0x08)
    pinncfg = (pinncfg & 0xF8) | (pin_config & PORT_ISC_gm);
  // Pullup resistor
  uint8_t temp = pin_config & 0x30;
  if(temp)
  {
    if(temp == 0x30)
      pinncfg ^= PORT_PULLUPEN_bm;    // Toggle pullup
    else if(temp == 0x20)
      pinncfg &= ~(PORT_PULLUPEN_bm); // Clear pullup
    else
      pinncfg |= PORT_PULLUPEN_bm;    // Set pullup
  }
  // Invert pin
  pin_config >>= 8;
  temp = pin_config & 0x0C;
  if(temp)
  {
    if(temp == 0x0C)
      pinncfg ^= PORT_INVEN_bm;    // Toggle invert
    else if(temp == 0x08)
      pinncfg &= ~(PORT_INVEN_bm); // Clear
    else
      pinncfg |= PORT_INVEN_bm;    // Set
  }
  // Write to PINnCTRL register
  *(portbase + 0x10 + bit_pos) = pinncfg;

  // Restore SREG
  SREG = oldSREG;
}
```

SpenceKonde, 2021-12-23:
Damn, that looks awesome - I was never happy with the need to OR the constants together, but templates are a dark art which I do not have experience with.

SpenceKonde, 2021-12-26:
Hmm, okay - so I do *NOT* know how to make that compile... Which isnt surprising since I don't really know how to use templates much less variadic ones.

MCUdude, 2021-12-26:
The issue may be because your cores already have pinConfigure defined. Try removing or renaming them.
Here is an updated version where pin_configure_t is a 16-bit enum.

```c++
// Compiles with MegaCoreX
typedef enum : uint16_t
{
 // OUTPUT
  PIN_DIR_SET        = 0x0001,
  PIN_DIRSET         = 0x0001,
  PIN_DIR_OUTPUT     = 0x0001,
  PIN_DIR_OUT        = 0x0001,
 // INPUT
  PIN_DIR_CLR        = 0x0002,
  PIN_DIRCLR         = 0x0002,
  PIN_DIR_INPUT      = 0x0002,
  PIN_DIR_IN         = 0x0002,
 // TOGGLE INPUT/OUTPUT
  PIN_DIR_TGL        = 0x0003,
  PIN_DIRTGL         = 0x0003,
  PIN_DIR_TOGGLE     = 0x0003,
 // HIGH
  PIN_OUT_SET        = 0x0004,
  PIN_OUTSET         = 0x0004,
  PIN_OUT_HIGH       = 0x0004,
 // LOW
  PIN_OUT_CLR        = 0x0008,
  PIN_OUTCLR         = 0x0008,
  PIN_OUT_LOW        = 0x0008,
// CHANGE/TOGGLE
  PIN_OUT_TGL        = 0x000C,
  PIN_OUTTGL         = 0x000C,
  PIN_OUT_TOGGLE     = 0x000C,
//Interrupt disabled but input buffer enabled
  PIN_ISC_ENABLE     = 0x0080,
  PIN_INPUT_ENABLE   = 0x0080,
 // Interrupt on change
  PIN_ISC_CHANGE     = 0x0090,
  PIN_INT_CHANGE     = 0x0090,
// Interrupt on rising edge
  PIN_ISC_RISE       = 0x00A0,
  PIN_INT_RISE       = 0x00A0,
// Interrupt on falling edge
  PIN_ISC_FALL       = 0x00B0,
  PIN_INT_FALL       = 0x00B0,
// Interrupt and input buffer disabled
  PIN_ISC_DISABLE    = 0x00C0,
  PIN_INPUT_DISABLE  = 0x00C0,
// Interrupt enabled with sense on low level
  PIN_ISC_LEVEL      = 0x00D0,
  PIN_INT_LEVEL      = 0x00D0,
// PULLUP ON
  PIN_PULLUP_ON      = 0x0100,
  PIN_PULLUP         = 0x0100,
  PIN_PULLUP_SET     = 0x0100,
// PULLUP OFF
  PIN_PULLUP_OFF     = 0x0200,
  PIN_PULLUP_CLR     = 0x0200,
  PIN_NOPULLUP       = 0x0200,
// PULLUP TOGGLE
  PIN_PULLUP_TGL     = 0x0300,
  PIN_PULLUP_TOGGLE  = 0x0300,
// PIN INVERT ON
  PIN_INVERT_ON      = 0x4000,
  PIN_INVERT_SET     = 0x4000,
// PIN INVERT OFF
  PIN_INVERT_OFF     = 0x8000,
  PIN_INVERT_CLR     = 0x8000,
// PIN_INVERT_TOGGLE
  PIN_INVERT_TGL     = 0xC000,
  PIN_INVERT_TOGGLE  = 0xC000,
} pin_configure_t;

pin_configure_t pinConfigure(const uint8_t digital_pin, const pin_configure_t mode)
{
  (void)digital_pin;
  return mode;
}

template<typename MODE, typename... MODES>
void pinConfigure(const uint8_t digital_pin, const MODE& mode, const MODES&... modes)
{
  // Start by or-ing together the arguments
  uint8_t pin_config = pinConfigure(digital_pin, mode) | pinConfigure(digital_pin, modes...);
 
  uint8_t bit_mask = digitalPinToBitMask(digital_pin);
  if(bit_mask == NOT_A_PIN || !pin_config) // Return if digital pin is invalid or the other parameters or out to zero
    return;

  uint8_t bit_pos  = digitalPinToBitPosition(digital_pin);
  volatile uint8_t *portbase = (volatile uint8_t*) digitalPinToPortStruct(digital_pin);

  // Write to selected pin direction register
  uint8_t setting = pin_config & 0x03; // Mask out direction bits (DIR, DIRSET, DIRCLR, DIRTGL)
  if(setting) 
    *(portbase + setting) = bit_mask;

  // Write to selected output register
  pin_config >>= 2;
  setting = pin_config & 0x03;
  if(setting)
    *(portbase + 4 + setting) = bit_mask;

  // Return if there is nothing more to configure
  if(!(pin_config & 0x3FFC))
    return;

  uint8_t oldSREG = SREG; // Store SREG
  cli(); // Disable interrupts

  // PINnCTRL register
  pin_config >>= 2;
  uint8_t pinncfg = *(portbase + 0x10 + bit_pos);
  // Input sense configuration (ISC)
  if(pin_config & 0x08)
    pinncfg = (pinncfg & 0xF8) | (pin_config & PORT_ISC_gm);
  // Pullup resistor
  uint8_t temp = pin_config & 0x30;
  if(temp)
  {
    if(temp == 0x30)
      pinncfg ^= PORT_PULLUPEN_bm;    // Toggle pullup
    else if(temp == 0x20)
      pinncfg &= ~(PORT_PULLUPEN_bm); // Clear pullup
    else
      pinncfg |= PORT_PULLUPEN_bm;    // Set pullup
  }
  // Invert pin
  pin_config >>= 8;
  temp = pin_config & 0x0C;
  if(temp)
  {
    if(temp == 0x0C)
      pinncfg ^= PORT_INVEN_bm;    // Toggle invert
    else if(temp == 0x08)
      pinncfg &= ~(PORT_INVEN_bm); // Clear
    else
      pinncfg |= PORT_INVEN_bm;    // Set
  }
  // Write to PINnCTRL register
  *(portbase + 0x10 + bit_pos) = pinncfg;

  // Restore SREG
  SREG = oldSREG;
}

void setup() {

}

void loop() {

  pinConfigure(13, PIN_OUT_SET, PIN_DIRCLR);
  delay(1000);

}

MCUdude, 2021-12-26:
Ah, you need `inline` in order to get it to compile:

https://github.com/MCUdude/MegaCoreX/commit/47ef471683d9856af8147705d410bf096ba282c8

SpenceKonde, 2021-12-30:
Hmm, uhhh... 

```
uint8_t pin_config = pinConfigure(digital_pin, mode) | pinConfigure(digital_pin, modes...);
```

pinConfigure with more than two modes fails to compile, complaining of there being no operator | defined for pin_configure_t and void - because pinConfigure with a single mode argument calls the helper function which returns a pin_configure_t, while the one with more argument options returns void. This looks like a tricky thing to fix if I'm not missing something.

MCUdude, 2022-01-01:
> This looks like a tricky thing to fix if I'm not missing something.

Happy new year!

Nah, just me being dumb. Fixed now.
https://github.com/MCUdude/MegaCoreX/commit/a7dd83c062ea4f0284bf243bbe4c4025e7dcd979

SpenceKonde, 2022-01-01:
Happy new year to you! 

Huh, okay, yeah, now it;s compiling and looks to be working. I notice that the flash usage of a single call to it is competitive with a call to the old version, but as you call it more, it's flash usage grows markedly faster... Do you happen to know why this would be?

MCUdude, 2022-01-01:
> Huh, okay, yeah, now it;s compiling and looks to be working. I notice that the flash usage of a single call to it is competitive with a call to the old version, but as you call it more, it's flash usage grows markedly faster... Do you happen to know why this would be?

I'm not sure why it's that much larger. The thing about templates is that things are figured out at compile time, so in theory all the ORing and function calling should have been done by the compiler.

Have a look at the OR template function below. Give it a try if you have time. No matter what you put into it, as long as it's know at compile time, the size will not grow much at all.

``` c++
template<typename T>
T orer(T first) {
  return first;
}

template<typename T, typename... Args>
T orer(T first, Args... args) {
  return first | orer(args...);
}

// call orer(arg1, arg2, arg3 etc)
```

SpenceKonde, 2022-01-02:
Based on 
```
void setup() {
  pinConfigure(PIN_PA7, PIN_PULLUP_ON | PIN_ISC_DISABLE | PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_DIR_SET | PIN_INPUT_ENABLE | PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_OUT_CLR | PIN_DIR_SET| PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA6, PIN_PULLUP_ON | PIN_ISC_DISABLE| PIN_INVERT_ON);
  pinConfigure(PIN_PA6, PIN_DIR_SET | PIN_INPUT_ENABLE| PIN_INVERT_ON);
  pinConfigure(PIN_PA6, PIN_OUT_CLR | PIN_DIR_SET| PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA5, PIN_PULLUP_ON | PIN_ISC_DISABLE| PIN_INVERT_ON);
  pinConfigure(PIN_PA5, PIN_DIR_SET | PIN_INPUT_ENABLE| PIN_INVERT_ON);
  pinConfigure(PIN_PA5, PIN_OUT_CLR | PIN_DIR_SET| PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA4, PIN_PULLUP_ON | PIN_ISC_DISABLE| PIN_INVERT_ON);
  pinConfigure(PIN_PA4, PIN_DIR_SET | PIN_INPUT_ENABLE| PIN_INVERT_ON);
  pinConfigure(PIN_PA4, PIN_OUT_CLR | PIN_DIR_SET| PIN_INPUT_DISABLE);
}
void loop() {
  
}
```
vs
``` 
void setup() {
  pinConfigure(PIN_PA7, PIN_PULLUP_ON | PIN_ISC_DISABLE | PIN_INVERT_ON);
  pinConfigure(PIN_PA4, PIN_OUT_CLR | PIN_DIR_SET| PIN_INPUT_DISABLE);
}
void loop() {
}
```
744 vs 496
24.8 bytes per call. 

With the equivalent (substitute |'s for commas):
1424 vs 1070
35.4 bytes per call. 

Now compare:
```

void setup() {
  pinConfigure(PIN_PA7, PIN_PULLUP_ON , PIN_ISC_DISABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA6, PIN_DIR_SET , PIN_INPUT_ENABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA5, PIN_OUT_CLR , PIN_DIR_SET, PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA4, PIN_PULLUP_ON , PIN_ISC_DISABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA3, PIN_DIR_SET , PIN_INPUT_ENABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA2, PIN_OUT_CLR , PIN_DIR_SET, PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA1, PIN_PULLUP_ON , PIN_ISC_DISABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PB0, PIN_DIR_SET , PIN_INPUT_ENABLE, PIN_INVERT_ON);
}
void loop() {
  
}
```
with 
```
void setup() {
  pinConfigure(PIN_PA7, PIN_PULLUP_ON , PIN_ISC_DISABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_DIR_SET , PIN_INPUT_ENABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_OUT_CLR , PIN_DIR_SET, PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA7, PIN_PULLUP_ON , PIN_ISC_DISABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_DIR_SET , PIN_INPUT_ENABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_OUT_CLR , PIN_DIR_SET, PIN_INPUT_DISABLE);
  pinConfigure(PIN_PA7, PIN_PULLUP_ON , PIN_ISC_DISABLE, PIN_INVERT_ON);
  pinConfigure(PIN_PA7, PIN_DIR_SET , PIN_INPUT_ENABLE, PIN_INVERT_ON);
}
void loop() {
  
}
```
1220 vs 1076
[... 277 lines cut ...]
 51a: 5f 4f         sbci  r21, 0xFF ; 255
 51c: be 01         movw  r22, r28
 51e: 6b 5f         subi  r22, 0xFB ; 251
 520: 7f 4f         sbci  r23, 0xFF ; 255
 522: 80 e0         ldi r24, 0x00 ; 0
 524: c8 dd         rcall .-1136    ; 0xb6 <unsigned int pinConfigure<pin_configure_t, pin_configure_t, pin_configure_t>(unsigned char, pin_configure_t const&, pin_configure_t const&, pin_configure_t const&)>
  pinConfigure(PIN_PA4, PIN_OUT_CLR , PIN_DIR_SET, PIN_INPUT_DISABLE);
 526: e9 82         std Y+1, r14  ; 0x01
 528: fa 82         std Y+2, r15  ; 0x02
 52a: 0b 83         std Y+3, r16  ; 0x03
 52c: 1c 83         std Y+4, r17  ; 0x04
 52e: 6d 82         std Y+5, r6 ; 0x05
 530: 7e 82         std Y+6, r7 ; 0x06
 532: 9e 01         movw  r18, r28
 534: 2f 5f         subi  r18, 0xFF ; 255
 536: 3f 4f         sbci  r19, 0xFF ; 255
 538: ae 01         movw  r20, r28
 53a: 4d 5f         subi  r20, 0xFD ; 253
 53c: 5f 4f         sbci  r21, 0xFF ; 255
 53e: be 01         movw  r22, r28
 540: 6b 5f         subi  r22, 0xFB ; 251
 542: 7f 4f         sbci  r23, 0xFF ; 255
 544: 80 e0         ldi r24, 0x00 ; 0
 546: b7 dd         rcall .-1170    ; 0xb6 <unsigned int pinConfigure<pin_configure_t, pin_configure_t, pin_configure_t>(unsigned char, pin_configure_t const&, pin_configure_t const&, pin_configure_t const&)>
```

SpenceKonde, 2022-01-02:
```
void pinConfigure(uint8_t pin, uint16_t pinconfig) {
  check_valid_digital_pin(pin);
  uint8_t bit_mask = digitalPinToBitMask(pin);
  b8:	fc 01       	movw	r30, r24
  ba:	ea 52       	subi	r30, 0x2A	; 42
  bc:	fd 47       	sbci	r31, 0x7D	; 125
  be:	40 81       	ld	r20, Z
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:54
  if (bit_mask == NOT_A_PIN) {
  c0:	4f 3f       	cpi	r20, 0xFF	; 255
  c2:	09 f4       	brne	.+2      	; 0xc6 <pinConfigure+0x10>
  c4:	3f c0       	rjmp	.+126    	; 0x144 <__EEPROM_REGION_LENGTH__+0x44>
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:57
    return;                             /* ignore invalid pins passed at runtime */
  }
  volatile uint8_t *portbase = (volatile uint8_t*) digitalPinToPortStruct(pin);
  c6:	fc 01       	movw	r30, r24
  c8:	ec 53       	subi	r30, 0x3C	; 60
  ca:	fd 47       	sbci	r31, 0x7D	; 125
  cc:	e0 81       	ld	r30, Z
  ce:	20 e2       	ldi	r18, 0x20	; 32
  d0:	e2 9f       	mul	r30, r18
  d2:	f0 01       	movw	r30, r0
  d4:	11 24       	eor	r1, r1
  d6:	fc 5f       	subi	r31, 0xFC	; 252
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:58
  uint8_t bit_pos = digitalPinToBitPosition(pin);
  d8:	8e 54       	subi	r24, 0x4E	; 78
  da:	9d 47       	sbci	r25, 0x7D	; 125
  dc:	dc 01       	movw	r26, r24
  de:	8c 91       	ld	r24, X
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:59
  uint8_t setting = pinconfig & 0x03; // grab direction bits
  e0:	96 2f       	mov	r25, r22
  e2:	93 70       	andi	r25, 0x03	; 3
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:60
  if (setting) {
  e4:	09 f0       	breq	.+2      	; 0xe8 <pinConfigure+0x32>
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:61
    *(portbase + setting) = bit_mask;
  e6:	41 83       	std	Z+1, r20	; 0x01
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:63
  }
  pinconfig >>= 2;
  e8:	9b 01       	movw	r18, r22
  ea:	36 95       	lsr	r19
  ec:	27 95       	ror	r18
  ee:	36 95       	lsr	r19
  f0:	27 95       	ror	r18
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:64
  setting = pinconfig & 0x03; // as above, only for output
  f2:	a2 2f       	mov	r26, r18
  f4:	a3 70       	andi	r26, 0x03	; 3
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:65
  if (setting) {
  f6:	29 f0       	breq	.+10     	; 0x102 <__EEPROM_REGION_LENGTH__+0x2>
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:66
    *(portbase + 4 + setting) = bit_mask;
  f8:	b0 e0       	ldi	r27, 0x00	; 0
  fa:	14 96       	adiw	r26, 0x04	; 4
  fc:	ae 0f       	add	r26, r30
  fe:	bf 1f       	adc	r27, r31
 100:	4c 93       	st	X, r20
C:\arduino-1.8.13-portable\portable\packages\megaTinyCore\hardware\megaavr\2.5.4\cores\megatinycore/wiring_digital.c:68
  }
  if (!(pinconfig & 0x03FFC)) return;
 102:	2c 7f       	andi	r18, 0xFC	; 252
 104:	3f 73       	andi	r19, 0x3F	; 63
 106:	23 2b       	or	r18, r19
[... 567 lines cut ...]
 26a:	21 f0       	breq	.+8      	; 0x274 <__EEPROM_REGION_LENGTH__+0x174>
C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr\cores\megatinycore/Arduino.h:517
    {
      if(temp == 0x0C)
 26c:	2c 30       	cpi	r18, 0x0C	; 12
 26e:	09 f0       	breq	.+2      	; 0x272 <__EEPROM_REGION_LENGTH__+0x172>
 270:	a8 cf       	rjmp	.-176    	; 0x1c2 <__EEPROM_REGION_LENGTH__+0xc2>
C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr\cores\megatinycore/Arduino.h:518
        pinncfg ^= PORT_INVEN_bm;    // Toggle invert
 272:	90 58       	subi	r25, 0x80	; 128
C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr\cores\megatinycore/Arduino.h:525
        pinncfg &= ~(PORT_INVEN_bm); // Clear
      else
        pinncfg |= PORT_INVEN_bm;    // Set
    }
    // Write to PINnCTRL register
    *(portbase + 0x10 + bit_pos) = pinncfg;
 274:	90 83       	st	Z, r25
C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr\cores\megatinycore/Arduino.h:528

    // Restore SREG
    SREG = oldSREG;
 276:	6f bf       	out	0x3f, r22	; 63
 278:	8b cf       	rjmp	.-234    	; 0x190 <__EEPROM_REGION_LENGTH__+0x90>
```

SpenceKonde, 2022-01-02:
What is very clear is that stuff is being done at runtime that should be getting done at compile time.

MCUdude, 2022-01-03:
Ok, I figured it out. The compiler wasn't able to figure out what was going on when the template function called itself. It usually can, but for some reason, not in this case. Fixed now.

https://github.com/MCUdude/MegaCoreX/commit/854d9e1041bef33e3bc50458f5efa686ab5118ec#diff-3425c70d2b78d865cd401a9c605f563eda19ce9cf9d095e63d5770d06423c949

mikrocoder, 2022-02-03:
Hi,
Let me show you my Pin Lib.  [AVRxDB_Pin.zip](https://github.com/SpenceKonde/DxCore/files/7994638/AVRxDB_Pin.zip)

SpenceKonde, 2022-02-04:
I forget where I was on the MCUDude implementation, The last thing I remember, things had been improvedm but there was still a substantial overhead, which matters since pinConfigure gets ported to megaTinyCore where it's serving parts with 2k of flash). But I've owed the classic tiny people a release for too long, ATTinyCore 2.0.0 goes out next. 

> Hi, Let me show you my Pin Lib. [AVRxDB_Pin.zip](https://github.com/SpenceKonde/DxCore/files/7994638/AVRxDB_Pin.zip)

What's the overhead like?

mikrocoder, 2022-02-04:
No overhead, has a small "footprint". Do you have a standard test sketch to compare? Switch any pins or similar. Then I'll adapt that.

SpenceKonde, 2022-03-12:
Sorry for not saying more here sooner. Basically, I can't pull in the code in the condition it is in. I was testing it on megaTinyCore, and the impact on codesize is catastrophic, and renders it unusable on some of the target platforms. 

It appears that the attempts mafr have still been unsuccessful at making sure the ORing always happens at compile time if at all possible. 

I am not qualified to rectify this myself; but I was wondering - would it make a difference if the variadic function was replaced with a macro? Wouldn't that get us to the point where, with compiletime constant values, a series of comma separated values could be made to shake out into a series of values separated by bitwise OR operators? Then any compiletime known ones would get constant folding applied; , and if they included things that weren't constant, it would still fold to the greatest extent possible, and it would never be bitwise or-ing compiletime constant values at runtime. 

I love the improved syntax of pinConfigure(pin,option1,option2,option3) (in fact,I really hate the syntax I was forced to use!). And, from a  logical or complexity stantpoint it make no sense that it couldn't be done - somehow., I see no reason that can't compile to something just as small. Except thathat the implementations we've been able to come up with don't. But I'm pretty useless at this because it's all aspects of the language that I don't really understand. As I've often remarked, I don't know C++,  I know C and Assembly. 

I guess you'd also have the same non-macro function that my implementation uses, which took only a single option argument that is the bitwise combination of the desirerd options, and that's what everything else would condense down to....

SpenceKonde, 2022-03-12:
@mikrocoder -  there are a few reasons I don't really like that code. 
None of them are technical. 

The first thing is, it is not in keeping with the style of Arduino. Every other Arduino function to work with a pin is structured as functionName(pin,options) - a function takes a pin identifier and options (if any) and does something. 

Second, - It's greek to me, I don't understand it very well. 

Third - I still worry if one tried to use if on a larger scale, you'd suddenly find that under some situation, constant folding failed, with catastrophic size and performance effects. The optimizer isn't much better with C++ than I am. For example, classes sem to compltely disable most optimization. If you try to guard Serial.begin() against illegal compiletime known values by doimg 
```
if (__builtin_constant_p(baud)) {
if (baud > maxbaud || baud < minbaud) {
badArg("The requested baud cannot be generated at this clock speed"); 
}
} 
```
(which is what I tried shortly after discovering that technique) It doesn't work. Because it's a class method. even though it's only called once with an integer literal argment, the baud rate is never treated as a compile known constant! C++ feaures appear to be kryptonite to the optimizer, and with flash space and memory at such a premium on the smaller devices, something that makes the code more "modern and object porented-like" at the expense of compiled binary size is not an improvement. Anything I do that increases compile size is something I really think about whether I can justify it, particularly if it might get ported to megaTinyCore which serves 2k parts/ 

In the case of a the automatic clearing of the reset flags and stashing of them in GPIOR0 at startup. I nearly didn't make that the default behavior. But as I considered what the consequences of not doing that were, it was clear since, no matter what I wrote, 95% of users wouldn't follow my copy-paste instructions to ensure reset flags were cleared after every startup. And the problem with that it prevents the chip from detecting tat it just experienced a "dirty reset". hence all expectations of init() and all api functions were violated, we might even be in a state where CPUINT considers us to be in an interrupt and so won't execute any ISRs but we're executing normal code. If execution ever got to 0x0000, it is guaranteed that the code has malfunction, and it is highly likely that doing anthing other than an immediate software reset is unlikely to restore any approximation of functioning, and could result in arbitrary incorrect behavior. I further realized that dirty resets are liekly the main cause of AVRs getting into a "bad state" amd requiring a manual reset to revive. In light of the magnitude of the problems that it would preempt, I decided it would be irresponsible of me to know how to catch that case and not use it by default. 

But support for alt pins with analogWrite? That's in DxCore, but I decided I couldn't justify porting to megaTinyCore. 

There are a number of cases where I got most of the way through implemening some new feature, and started thinking about whether the usefulness justified the flash cost, decided it did not, and removed it.

MCUdude, 2022-03-12:
> Sorry for not saying more here sooner. Basically, I can't pull in the code in the condition it is in. I was testing it on megaTinyCore, and the impact on codesize is catastrophic, and renders it unusable on some of the target platforms.

It's been a while, and I don't remember all the details, but if you have a look at the current [MegaCoreX pinConfigure implementation](https://github.com/MCUdude/MegaCoreX/blob/0d7b0d1a3a28f351ab796c8aaabc4ba1c267ff80/megaavr/cores/coreX-corefiles/Arduino.h#L152-L319), this is able to OR together the parameters at compile time, and is just as small as your implementation with no overhead. The trick was to move the entire OR logic template out of the pinConfigure template.

SpenceKonde, 2022-03-12:
Well, I have to apologize. I'm rather emvbarassed. I re-ran the tests with the version I'd put into megaTinyCore. 
The results did not match my notes at all .

Sometimes your implementation. wins and sometimes mine does... 
What is really interesting tough?

If you use the grammar for my version, with your implementation, the result is the worst of all!! 

So please disregard ,my inaccrate objections. 

I will release the next version, then merge this in.to master and then we can see whether people find problemswith it.

SpenceKonde, 2022-03-12:
Thanks and sorry for bot the long delay an inaccrate claims;

mikrocoder, 2022-03-12:
Hello,

our approaches are just different. :wink: Everyone can handle this as they wish. In any case, I would like to get away from the define macro programming. The risk for errors is too high. I just wanted to show it. It's okay if you don't like it. It's just OOP.

I don't like these multi functions with hundreds of parameters. That is also very error prone. One typo and a pin is misconfigured without the compiler can complain. Or a pin is changed at another place unintentionally. You can overwrite anything with anything.

You can handle this as you like. :wink:

mikrocoder, 2022-03-12:
Hi,

you know what's funny now. I'm testing the Logic and Event examples right now because of deprecated warnings. That's what you wanted me to test. You don't like my pin classes. But the event class is okay? :smile: And with Logic the Arduino user should suddenly deal directly with hardware accesses. That makes your rejection a bit contradictory. :wink:

SpenceKonde, 2022-03-12:
Those libraries are written in a very different style than they would be if I had written them. - they were by @MCUdude - I always find it striking how in some ways we are driven to the same thing - universal cores for all the decent arduino parts, chp-centric rather than board-centric, etc etc  - and et, our tastes in coding style could hardly be more different than they are. The most recent version of the event library was the product of considerable debate, the sort of debate where, at t's heart, the disagreement is on what that goal should be. 

For a great example of what happened when neither of us were aware of the others work, compare the Flash library in DxCore (which I wrote) with the one in megaTinyCore (which he wrote).

Specifically on event, there is a desperate need to have some form of abstraction hat is vaguely usablel; and he had a better idea of how to do it that I did.

The Ex-series is going to need a considerable amount of work there by the way, - the are changing up how pin and PIT events work. The end goal. I think, its good and wise make all channels the same) but that obviously means some major changes are going to be made (you can see hints of this in the initial Ex-seres ATPACK, but they redacted all the interesting parts.

SpenceKonde, 2022-03-12:
He also did Comparator, ZCD, and OPAMP libaries.. Without which there wouldn't be wrappers aroundthat funcrionalty at all. The key difference is that without his libraries, there was *no* wrapper around these things. I don't object to your library existing, but I'd need to do a lot of of work if I was to integrate it with the cores, just to to port it across megaTinyCore and DxCore to all parts, and it's functionality that already exists.through standard API functions.

SpenceKonde, 2022-11-25:
How strange!  With only one or two  calls to pinConfigure, or with lots of calls to it, the commas compile significantly smaller. In the 3-6 call range though, if they're all bitwise OR's, they come out smaller. But the bitwise or otherwise loses! What?!

Anyway, this is at a point where I'm comfortable merging it.

commit 615ec920 (2022-11-25): MCUDude's pinConfigure implementation.

Fix #214
closed 2022-11-25 by SpenceKonde