SpenceKonde/DxCore #221: UPDI download problems 1.4.4

microPaul, 2021-12-30:
I'm still having UPDI download problems even after upgrading to 1.4.4.

I'm using a new homebrew DB32 board (old tests were with homebrew DB64 board).  My Serial-UPDI programmer is based on the CH340 and I'm using a parallel 4.7K/Schottky.

To confirm that my board was OK, I went back to 1.3.9 and that works fine for me with UPDI download.

Here's the error message I get with my blink sketch when using 1.4.4.

`Sketch uses 4144 bytes (3%) of program storage space. Maximum is 131072 bytes.
Global variables use 317 bytes (1%) of dynamic memory, leaving 16067 bytes for local variables. Maximum is 16384 bytes.
Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port COM14 at 230400 baud.
Target: avr128db32
Set fuses: ['5:0b11001001', '6:0b00001100', '7:0x00', '8:0x00']
Action: write
File: C:\Users\p\AppData\Local\Temp\arduino_build_157165/DB64-Rev0-blink-211017-01.ino.hex
Connecting to any uart{upload.with_programmer_not_available_for_bootloader_boards}
Traceback (most recent call last):
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pyedbglib\hidtransport\cyhidapi.py", line 4, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4/tools/prog.py", line 266, in <module>
    main()
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4/tools/prog.py", line 121, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4/tools/prog.py", line 163, in pymcuprog_basic
    backend.connect_to_tool(toolconnection)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pymcuprog\backend.py", line 150, in connect_to_tool
    self.transport = hid_transport()
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pyedbglib\hidtransport\hidtransportfactory.py", line 46, in hid_transport
    from .cyhidapi import CyHidApiTransport
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pyedbglib\hidtransport\cyhidapi.py", line 9, in <module>
    raise ImportError(MSG)
ImportError: This transport class requires cython and hidapi to be installed:

> pip install cython

> pip install hidapi

the selected serial port 
 does not exist or your board is not connected
`

SpenceKonde, 2021-12-30:
Are you attempting to do upload using programmer while an optiboot
definition is selected?

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

microPaul, 2021-12-30:
just rechecked (because I do make mistakes).

Target is a DB32 and no bootloader is installed and I'm trying to compile and upload using "no bootloader" -- Board: AVR DB-series   (no bootloader).  Everything else should be (intended to be) the default options.

Here's a screen shot:

[image]

microPaul, 2021-12-30:
And this is the error I'm getting

`Sketch uses 4048 bytes (3%) of program storage space. Maximum is 131072 bytes.
Global variables use 287 bytes (1%) of dynamic memory, leaving 16097 bytes for local variables. Maximum is 16384 bytes.
Arduino <---> pymcuprog bridge by Quentin Bolsee and Spence Konde
Version 1.1.0 - May 2021
Using serial port COM14 at 230400 baud.
Target: avr128db32
Set fuses: ['5:0b11001001', '6:0b00001100', '7:0x00', '8:0x00']
Action: write
File: C:\Users\p\AppData\Local\Temp\arduino_build_226156/DB32-Uno--blink-211204-01.ino.hex
Connecting to any uart{upload.with_programmer_not_available_for_bootloader_boards}
Traceback (most recent call last):
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pyedbglib\hidtransport\cyhidapi.py", line 4, in <module>
    import hid
ModuleNotFoundError: No module named 'hid'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4/tools/prog.py", line 266, in <module>
    main()
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4/tools/prog.py", line 121, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4/tools/prog.py", line 163, in pymcuprog_basic
    backend.connect_to_tool(toolconnection)
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pymcuprog\backend.py", line 150, in connect_to_tool
    self.transport = hid_transport()
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pyedbglib\hidtransport\hidtransportfactory.py", line 46, in hid_transport
    from .cyhidapi import CyHidApiTransport
  File "C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\tools\libs\pyedbglib\hidtransport\cyhidapi.py", line 9, in <module>
    raise ImportError(MSG)
ImportError: This transport class requires cython and hidapi to be installed:

> pip install cython

> pip install hidapi

the selected serial port 
 does not exist or your board is not connected
`

ITDiscovery, 2022-01-06:
Sorry about being late to the party, but yeah...I can confirm microPaul's issue and I also went back to 1.3.10 and it works fine.

I did try doing a pip3 install hidapi, but that blew chunks on my MacBook. In between doing some asm samples, pls let me know if I can help with testing.

jeffs555, 2022-01-07:
Was having the same problem. Replaced platform.txt with the one from that fix upload bug #221 link above and then it couldn't even find python3 directory. 

Just deleted    {upload.with_programmer_not_available_for_bootloader_boards}    from the last line of the 1.4.4 platform.txt and it appears to be working.

SpenceKonde, 2022-01-08:
Sorry I have had a very busy week and am only now getting around to rekeasing the new versiobn.

jeffs555, 2022-01-08:
No need to be sorry. Just getting started with AVR128DB and very appreciative of all your hard work.

ITDiscovery, 2022-01-08:
Confirmed on jeffs555's fix and I am on a Mac. Nice work Jeff!! 

No worries Spence on any delay's for getting to 1.4.5 (at least from me) your efforts are very well appreciated.

SpenceKonde, 2022-01-11:
1.4.5 is available

closed 2022-01-11 by SpenceKonde