SpenceKonde/DxCore #222: teeny tiny nit - bootloader LED flash

microPaul, 2021-12-30:
I love the fact that you flash an LED on A7 when the bootloader is installed.  My oscilloscope tells me that it's three flashes, but to my old slow eyes and slow brain it's difficult for me to tell if it's 2 or 3 flashes.   Next time you are working on to bootloader would it be possible to add a little more "off time" between the three flashes?  It looks like the off time is currently about 68 ms.  Maybe something around 150ms of off time would make the individual flashes more distinct.  Thanks.

SpenceKonde, 2022-01-04:
I will see what I can do - the blink is generated through very crude means, and is not intended to be exact.

SpenceKonde, 2022-01-12:
it looks like the flashing is 1/15th of a second per on or off period, generated by executing a 10-clock instruction 4,000,000/150 = 26,666 times. 
classic Optiboot does 1/16th, and megaTinyCore does 1/15th unless OSCCFG fuse is set for 16 MHz in which case it does 1/12th of a second.  As the comments make abundantly clear the triple blink pattern is in no way intended to be precise. 

It is intended to be distinctive enough that you can say "Oh, the bootloader is running" or "Hm, why isn't the bootloader running?" or - probably most usefully - "wtf, why is the bootloader running, it must be resetting when it shouldn't.... (I'd better re-read that section of the reset guide and review my code for causes of unwanted resets" I think it's more important that it look "the same as optiboot always does" to ensure that it's recognized for what it is, than that one be able to count the flashes.

closed 2022-01-12 by SpenceKonde