SpenceKonde/DxCore #223: Problems with "1.4.4\cores\dxcore/core_devices.h" with respect to RTC

microPaul, 2021-12-30:
I realize the Microchip sort of messed us up by changing some the core definitions between ATtiny1 and AVR-Dx even though the peripherals look like they are the same.

I'm trying to configured the PIT on an AVR128DB32 using the Arduino IDE and the DxCore by writing directly to the peripheral using the values Microchip specified in the MPLAB X IDE  AVR-Dx_DFP file "ioavr128db32.h".  However, when I try to do that with the Dx core 1.4.4 (or 1.3.9) I get the following error:

`In file included from C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\cores\dxcore/Arduino.h:27:0,
                 from sketch\DB32_Sleeping_Lighthouse_211230-01.ino.cpp:1:
p:\Users\p\Documents\Arduino\DB32_Sleeping_Lighthouse_211230-01\DB32_Sleeping_Lighthouse_211230-01.ino: In function 'void init32kOscRTCPIT()':
C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\cores\dxcore/core_devices.h:54:35: error: 'RTC_CLKSEL_INT1K_gc' was not declared in this scope
     #define RTC_CLKSEL_OSC1K_gc   RTC_CLKSEL_INT1K_gc
                                   ^
p:\Users\p\Documents\Arduino\DB32_Sleeping_Lighthouse_211230-01\DB32_Sleeping_Lighthouse_211230-01.ino:149:16: note: in expansion of macro 'RTC_CLKSEL_OSC1K_gc'
   RTC.CLKSEL = RTC_CLKSEL_OSC1K_gc;  //Select 1.024 kHz from 32KHz Low Power Oscillator (OSCULP32K) as clock source
                ^~~~~~~~~~~~~~~~~~~
C:\Users\p\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.4\cores\dxcore/core_devices.h:54:35: note: suggested alternative: 'RTC_CLKSEL_INT32K_gc'
     #define RTC_CLKSEL_OSC1K_gc   RTC_CLKSEL_INT1K_gc
                                   ^
p:\Users\p\Documents\Arduino\DB32_Sleeping_Lighthouse_211230-01\DB32_Sleeping_Lighthouse_211230-01.ino:149:16: note: in expansion of macro 'RTC_CLKSEL_OSC1K_gc'
   RTC.CLKSEL = RTC_CLKSEL_OSC1K_gc;  //Select 1.024 kHz from 32KHz Low Power Oscillator (OSCULP32K) as clock source
                ^~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board AVR DB-series (no bootloader).
`

This is a small problem as I can just use 0x_ _ values in my code rather than things like "RTC_CLKSEL_OSC1K_gc" instead, but I thought I should document the issue.

SpenceKonde, 2022-01-02:
Ack, got it backwards
INT1K s supposed to be defined to OSC1K

SpenceKonde, 2022-01-06:
this is fixed

closed 2022-01-06 by SpenceKonde