SpenceKonde/DxCore #224: Upload via Atmel-ICE on core 1.4.4 : Part not found

Di-Ny, 2022-01-03:
It seems that there is a {upload.workaround} not filled by anything, or inapropriately stuck to the part number since the 1.4.4 upgrade. I am using Visual Micro on Visual Studio. 

Here is the issue displayed in the output window:
```
Uploading 'Project' to 'AVR DB-series (no bootloader)' using 'Atmel-ICE (UPDI mode)'  
Error during upload using programmer  
The uploader process failed  
  
avrdude: AVR Part "avr128db48{upload.workaround}" not found.  
The uploader process failed  
The uploader returned an error  
Valid parts are:  
...
```

SpenceKonde, 2022-01-04:
uhoh. I'll look into this.

MX682X, 2022-01-04:
Something similar here: I was not able to upload anything to my Curiosity nano AVR128DA48 until I removed `{upload.with_programmer_not_available_for_bootloader_boards}` that was appended after `-c{protocol}` (in the platforms.txt file). Since this piece was not defined (I guess), it was passed as a string directly to avrdude which wasn't able to find a programmer with that name. I was uploading directly, not through the bootloader.

SpenceKonde, 2022-01-04:
Yeah that issue was fixed with 1.4.4, but I don't know why there is an upload.workaround in DxCore. There shouldn't be.

SpenceKonde, 2022-01-04:
And there *isn;t* any {upload.workaround} in DxCore! 

Do you also have megaTinyCore installed? That DOES have an upload.workaround, and if you were uploading to megaTinyCore, then switch to a DxCore part, due to an IDE bug, the following will occur:
* The selected programmer will not be shown (because the "selected" programmer is the one defined by megaTinyCore. That one is not an option when using DxCore, even though one of the same name is. 
* It will still try to use that programmer, however. 
* Which would work fine, except that IT USES THE OTHER CORE'S PLATFORM.TXT which contains references to things that are specific to that core. 

The fix is simple - select an option from the tools -> programmer menu.

MX682X, 2022-01-04:
After redownloading 1.4.4 through the boardmnaager to get the stock core, I still can't get it to work. I tried selecting another chip and reselected curiosity as programmer. Still getting 

```avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

avrdude: Can't find programmer id "curiosity_updi{upload.with_programmer_not_available_for_bootloader_boards}"

Valid programmers are:
...
```

MX682X, 2022-01-04:
And I just noticed you removed it like 30 minutes ago with  https://github.com/SpenceKonde/DxCore/commit/5ed9214587e3333bc68e6e2472b3d1ec6b05fbe2

SpenceKonde, 2022-01-04:
> And I just noticed you removed it like 30 minutes ago with 5ed9214

Yeah thought I committed that days ago. Still unpacking after a move so am
dusorganized

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2022-01-04:
Both cores are getting emergency releases tomorrow

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

Di-Ny, 2022-01-04:
Ok thank you, the simple fix is working. It also refused to burn the bootloader correctly for the same kind of reasons. 
I updated megaTinyCore (which was kinda old), restarted the IDE, changed a random option like BOD -> disabled. Then everything works again 👍

SpenceKonde, 2022-01-13:
The emergecny release from the past weekend corrects this issue.

closed 2022-01-13 by SpenceKonde