SpenceKonde/DxCore #226: New Wire.endTransmission() has incorrect return values.

SpenceKonde, 2022-01-06:
From #225 
Per https://www.arduino.cc/en/Reference/WireEndTransmission
'they should be:
```
0:success
1:data too long to fit in transmit buffer
2:received NACK on transmit of address
3:received NACK on transmit of data
4:other error
```

Currently it is acting like a write(), and returning the number of bytes written instead, causing success to look like failure and failure to look like success. 

The new behavior breaks the extremely common I2C scanner sketches (which I never use because I read the bloody datasheet and know what the address of the thing I'm talking to is without scanning every available address - but people always seem to use that sort of thing)

MX682X, 2022-01-06:
Easy to fix. Sorry for the inconvinience, I thought the implementation I was using as reference was returning the written bytes too.

SpenceKonde, 2022-01-06:
Yeah, I think I've got a fix ready. 

Looks like only requestFrom() returns a number of bytes, and no error code (you must infer that an error occurred when you notice that you didn't get as much as you asked for), while endTransaction returns an error code, and if it's not success, you know less than the intended number of bytes was transferred but not how many

The API could have been better designed here (maybe a 2-byte struct would have been better, with the number of bytes and an error code) but so could a lot of things.

commit 9ea42e3d (2022-01-06): Fix #226
closed 2022-01-06 by SpenceKonde