SpenceKonde/DxCore #229: AStyle CI checker is dead. What the bloody hell is it now? last month markdownlint blew up, now astyle?

SpenceKonde, 2022-01-15:
Continuous disintegration is more like it. 

Homepage linked in the log output doesn't resolve as a domain name.

SpenceKonde, 2022-01-24:
Yeah, that action specifically had vanished and was failing with a hostname not resolved for several days. 

I will have to look at that clangformat, it looks like it offers the flexibility that would let me start stylechecking the core proper (which is curently excluded because it astyle doesn't let you indent all preprocessor conditionals, and I can't read my own code without that))

closed 2022-01-24 by SpenceKonde

per1234, 2022-01-24:
The [`per1234/artistic-style-action`](https://github.com/per1234/artistic-style-action) action used in that workflow is something I created (it actually was made specifically for https://github.com/SpenceKonde/megaTinyCore/pull/236). I haven't done anything to it so I'm sure the problem wasn't related to my repo. But my action downloads Artistic Style from its resting place on SourceForge, so it might have been a transient outage in that system.

Even though I am now recommending ClangFormat, I will definitely keep the Artistic Style action alive. It is a fairly simple action, so not likely to require frequent maintenance.