SpenceKonde/DxCore #232: Serial issue/not working on avr128db48 at least?

apws, 2022-01-17:
After testing the same sketch with 4 different boards - cnano avr 128db48, cnano tiny1627, XMC2Go and Leonardo in DxCore, it seems that Serial does nothing - I have identified CDC pins also (here PB1,PB0) and flashed correct optiboot, but Serial monitor shows nothing in arduino ide, nor in vscode/arduino-ext. Initially I thought that something is very bad in vscode/arduino, so then I identified that my print(".") are not shown as NEVER flushed without any println() ... but thats really problem of theirs serial monitor.

I tried any combination of compilation/upload in arduino ide too, but serial simply doesnt sent nothing to monitor, or there is some weird behavior of settings (baud?) ... I dont know.

the sketch used is below
[main.txt](https://github.com/SpenceKonde/DxCore/files/7883785/main.txt)

apws, 2022-01-18:
well, now I placed on breadboard plain avr128db28 and everything works with PA0,PA1 as UART
So something was weird only with CuriosityNano avr128db48 and his urt3 PB0, PB1 ??

SpenceKonde, 2022-01-19:
I had the same problem with curiosity nano.... There;s something weird about the serial on those things, it's something with the USB interface they have... I dislike the boards enough that I haven't investigated further.

closed 2022-01-19 by SpenceKonde