SpenceKonde/DxCore #239: Question to analogReadEnh( )

mikrocoder, 2022-01-23:
Hi,

when I write
```analogReadEnh(PIN_PD5, ADC_ACC64)```
or
```analogReadEnh(PIN_PD5, ADC_ACC_64)  ```
then ADC_ACC64 or ADC_ACC_64 is not known.
However, according to your description it should work like this.
What am I doing wrong?
Thanks.

closed 2022-01-24 by SpenceKonde

SpenceKonde, 2022-01-24:
Agree, it should, what you're doing wrong is using a core that never got the constants copied to it's Arduiino.h, and is maintained by someone who isn't very good at testing new features. 

I really ought to have a sketch that's compiled as part of CI test that tries all the legal arguments to that function and makes sure they compile.

SpenceKonde, 2022-01-24:
These are the defines that were missing

```
#define ADC_ACC2        (0x81)
#define ADC_ACC4        (0x82)
#define ADC_ACC8        (0x83)
#define ADC_ACC16       (0x84)
#define ADC_ACC32       (0x85)
#define ADC_ACC64       (0x86)
#define ADC_ACC128      (0x87)
```

mikrocoder, 2022-01-24:
Hi,

all good, can happen. Thanks. You will never find all the errors through your own tests, because always tests the "wrong" parts that work anyway.  :-)   Own experience.  :-)

Another question.
Is there an overview which registers you change by default? For example Portmux, USART, I2C, Timer etc.?

commit 2fdea6d4 (2022-01-24): Fix #239 (ADC_ACCn constants missing from DxCore)

Also correct related error message).