SpenceKonde/DxCore #240: DA28 - interrupts not working on port F

obdevel, 2022-01-28:
I've been troubleshooting a problem and I can't get pins PF0 or PF1 to respond to a pin change interrupt. I've tried all three interrupt menu options and even swapped out the chip in case I'd blown the pin. It works fine on e.g. port D. Unfortunately, the board is already laid out to use PF1, otherwise I could work around it.

```
// test code
pinMode(INTPIN, INPUT_PULLUP);
attachInterrupt(INTPIN, isr1, LOW);        // also tried FALLING
```

The interrupt on PF1 is driven by an MCP2515 CAN controller chip but I've removed it and just used a jumper to test the signal.

Chip: AVR128DA28 + bootloader
Core version: 1.4.6 latest
IDE version: 1.8.19 latest
macOS 10.13.6

[image]

SpenceKonde, 2022-01-28:
That's because you are using a 28 pin part and I have't done a release to
fix pins 0 and 1 of all ports on 28 pin parts yet because I've been working
on ATTinyCore and I cannot seem to get a single UPDI programmer to work.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

obdevel, 2022-01-28:
Ok, no rush, so long as it's a known issue and a solution is coming. I have other parts of this project I can work on.

SpenceKonde, 2022-01-31:
Fixed in current released version, 1,4.6

closed 2022-02-01 by SpenceKonde

hoka83, 2024-04-24:
> I've been troubleshooting a problem and I can't get pins PF0 or PF1 to respond to a pin change interrupt. I've tried all three interrupt menu options and even swapped out the chip in case I'd blown [...]

hello, please can you explain how can i program avr128DA28 using arduino uno as updi