SpenceKonde/DxCore #256: RTC_CLKSEL Macros Undefined

rocketscream, 2022-02-19:
I wrote the [RTCAVRZero](https://github.com/rocketscream/RocketScream_RTCAVRZero) library but it was heavily tested on @MCUdude [MegaCoreX](https://github.com/MCUdude/MegaCoreX). When trying to compile the library on an AVR128DB32 (or any of the variant I believe), an error of missing macro like RTC_CLKSEL_INT32K_gc and RTC_CLKSEL_TOSC32K_gc pops out. By right I believe both of the device family are tightly related with most register having same name but as you mentioned in the code, it is not the case. So, I took a look on the core_devices.h and found that maybe you might have taken care of this. Un-commenting BACKWARD_COMBATIBILITY_MODE as the name suggest, I thought it would solve the problem, but it doesn't. 

With the `BACKWARD_COMBATIBILITY_MODE` un-commented:
```
/home/rocketscream/Arduino/libraries/RocketScream_RTCAVRZero/src/RocketScream_RTCAVRZero.cpp:46:18: error: 'RTC_CLKSEL_TOSC32K_gc' was not declared in this scope
     RTC.CLKSEL = RTC_CLKSEL_TOSC32K_gc;
                  ^~~~~~~~~~~~~~~~~~~~~
/home/rocketscream/Arduino/libraries/RocketScream_RTCAVRZero/src/RocketScream_RTCAVRZero.cpp:46:18: note: suggested alternative: 'RTC_CLKSEL_XOSC32K_gc'
     RTC.CLKSEL = RTC_CLKSEL_TOSC32K_gc;
                  ^~~~~~~~~~~~~~~~~~~~~
                  RTC_CLKSEL_XOSC32K_gc
/home/rocketscream/Arduino/libraries/RocketScream_RTCAVRZero/src/RocketScream_RTCAVRZero.cpp:52:17: error: 'RTC_CLKSEL_INT32K_gc' was not declared in this scope
    RTC.CLKSEL = RTC_CLKSEL_INT32K_gc;
                 ^~~~~~~~~~~~~~~~~~~~
/home/rocketscream/Arduino/libraries/RocketScream_RTCAVRZero/src/RocketScream_RTCAVRZero.cpp:52:17: note: suggested alternative: 'RTC_CLKSEL_OSC32K_gc'
    RTC.CLKSEL = RTC_CLKSEL_INT32K_gc;
                 ^~~~~~~~~~~~~~~~~~~~
                 RTC_CLKSEL_OSC32K_gc
```

SpenceKonde, 2022-02-24:
Goddamnit, I had typed out like a 2 page reply to this! Where he hell did it go? 

Read that mode name again, and the text above it. That is not a typo. That define is intended to disable all the compatibility defines like the ones you were hoping to use.

Default is backwards compatibility. Uncommenting that line gives you combatability, which is the exact opposite. 

If you simply undo those changes you made to core_devices.h, those macros should work.   I don't see how that could be because unless in backwards combatibility mode, we check for RTC_CLKSEL register being defined (it is) and then define them....  

```c
  #if defined(RTC_CLKSEL)
    #define RTC_CLKSEL_INT32K_gc  RTC_CLKSEL_OSC32K_gc
    #define RTC_CLKSEL_INT1K_gc   RTC_CLKSEL_OSC1K_gc
    #define RTC_CLKSEL_TOSC32K_gc RTC_CLKSEL_XTAL32K_gc
  #endif
````

Are you saying that with the core in it's stock condition and that line left commented out, it doesn't work????

There may be some relevant design decisions regarding what cases should be handled and not handled, and what things should receive similar treatment on megaTinyCore,

The cases I am aware of that apply are:
RTC clock selections

TCA event inputs - there was previously one event channel input. Now there are two, and accordingly two bitfields to select the action (EVACTA, EVACTB). 

TCB - there was previously no option to cascade or count on events; and with this change the names of the clock options got changed. 
I know I did something about this since code works, but I don't recall where or how I handled it. 

Depending on the version of the toolchain package (seriously Microchip?)
FRQSEL/FREQSEL - different versions of the ATPACKs have different names for this register and the group codes. 
CRYSTAL vs XTAL - for external HF

For the tinyAVR parts specifically: 
EVSYS on 0/1-series tiny significantly different. Only impacts tiny 0/1-series compatibility, handled by Event library for those parts only.

SpenceKonde, 2022-02-24:
Aaaah! It's XOSC32k, not XTAL32k

otherwise it should work as intended and that will be corrected in the immenant 1.4.9 release, (because we have a critical bugfix to attachInterrupt - there was another bug in the new implementation.

closed 2022-02-24 by SpenceKonde