SpenceKonde/DxCore #266: Add support for Generic Autobaud Mode

SpenceKonde, 2022-03-12:
My current thinking is to adapt begin to take a signed baud value instead of unsigned (or maybe it would be more efficient to just set the MSB? I dunno)

Either way, it would get either multiplied by -1 or have the MSB stripped if that was detected; result would be starting baud rate. A new method would be added: bool AutoBaudWFB(); it would return false if the command was invalid vecause we were not in autobaud mode, otherwise it would setthe WFB bit. 

Mentioning this because apparently there are people bullying the current implementatiion into working with autobaud. I didn't think it sounded very useful, but not everyone agrees. LIN constrained autobaud will not be supported.

microPaul, 2022-03-25:
I don't have any use cases for autobaud.  usually I find autobaud to be more of a nuisance in the systems I've used that make use of autobaud.

SpenceKonde, 2022-03-27:
I tend to agree on it being less than useful. But I've had several people ask about it, and it's not hard to implement like that...

SpenceKonde, 2022-06-18:
This is implemented but untested now.

closed 2022-06-18 by SpenceKonde