SpenceKonde/DxCore #272: bootloader file location

ArduinoShop, 2022-04-09:
I just set up my IDE (1.8.10) with this great boards package - I don't have my chips yet but I did a dry run just compiling "blink".
I selected the AVR128DB28 with Optiboot.

All went well except the IDE showed and error with the location of the bootloader file - seemed to be repeating a PATH variable..

I changed line 1989 of "boards.txt"  to "avrdbopti.bootloader.file=hex/{bootloader.class}{bootloader.port}{bootloader.timeout}.hex"
and the problem went away.

Just waiting for the chips to arrive to dive in...

ArduinoShop.ca

SpenceKonde, 2022-04-09:
Hmm, that's strange. Can you please post the full output in the console? 

When I try to reproduce, i get File: C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr/bootloaders/hex/optiboot_dx128_ser0_alt.hex
And the file is present at
C:\Users\Spence\Documents\Arduino\hardware\DxCore\megaavr\bootloaders\hex\optiboot_dx128_ser0_alt.hex

ArduinoShop, 2022-04-16:
I foolishly didn't back up the boards.txt file so tried to reconstruct the "bad" version...
Please Note that I am using Linux(Ubuntu 21.10)

hopefully this will at least point you in the direction...  seems to be a $PATH shell variable problem

like it is already set but it is in the "boards.txt" file as well.

This is what I get

Bootloader file specified but missing: /home/pb/.arduino15/packages/DxCore/hardware/megaavr/1.4.10/bootloaders/home/pb/.arduino15/packages/DxCore/hardware/megaavr/1.4.10/bootloaders/hex/optiboot_dx128_ser0_alt.hex

??? somewhere the $HOME shell variable seems to be getting duplicated ??

I have since been able to upload the bootloader using jtag2updi and did not encounter this problem.  So far no joy with SerialUPDI - but resistor/diode problem I think..

SpenceKonde, 2022-04-16:
Sad to  say I don't have much to offer here.

You didn't by chance install the ide from a package manager, righ? Tjhose are known to generate broken installs

SpenceKonde, 2022-04-28:
And yeah as you guessed, $HOME should only be there once. But this problem seems unique to your system; we have many users on linux who aren't experiencing that problem. If you haven't solved it, I would suggest posting on Arduino forum and see if thems folk over there have any ideas. Feel free to reopen this if more information comes to light from there that points to the core - but I think this is an Arduino bug or system config issue. (or, as I mentioned, a bug caused y using a package mananger version of Arduino; never install arduino from a package manager. Because the people maintaining those package repos don't use Arduino or know how it's supposed to work (and the Arduino people don't know how to do a linux software repo version of the IDE - so they can't march over to the mailing list or github repo and tell them what they're doing wrong). If you're arouind niche software for long, you start to see this everywhere..... at my old job, the Kerberos people only knew how security worked through that, and the people who made the components of the product which had to connect securely didn't know wtf that was, only that it was hard to make work.

closed 2022-04-28 by SpenceKonde