SpenceKonde/DxCore #278: Best #define to uniquely identify this core ?

obdevel, 2022-05-21:
What's the best #define or combination thereof to uniquely identify this core and AVR family in portable code ?

ARDUINO_ARCH_MEGAAVR is common with the Arduino NanoEvery (4809). 
\_\_AVR_XMEGA__ seems unique to this core/family.

I need to differentiate between the various AVR families and Tiny families, as well as others (ancient AVR, ESP, etc, etc) but I have those covered already.

I can't imagine there'll ever be a choice of cores for this AVR family (!!) so they're the same thing, to all intents and purposes.

I'm currently doing, which seems to work:

```
#ifdef ARDUINO_ARCH_MEGAAVR
#ifdef __AVR_XMEGA__
// it's DxCore
#else
// it's a NanoEvery
#endif
#else
// it's something else
// etc, etc, etc
#endif
```

FWIW the only difference I've come across in my code so far is TCB_CLKSEL_DIV2_gc vs TCB_CLKSEL_CLKDIV2_gc.

SpenceKonde, 2022-05-21:
For these parts use:
if defined(DXCORE) and that's a string representation of the version
DXCORE_MAJOR DXCORE_MINOR amd DXCORE_PATCH
if defined(MEGATINYCORE)  and that's a string representation of the
version    MEGATINYCORE_MAJOR MEGATINYCORE_MINOR amd DXCORE_PATCH and
MEGATINYSERIES for 0, 1, or 2-series parts,. i
if defined(MEGACOREX) for megaAVR 0-series using that core
if none of the above, but __AVR_XMEGA__ or ARDUINO_ARCH_MEGAAVR is definedthen
it's a modern (post 2016) chip..but not running on a third party core that
actually works correctly (ie it's using an official core).. Expect trouble
ahead.

From the hardware side,
Internally there are very few differences other than the explicit changes
(ADC totally different on 2-series, TCD only present for 1-series and Dx.
CCL int vector for non 0/1-series, and the second TCA- event input in
inadition to those TCB changes.

If __AVR__ARCH_ _ == 103,flash is less than 48k and all is mapped to the
datas pace -This is always the case on a modern AVR that is't a Dx or Ex. t
on Dx (and apparently Ex) series may have more than that muich flash and
will show as 102 (64k) or 104 (128k), and only have a 32k chunk of flash
memory mapped at any given time, so . When __AVR_ARCH__ == 103, the linker
knows it can automatically put constants into progmem and not copy them to
ram. Otherwise that step is required and that forces back things like the
F() macro on those devices)

The cores on the other hand, internally.
DxCore and megaTinyCore are very similar to eachother but less similar to
MegaCoreX, and very different from the official arduino core, the use of
which cannot be recommended.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***

SpenceKonde, 2022-05-21:
``

closed 2022-05-21 by SpenceKonde