SpenceKonde/DxCore #279: Clash/duplication between Logic & Comparator libraries

grandaspanna, 2022-05-22:
Getting a compilation error if I include both the Logic and Comparator libraries.

Probably a simple fix, but I don't know how/where.

With minimalist program:
```
#include <Logic.h>
#include <Comparator.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

If get the following errors:
```
Compiling sketch...
"C:\\Users\\blahblah\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=avr128da32 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DCORE_ATTACH_ALL -DTWI_MORS_SINGLE -DMILLIS_USE_TIMERB2 -DARDUINO=10819 -DARDUINO_avrda -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.4.10\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=4UL -DDXCORE_PATCH=10UL -DDXCORE_RELEASED=1 "-IC:\\Users\\blahblah\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\cores\\dxcore/api/deprecated" "-IC:\\Users\\blahblah\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\cores\\dxcore" "-IC:\\Users\\blahblah\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\variants\\32pin-standard" "-IC:\\Users\\blahblah\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\libraries\\Logic\\src" "-IC:\\Users\\blahblah\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\libraries\\Comparator\\src" "C:\\Users\\blahblah\\AppData\\Local\\Temp\\arduino_build_57917\\sketch\\CCL_comp_test.ino.cpp" -o "C:\\Users\\blahblah\\AppData\\Local\\Temp\\arduino_build_57917\\sketch\\CCL_comp_test.ino.cpp.o"
In file included from C:\Users\blahblah\Documents\Arduino\CCL_comp_test\CCL_comp_test.ino:2:0:
C:\Users\blahblah\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\libraries\Comparator\src/Comparator.h:63:8: error: multiple definition of 'enum out::output_t'
   enum output_t : uint8_t {
        ^~~~~~~~
In file included from C:\Users\blahblah\Documents\Arduino\CCL_comp_test\CCL_comp_test.ino:1:0:
C:\Users\blahblah\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\libraries\Logic\src/Logic.h:165:8: note: previous definition here
   enum output_t : uint8_t {
        ^~~~~~~~
In file included from C:\Users\blahblah\Documents\Arduino\CCL_comp_test\CCL_comp_test.ino:2:0:
C:\Users\blahblah\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\libraries\Comparator\src/Comparator.h:70:8: error: multiple definition of 'enum out::pinswap_t'
   enum pinswap_t : uint8_t {
        ^~~~~~~~~
In file included from C:\Users\blahblah\Documents\Arduino\CCL_comp_test\CCL_comp_test.ino:1:0:
C:\Users\blahblah\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\libraries\Logic\src/Logic.h:169:8: note: previous definition here
   enum pinswap_t : uint8_t {
        ^~~~~~~~~
Using library Logic at version 1.1.2 in folder: C:\Users\blahblah\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\libraries\Logic 
Using library Comparator at version 1.2.0 in folder: C:\Users\blahblah\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\libraries\Comparator 
exit status 1
Error compiling for board AVR DA-series (no bootloader).

```

SpenceKonde, 2022-05-22:
@MCUdude  - what's the correct way to fix this one?

MCUdude, 2022-06-12:
I've come up with a fix: https://github.com/MCUdude/MegaCoreX/issues/150

SpenceKonde, 2022-07-23:
These changes need to be pulled in from megaTinyCore now and corresponding changes made to ZCD and Opamp.

SpenceKonde, 2022-08-13:
I think this is fixed but can't tell until core compiles anything successfully

SpenceKonde, 2022-11-09:
The fix has been ported and this core has some functionality intact. 

As long as you have millis disabled at least. If you don't, everything is fucked

closed 2022-11-09 by SpenceKonde