SpenceKonde/DxCore #281: digitalWriteFast error: badArg("Fast digital pin must be a constant")

nabelekt, 2022-05-27:
If I build
```
void setup()
{
    toggle_pin_on_off(PIN_PD7);
    toggle_pin_on_off(PIN_PD6);
    toggle_pin_on_off(PIN_PD7);
}

void loop()
{
}

void toggle_pin_on_off(uint8_t pin)
{
    digitalWriteFast(pin, HIGH);
    digitalWriteFast(pin, LOW);
}
```
for AVR128DA28, I get:
```
...
Linking everything together...
"C:\\Users\\nabel\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-gcc" -Wall -Os -g -flto -fuse-linker-plugin -mrelax -Wl,--gc-sections,--section-start=.text=0x0,--section-start=.FLMAP_SECTION1=0x8000,--section-start=.FLMAP_SECTION2=0x10000,--section-start=.FLMAP_SECTION3=0x18000 -mmcu=avr128da28 -Wl,-u,vfprintf -lprintf_flt -o "C:\\Users\\nabel\\AppData\\Local\\Temp\\arduino_build_231818/digitalWriteFast_badArg.ino.elf" "C:\\Users\\nabel\\AppData\\Local\\Temp\\arduino_build_231818\\sketch\\digitalWriteFast_badArg.ino.cpp.o" "C:\\Users\\nabel\\AppData\\Local\\Temp\\arduino_build_231818/..\\arduino_cache_198558\\core\\core_8108a44d374121fbcdadd8359ce76c9f.a" "-LC:\\Users\\nabel\\AppData\\Local\\Temp\\arduino_build_231818" -lm
In function 'check_constant_pin',
    inlined from 'digitalWriteFast' at C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore\wiring_digital.c:424:3,
    inlined from 'toggle_pin_on_off' at S:\google_drive\cattle_tracker\development\code\repos\ear_tag\issue_testing\digitalWriteFast_badArg\digitalWriteFast_badArg.ino:14:21:
C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore\Arduino.h:58:5: error: call to 'badArg' declared with attribute error: 
     badArg("Fast digital pin must be a constant");
     ^
In function 'check_constant_pin',
    inlined from 'digitalWriteFast' at C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore\wiring_digital.c:424:3,
    inlined from 'toggle_pin_on_off' at S:\google_drive\cattle_tracker\development\code\repos\ear_tag\issue_testing\digitalWriteFast_badArg\digitalWriteFast_badArg.ino:15:21:
C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore\Arduino.h:58:5: error: call to 'badArg' declared with attribute error: 
     badArg("Fast digital pin must be a constant");
     ^
lto-wrapper.exe: fatal error: C:\Users\nabel\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4b/bin/avr-gcc returned 1 exit status
compilation terminated.
c:/users/nabel/appdata/local/arduino15/packages/dxcore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino4b/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board AVR DA-series (no bootloader).
```

I have found that some variations on the code build fine. All of the below build ok:
```
void setup()
{
    toggle_pin_on_off(PIN_PD7);
    toggle_pin_on_off(PIN_PD6);
    // toggle_pin_on_off(PIN_PD7);
}

void loop()
{
}

void toggle_pin_on_off(uint8_t pin)
{
    digitalWriteFast(pin, HIGH);
    digitalWriteFast(pin, LOW);
}
```
```
void setup()
{
    digitalWriteFast(PIN_PD7, HIGH);
    digitalWriteFast(PIN_PD7, LOW);
    toggle_pin_on_off(PIN_PD6);
    toggle_pin_on_off(PIN_PD7);
}

void loop()
{
}

void toggle_pin_on_off(uint8_t pin)
{
    digitalWriteFast(pin, HIGH);
    digitalWriteFast(pin, LOW);
}
```

Any ideas what is going on here and how to fix the issue? Thanks!

nabelekt, 2022-05-27:
Using `digitalWrite` inplace of `digitalWriteFast` works fine. I'll do that for now.

SpenceKonde, 2022-05-31:
The issue is exactly what it says, The pin argument for digitalWriteFast must be constant, In the code you posted it is not (it is this lookup of the arduino pin number, as well as turning off of PWM, that is responsible for digitalWrite() taking 100-200 times as long as digitalWriteFast)

In the ones that work, the function that calls digitalWriteFast is only being referenced twice, and the optimizer inlines it, so your code becomes 
```
void setup()
{
    //with 2 calls ifs no less spaceefficient to inline the two calls
    digitalWriteFast(PIN_PD7, HIGH); //sbi VPORTD.OUT, 7
    digitalWriteFast(PIN_PD7, LOW);  //sbi VPORTD.OUT, 7
    digitalWriteFast(PIN_PD6, HIGH); //sbi VPORTD.OUT, 6
    digitalWriteFast(PIN_PD6, LOW);  //sbi VPORTD.OUT, 6
   // 4 isns total
}
```
Instead of 
```
void setup()
{
    toggle_pin_on_off(PIN_PD7); //rcall 
    toggle_pin_on_off(PIN_PD6); // rcall 
    // 2 isns here
}
void toggle_pin_on_off(uint8_t pin) {
  digitalWriteFast(pin,HIGH); //sbi
  digitalWriteFast,pin,LOW);  //cbi
  // `ret`urn. 
 // 5 instructions, so the inlined approach wins
}
```

But when you put in a third call - now the function call overhead is is no longer larger than the overhead of two digitalWriteFast()
So the optimizer, instructed to optmize for size, compiles

```
void setup() {
  toggle_pin_on_off(PIN_PD7); //ldi r24, PIN_PD7; rcall toggle_pin_on_off(); 
  toggle_pin_on_off(PIN_PD6); //ldi r24, PIN_PD7; rcall toggle_pin_on_off(); 
  toggle_pin_on_off(PIN_PD5); //ldi r24, PIN_PD7; rcall toggle_pin_on_off();
}
// 6 isns
void toggle_pin_on_off(uint8_t pin) {
  digitalWriteFast(pin,HIGH);
  digitalWriteFast,pin,LOW); 
}
//3 isns counting the ret: 9 
```
Now one might observe that a third call should also be smaller with inlining (6 isns), or even 4 (8 isns). Thisis an example of the compiler sucking at times. 

Anyway the moment it's inlined,  the pin passed to digitalWriteFast is no longer constant, so it is not possible to turn it into the a SBI or CBI and throws the error. 
I think this would fix it correctly by forcing the inlining
```
inline __attribute__((always_inline)) void toggle_pin_on_off(uint8_t pin) {
  digitalWriteFast(pin,HIGH);
  digitalWriteFast,pin,LOW); 
}
```
As would 

#define toggle-pin-on-off(pin) (digitalWriteFast(pin,HIGH); digitalWriteFast(pin,LOW);)

closed 2022-05-31 by SpenceKonde