SpenceKonde/DxCore #284: MVIO Wire.swap(2) not working on AVR32DB28?

nomakewan, 2022-06-18:
I have a project where I am communicating with a KXTJ3-1057 accelerometer over I2C in Fast Mode. Currently this is being prototyped on a breadboard using the AVR32DB28. If I use an external logic level shifter (NOYITO PCA9306) this code works perfectly fine using PA2/PA3, exactly as it did with the old ATMEGA328P I had been using before.

However, if I enable Multi-Voltage IO, and follow Microchip's design specification for using the internal OPAMP to drive 2.5V to VDDIO2 (AN3636), and then set `Wire.swap(2)` before `Wire.begin()` to switch I2C to PC2/PC3, the sensor is dead as a doornail. There is in fact 2.5V going to both VDDIO2 and the KXTJ3-1057 but it appears the I2C communication isn't working. I have also tried `Wire.pins(PIN_PC2, PIN_PC3)` which did not work, as well as `PORTMUX.TWIROUTEA = (PORTMUX.TWIROUTEA & 0xFC) | 0x02` which did not work either. The KXTJ3-1057 is on an evaluation board which includes 2.1k pullups already. I have attempted physically adding 4.7k pullup resistors to the circuit, but to no avail.

Am I missing something here, or is there something funky going on with the AVR32DB28?

SpenceKonde, 2022-06-18:
Two quick questions here:

1. Sanity check: MVIO.STATUS == 1? If it's not, that implies that VDDIO2 is not being correctly powered.
2. Double check that the pins at at a HIGH logic level, and that the pins digitalRead() as HIGH. Do you see any output on the 'scope when trying to talk to the slave in MVIO mode?

nomakewan, 2022-06-18:
Well, you nailed it. I hadn't gotten around to getting serial data working yet, so I went ahead and burned a bootloader so I could run a single adapter with this chip. Sure enough, once the sensor is plugged in MVIO.STATUS is 0 on startup, and PIN_PC2 and PIN_PC3 are both low. Realizing that this is likely a voltage settle issue, I went ahead and delayed the code a little as a test and sure enough, MVIO.STATUS went to 1 and both pins went high, and the KXTJ3 was correctly initialized and responding as expected. Consider this issue closed!

While doing this experiment, however, I did find a bug in the "Burn Bootloader" function. If "Optiboot" is selected as the chip type and you click "Burn Bootloader," the command to prog.py is missing quotation marks around the various paths in the command. As my username has a space in it, this causes the Burn Bootloader to fail. I ran the command manually in a command prompt with quotation marks added around all the directory paths to get around this and successfully burn the bootloader.

closed 2022-06-18 by nomakewan

SpenceKonde, 2022-06-19:
Interesting. 

Worth noting that if the library does anything in the *constructor* rather than a begin/start/init/etc method, that happens before any core initialization - much less the call to setup; it's not a recipe for success, needless to say. I would consider it to be a library bug if it's performing I2C transactions in a class constructor, though there is a workaround to run code before constructors. See Ref_callbacks.md. But yeah I'm not the least bit surprised that if you were rushing through the initialization, the DAC wouldn't have come online in time. My solution would be a while 

```
uint16_t countdown == 65535; 
while (!MVIO.STATUS & 1 && countdown) { //spin until MVIO is ready // 2 + 1 c;lock;
    countdown --; // 1 clk
}
if (!countdown) {
  signal_error_condition to user()'
}
```

The dac has a non-instant conversion time (you also need to be careful about how much current you draw from the opamp. It's ah, a bit wimpy even with the opamp buffering it
I'll be sure to fix the quotes around filenames, will create new issue to keep from getting lost

nomakewan, 2022-06-19:
The library in question fires `Wire.swap(2)`, `Wire.begin()` and `Wire.setClock()` under a begin method, as it should. Also, I was just using the internal Opamp0 to supply the 2.5V to VDDIO2 while connecting the positive end to the VDD/2 wiper, not the DAC, per Microchip AN3636. However, clearly settle time was still an issue here, as adding a 10ms delay allowed everything to work. Oddly enough, having only a `while(MVIO.STATUS != 1)` did allow the system to wait until MVIO was up and also for PC2 and PC3 to go high, but wasn't enough for the TWI to be ready as my sensor still failed to initialize until I added an additional 10ms of delay. I'll try your countdown version next.

It is worth noting that the settle time and/or delay does not appear to have anything to do with the TWI peripheral or the PORTMUX system. The settle time lies either with the OPAMP, MVIO, or a combination of both. This seems strange to me as `Opamp0.start()` shouldn't return until the opamp is actually finished settling, and likewise having `while(MVIO.STATUS != 1)` should wait until MVIO is ready, and yet despite that, TWI was still not ready for communication.

As for current draw, the sensor peripheral at full tilt takes roughly 155 microamps according to its datasheet. The internal opamp should be more than capable of supplying that level of current, plus whatever MVIO itself may require.